coolsnowwolf/lede #9020: 麻烦看看这个是什么问题

GreatMichaelLee, 2022-03-07:
搜问题没搜到相关的。

SHELL= flock /home/builder/lede_x86/tmp/.expat-2.4.6.tar.xz.flock -c '          /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "expat-2.4.6.tar.xz" "de55794b7a9bc214852fdc075beaaecd854efe1361597e6268ee87946951289b" "" "@SF/expat"    '
MAKEFLAGS="" /home/builder/lede_x86/staging_dir/host/bin/ninja  -j1 -C /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6
ninja: Entering directory `/home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6'
[1/15] cd /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6 && /usr/bin/docbook2man /home/bu....6/doc/xmlwf.xml && mv XMLWF.1 /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6/doc/xmlwf.1
FAILED: CMakeFiles/xmlwf-manpage doc/xmlwf.1 /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6/CMakeFiles/xmlwf-manpage /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6/doc/xmlwf.1
cd /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6 && /usr/bin/docbook2man /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6/doc/xmlwf.xml && mv XMLWF.1 /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6/doc/xmlwf.1
Using catalogs: /etc/sgml/catalog
Using stylesheet: /usr/share/docbook-utils/docbook-utils.dsl#print
Working on: /home/builder/lede_x86/build_dir/hostpkg/expat-2.4.6/doc/xmlwf.xml
onsgmls:/usr/share/xml/docbook/schema/dtd/4.2/docbookx.dtd:112:17:E: "X20AC" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:40:19:E: "X21B6" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:41:19:E: "X21B7" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:42:17:E: "X21D3" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:43:18:E: "X21CA" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:44:18:E: "X21C3" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:45:18:E: "X21C2" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:46:18:E: "X21DA" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:47:17:E: "X219E" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:48:18:E: "X21C7" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:49:19:E: "X21A9" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:50:19:E: "X21AB" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:51:19:E: "X21A2" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:52:18:E: "X21BD" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:53:18:E: "X21BC" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:54:17:E: "X21D4" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:55:17:E: "X2194" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:56:19:E: "X21C6" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:57:19:E: "X21C4" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:58:18:E: "X21AD" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:59:19:E: "X21CC" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:60:19:E: "X21CB" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:61:16:E: "X21B0" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:62:16:E: "X21A6" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:63:18:E: "X22B8" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:64:18:E: "X2197" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:65:18:E: "X21CD" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:66:18:E: "X219A" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:67:18:E: "X21CE" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:68:18:E: "X21AE" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:69:18:E: "X219B" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:70:18:E: "X21CF" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:71:18:E: "X2196" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:72:18:E: "X21BA" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:73:18:E: "X21BB" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:74:18:E: "X21DB" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:75:17:E: "X21A0" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:76:18:E: "X21C9" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:77:19:E: "X21AA" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:78:19:E: "X21AC" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:79:19:E: "X21A3" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:80:18:E: "X219D" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:81:18:E: "X21C1" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:82:18:E: "X21C0" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:83:16:E: "X21B1" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:84:18:E: "X2198" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:85:18:E: "X2199" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:86:17:E: "X21D1" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:87:18:E: "X21C8" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:88:17:E: "X21D5" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:89:17:E: "X2195" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:90:18:E: "X21BF" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:91:18:E: "X21BE" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:92:18:E: "X21D0" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:93:18:E: "X2194" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:94:18:E: "X2194" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:95:18:E: "X21D2" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsb.ent:40:18:E: "X2210" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsb.ent:41:19:E: "X2306" is not a function name
[... 130 lines cut ...]
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:44:19:E: "X22C8" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:45:17:E: "X223D" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:46:18:E: "X22CD" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:47:17:E: "X224E" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:48:18:E: "X224F" is not a function name
onsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:49:17:E: "X2257" is not a function name
[... cut ...]
time: package/feeds/packages/expat/host-compile#0.28#0.10#0.30
    ERROR: package/feeds/packages/expat [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/expat/host/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

GreatMichaelLee, 2022-03-07:
麻烦看看是不是跟这个提交相关 https://github.com/coolsnowwolf/lede/commit/31ab8070884f79440d9db2c4f99d233934db2f0f

Dawneng, 2022-03-07:
好像是这个包下载不下来

Beginner-Go, 2022-03-07:
我看看

breakings, 2022-03-07:
试下 
```
make package/feeds/packages/expat/{clean,compile}
make package/feeds/packages/expat/host/{clean,compile}
```

GreatMichaelLee, 2022-03-07:
> make package/feeds/packages/expat/host/{clean,compile}

builder@Build-Server:~/lede_x86$ make package/feeds/packages/expat/{clean,compile}
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/clean
 make[2] -C feeds/packages/libs/expat clean
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/compile
 make[2] -C package/libs/toolchain compile
 make[2] -C feeds/packages/libs/expat compile
builder@Build-Server:~/lede_x86$ make package/feeds/packages/expat/host/{clean,compile}
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/host/clean
 make[2] -C feeds/packages/libs/expat host-clean
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/host/compile
 make[2] -C feeds/packages/libs/expat host-compile
    ERROR: package/feeds/packages/expat [host] failed to build.
make -r package/feeds/packages/expat/host/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：package/feeds/packages/expat/host/compile] 错误 1
builder@Build-Server:~/lede_x86$ make -j1 package/feeds/packages/expat/host/{clean,compile} V=sc
builder@Build-Server:~/lede_x86$ make package/feeds/packages/expat/{clean,compile}
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/clean
 make[2] -C feeds/packages/libs/expat clean
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/compile
 make[2] -C package/libs/toolchain compile
 make[2] -C feeds/packages/libs/expat compile
builder@Build-Server:~/lede_x86$ make package/feeds/packages/expat/host/{clean,compile}
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/host/clean
 make[2] -C feeds/packages/libs/expat host-clean
make[2]: Entering directory '/home/builder/lede_x86/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/builder/lede_x86/scripts/config'
 make[1] package/feeds/packages/expat/host/compile
 make[2] -C feeds/packages/libs/expat host-compile
    ERROR: package/feeds/packages/expat [host] failed to build.
make -r package/feeds/packages/expat/host/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：package/feeds/packages/expat/host/compile] 错误 1
builder@Build-Server:~/lede_x86$ make -j1 package/feeds/packages/expat/host/{clean,compile} V=sc

一样，上面加V=sc后错误和之前一样。

WYC-2020, 2022-03-07:
干得漂亮哈哈

Beginner-Go, 2022-03-07:
[image]
我这边似乎没问题

WYC-2020, 2022-03-07:
[image]
其实吧这个库到处都有，并且版本还不统一

Beginner-Go, 2022-03-07:
https://github.com/coolsnowwolf/packages/commit/f31dc1bc2932a7bd2cea20714b2c8b26222cdfb4
都是2.4.6

WYC-2020, 2022-03-07:
> [coolsnowwolf/packages@f31dc1b](https://github.com/coolsnowwolf/packages/commit/f31dc1bc2932a7bd2cea20714b2c8b26222cdfb4) 都是2.4.6

看到了今天才更新的

breakings, 2022-03-07:
可能是这几个地方的包都提供了host-build，再尝试清除一下 `make tool/expat/clean`

WYC-2020, 2022-03-07:
哈哈 看了只能用你们的万能大法了

doumeki, 2022-03-07:
> 哈哈 看了只能用你们的万能大法了

啥万能大法，已经make dirclan / make clean 过了

WYC-2020, 2022-03-07:
> > 哈哈 看了只能用你们的万能大法了
> 
> 啥万能大法，已经make dirclan / make clean 过了

你哪个不算万能大法

WYC-2020, 2022-03-07:
我刚都有重新全新编译了 也没遇到你们说的这个问题，哈哈看来bug 怕我

ricksuzade-maker, 2022-03-07:
也发现这个问题了，确实存在，连日志都一模一样

jliveusa, 2022-03-07:
同样问题~~

breakings, 2022-03-07:
可能是这个2.4.6的源代码有问题 升级到2.4.7就完全正常

xopenwrt, 2022-03-07:
同样问题

breakings, 2022-03-07:
> 同样问题

我PR了2.4.7 心急的可以手动改下试试

WYC-2020, 2022-03-07:
这么多翻车的哈哈

gswangmd-gssi, 2022-03-07:
更新后依然翻车

coolsnowwolf, 2022-03-07:
rm -rf tmp 重新生成缓存

gswangmd-gssi, 2022-03-07:
执行rm -rf tmp后一样

Biercenter, 2022-03-07:
刚编译发现出错后在犹豫要不要使用大法的时候上来看了一下，大家都一样，我就放心了😊

ricksuzade-maker, 2022-03-07:
我全删了重新拉，看看能不能解决

WYC-2020, 2022-03-07:
> 我全删了重新拉，看看能不能解决

他们已经升级了，说的是2.4.7已经解决，我还没升级明天再看

breakings, 2022-03-07:
我大概知道什么原因了 CMAKE_HOST_OPTIONS += $(OPTIONS) 没这个选项了

> > 我全删了重新拉，看看能不能解决
> 
> 他们已经升级了，说的是2.4.7已经解决，我还没升级明天再看

RodmanWang, 2022-03-07:
2.4.7并没有解决问题，大佬已经回滚了

breakings, 2022-03-07:
> 2.4.7并没有解决问题，大佬已经回滚了

得重新加回CMAKE_HOST_OPTIONS

WYC-2020, 2022-03-07:
> > 2.4.7并没有解决问题，大佬已经回滚了
> 
> 得重新加回CMAKE_HOST_OPTIONS

看到了，这个问题我这边没法复现，明天再看下吧，cmake有去掉什么东西？

breakings, 2022-03-07:
> > > 2.4.7并没有解决问题，大佬已经回滚了
> > 
> > 
> > 得重新加回CMAKE_HOST_OPTIONS
> 
> 看到了，这个问题我这边没法复现，明天再看下吧，cmake有去掉什么东西？

这个commit把它去掉了 加回来应该就正常了 https://github.com/coolsnowwolf/packages/commit/0b0a7334a04556ec62b0f9cbae99eca0d7d00a23

WYC-2020, 2022-03-07:
> > > > 2.4.7并没有解决问题，大佬已经回滚了
> > > 
> > > 
> > > 得重新加回CMAKE_HOST_OPTIONS
> > 
> > 
> > 看到了，这个问题我这边没法复现，明天再看下吧，cmake有去掉什么东西？
> 
> 这个commit把它去掉了 加回来应该就正常了 [coolsnowwolf/packages@0b0a733](https://github.com/coolsnowwolf/packages/commit/0b0a7334a04556ec62b0f9cbae99eca0d7d00a23)

这就不清楚了，明天看下，不过看这个错误感觉和这个关系不大，你们验证吧

breakings, 2022-03-07:
> 

已经本地验证就是这个错误，不加就不行，加了就正常通过，因为本来是有这个选项的。

WYC-2020, 2022-03-07:
> > > > > > 2.4.7并没有解决问题，大佬已经回滚了
> > > > > 
> > > > > 
> > > > > 得重新加回CMAKE_HOST_OPTIONS
> > > > 
> > > > 
> > > > 看到了，这个问题我这边没法复现，明天再看下吧，cmake有去掉什么东西？
> > > 
> > > 
> > > 这个commit把它去掉了 加回来应该就正常了 [coolsnowwolf/packages@0b0a733](https://github.com/coolsnowwolf/packages/commit/0b0a7334a04556ec62b0f9cbae99eca0d7d00a23)
> > 
> > 
> > 这就不清楚了，明天看下，不过看这个错误感觉和这个关系不大，你们验证吧
> 
> 已经本地验证就是这个错误，不加就不行，加了就正常通过，因为本来是有这个选项的。

哪的分析一下了，因为我看这些提交都是从op官方那边合并过来的，合到这里不行的分析分析了

breakings, 2022-03-07:
> rm -rf tmp 重新生成缓存

我重新提交PR了，https://github.com/coolsnowwolf/packages/commit/0b0a7334a04556ec62b0f9cbae99eca0d7d00a23 这里删除了CMAKE_HOST_OPTIONS会出事。

breakings, 2022-03-07:
> 

有些情况不能完全照搬官方的。

WYC-2020, 2022-03-07:
我好像知道原因，好像这个合并有的没有按op官方的合并，官方op的tools下面expat的makefile和官方就不一样，但是package又是一样的，可能是这个导致

WYC-2020, 2022-03-07:
你们没人验证我明天自己验证一下，这个合并有的时候是有连锁反应的不能只提交一部分

breakings, 2022-03-07:
> 

这个tools下面的本来也一样的，大雕又改回不用cmake了，但libs下面那个确实得按我PR的那样才行。

breakings, 2022-03-07:
> 我好像知道原因，好像这个合并有的没有按op官方的合并，官方op的tools下面expat的makefile和官方就不一样，但是package又是一样的，可能是这个导致

这个tools下面的本来也一样的，大雕又改回不用cmake了，但libs下面那个确实得按我PR的那样才行。

WYC-2020, 2022-03-07:
> > 我好像知道原因，好像这个合并有的没有按op官方的合并，官方op的tools下面expat的makefile和官方就不一样，但是package又是一样的，可能是这个导致
> 
> 这个tools下面的本来也一样的，大雕又改回不用cmake了，但libs下面那个确实得按我PR的那样才行。

明天折腾下吧哈哈

breakings, 2022-03-07:
> > > 我好像知道原因，好像这个合并有的没有按op官方的合并，官方op的tools下面expat的makefile和官方就不一样，但是package又是一样的，可能是这个导致
> > 
> > 
> > 这个tools下面的本来也一样的，大雕又改回不用cmake了，但libs下面那个确实得按我PR的那样才行。
> 
> 明天折腾下吧哈哈

我反正在本地验证过了

breakings, 2022-03-07:
> > > 我好像知道原因，好像这个合并有的没有按op官方的合并，官方op的tools下面expat的makefile和官方就不一样，但是package又是一样的，可能是这个导致
> > 
> > 
> > 这个tools下面的本来也一样的，大雕又改回不用cmake了，但libs下面那个确实得按我PR的那样才行。
> 
> 让大雕吧你的提交合并了吧

是要早点合并才好，要不然明天那些云编译的全炸锅了，哈哈。

heimeiqiusama, 2022-03-07:
我也遇到了这个问题

binge8, 2022-03-07:
我也遇到了这个问题

breakings, 2022-03-07:
> 我也遇到了这个问题

最新的PR已经加回来了 等大雕合并吧

closed 2022-03-07 by coolsnowwolf

WYC-2020, 2022-03-08:
刚看了一下，确实是这个合并不完整导致的错误，官方那边的commit。再这边有几个提交漏了，我先自己按官方的改了编译看下，应该不会出问题，这个提交其实就是去掉多余的编译，因为tools里面已经编译了，包里面不用再去编译这个