coolsnowwolf/lede #9024: ipsec server依赖冲突

binge8, 2022-03-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ipsec server依赖冲突

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：看截图
[image]

mitsukileung, 2022-03-08:
把luci-app-ipsec-vpnd去掉吧
ipsec-server也支持ipsec xauth psk的

binge8, 2022-03-08:
> 把luci-app-ipsec-vpnd去掉吧 ipsec-server也支持ipsec xauth psk的 我反而用l2tp ipsec psk 连不上

谢谢，我试试

WYC-2020, 2022-03-08:
把依赖修改成一样就解决了

mitsukileung, 2022-03-08:
> 把依赖修改成一样就解决了

两个LUCI有重复的吧？server也支持ipsec xauth psk，不知为何l2tp连不上

WYC-2020, 2022-03-08:
[image]
 是这里冲突了，luci 也要往init.d 复制ipsec 结果strongswan-ipsec这个也要复制，所以冲突了

WYC-2020, 2022-03-08:
对比一下 如果一样去掉一个地方的复制即可

closed 2022-03-08 by binge8

WYC-2020, 2022-03-11:
ipsec server的在线列表好像是失效的，无法展示在线客户

WYC-2020, 2022-03-11:
> > ipsec server的在线列表好像是失效的，无法展示在线客户
> 
> 好像只有用L2TP来连的时候才能查看在线列表吧

已经修复我自己

mitsukileung, 2022-03-11:
> > > ipsec server的在线列表好像是失效的，无法展示在线客户
> > 
> > 
> > 好像只有用L2TP来连的时候才能查看在线列表吧
> 
> 已经修复我自己

IPSec Xauth PSK也有可以显示在线用户？

mitsukileung, 2022-03-11:
> > > ipsec server的在线列表好像是失效的，无法展示在线客户
> > 
> > 
> > 好像只有用L2TP来连的时候才能查看在线列表吧
> 
> 已经修复我自己

大佬，我编译出来的 filebrowser 不能启动，文件权限默认是0664，需要手动把 /etc/init.d/filebrowser 这个文件权限改成 0775才可以运行。请问怎么可以在编译固件的时候把权限也改成0775 ？