coolsnowwolf/lede #9025: iptables无法初始化

leochien0102, 2022-03-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：iptables无法初始化

### (2) 路由器型号和固件版本

A：斐讯N1 最近lede源码 + 5.16.12-flippy-70+内核

### (3) 详细日志

A：
```
Tue Mar  8 07:55:37 2022 daemon.notice netifd: Network device 'eth0' link is up
Tue Mar  8 07:55:37 2022 kern.info kernel: [   20.569711] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Tue Mar  8 07:55:37 2022 daemon.notice netifd: bridge 'br-lan' link is up
Tue Mar  8 07:55:37 2022 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Mar  8 07:55:40 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables-save v1.8.7 (legacy): Cannot initialize: iptables who? (do you need to insmod?)
Tue Mar  8 07:55:40 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr:
Tue Mar  8 07:55:40 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables-save v1.8.7 (legacy): Cannot initialize: iptables who? (do you need to insmod?)
Tue Mar  8 07:55:40 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr:
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables v1.8.7 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Perhaps iptables or your kernel needs to be upgraded.
Tue Mar  8 07:55:41 2022 user.err ssr-rules[2583]: Can't redirect, please check the iptables.
Tue Mar  8 07:55:41 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: ssr-rules[2583]: Can't redirect, please check the iptables.
```

## 3. 补充说明：
**2月18日源码编译的固件没有此问题**

WYC-2020, 2022-03-08:
你这个是这个防火墙nat表没有，这个不清楚，暂时没遇到过

coolsnowwolf, 2022-03-08:
docker 问题要看看内核了，不归op管

WYC-2020, 2022-03-08:
> docker 问题要看看内核了，不归op管

他这个应该是ssr的可以去隔壁反馈 @leochien0102

leochien0102, 2022-03-08:
> docker 问题要看看内核了，不归op管

并不是Docker，只是N1需要替换amlogic的内核，2月18日的源码编译的同样使用的是flippy内核，包括最新的5.16.12-flippy-70+内核，都是没有问题的。就是新的op源码编译后的有问题，所有iptable表都是空的。
[image]
尝试了在正常使用版本中导出iptables，
```
iptables-save > /mnt/mmcblk2p4/iptables.save
```
并在新版中尝试导入，结果就是报错，无法初始化
```
iptables-restore < /mnt/mmcblk2p4/iptables.save 
iptables-restore v1.8.7 (legacy): iptables-restore: unable to initialize table 'raw'

Error occurred at line: 2
Try `iptables-restore -h' or 'iptables-restore --help' for more information.
```

leochien0102, 2022-03-08:
> > docker 问题要看看内核了，不归op管
> 
> 他这个应该是ssr的可以去隔壁反馈 @leochien0102

不算是ssr的问题，是整个iptable都没有了

WYC-2020, 2022-03-08:
组合怪 那无能为力 哈哈

leochien0102, 2022-03-08:
找到问题所在了，是固件升级脚本的锅，导致分区出错造成的，用U盘重新写入设备emmc，重新分区就好了。

closed 2022-03-08 by leochien0102