coolsnowwolf/lede #9027: ERROR: package/feeds/packages/node-yarn [host] failed to build.

netlovehf, 2022-03-08:
make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0'
mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp
touch /home/hzybb/lede/build_dir/hostpkg/node-v16.14.0/.built
touch /home/hzybb/lede/staging_dir/hostpkg/stamp/.node_installed
make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#5506.36#407.37#5906.63
make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
. /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.10.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10 -xf - --strip-components 1
[ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10/.prepared8a2eabb9d4fe455fbc6f609bd74ff67f_6664517399ebbbc92a37c5bb081b5c53
(cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10/.configured
CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_prefix=/home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10/host-install/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-b98Mz1OLC3 npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-b98Mz1OLC3 npm install -g /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10
TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
make[3]: *** [Makefile:71: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.10/.built] Error 1
make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
time: package/feeds/packages/node-yarn/host-compile#0.39#0.73#1.35
    ERROR: package/feeds/packages/node-yarn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
make[2]: Leaving directory '/home/hzybb/lede'
make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

breakings, 2022-03-08:
可能得升级下yarn的版本了

netlovehf, 2022-03-08:
> 可能得升级下yarn的版本了，如果你是本地编译的话 试下这个维护者的 https://raw.githubusercontent.com/nxhack/openwrt-node-packages/master/node-yarn/Makefile

请问本地源文件替换哪个文件？

breakings, 2022-03-08:
> > 可能得升级下yarn的版本了，如果你是本地编译的话 试下这个维护者的 https://raw.githubusercontent.com/nxhack/openwrt-node-packages/master/node-yarn/Makefile
> 
> 请问本地源文件替换哪个文件？

当然是feeds/packages/lang/node-yarn/Makefile

WYC-2020, 2022-03-08:
咋感觉很多更新都是没有经过验证的，都是一堆堆的错误 哈哈

breakings, 2022-03-08:
> 咋感觉很多更新都是没有经过验证的，都是一堆堆的错误 哈哈

这个yarn还是2020年的版本了，这个lang下面的包本来也不经常维护

WYC-2020, 2022-03-08:
> > 咋感觉很多更新都是没有经过验证的，都是一堆堆的错误 哈哈
> 
> 这个yarn还是2020年的版本了，这个lang下面的包本来也不经常维护，OP官方也还是这版本，一年多了。

这不管多久，这种提交更新应该要验证一下是否可以编译过才对，这种明显的错误，那就应该不暂时提交

WYC-2020, 2022-03-08:
看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈

breakings, 2022-03-08:
> 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈

看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。

breakings, 2022-03-08:
> > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> 
> 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。

WYC-2020, 2022-03-08:
> > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> 
> 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。

慢慢修吧，升级后可以测试一下14能兼容不，到时候提一个pr

breakings, 2022-03-08:
> > > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> > 
> > 
> > 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。
> 
> 慢慢修吧，升级后可以测试一下14能兼容不，到时候提一个pr

等一下那位老兄的结果，哈哈

netlovehf, 2022-03-08:
> > > > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> > > 
> > > 
> > > 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。
> > 
> > 
> > 慢慢修吧，升级后可以测试一下14能兼容不，到时候提一个pr
> 
> 等一下那位老兄的结果，哈哈

替换了一下，还是错误
make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.23#1.88#2.77
make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.node-v16.14.0.tar.xz.flock -c '      /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "node-v16.14.0.tar.xz" "05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623" "" "https://nodejs.org/dist/v16.14.0"    '
make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#0.20#8.61#11.49
make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c '          /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/yarn/-/yarn-1.22.17.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1216k  100 1216k    0     0   855k      0  0:00:01  0:00:01 --:--:--  855k
. /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1
[ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepared621e8a302ca41c8c98a0a93cfc5b16c0_6664517399ebbbc92a37c5bb081b5c53
(cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured
cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-EBuSm55VKe npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-EBuSm55VKe npm install --production --global-style --no-save --omit=dev --no-package-lock
TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
make[3]: *** [Makefile:115: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1
[... cut ...]
time: package/feeds/packages/node-yarn/host-compile#1.08#16.88#20.35
    ERROR: package/feeds/packages/node-yarn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
make[2]: Leaving directory '/home/hzybb/lede'
make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

breakings, 2022-03-08:
> > > > > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> > > > 
> > > > 
> > > > 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。
> > > 
> > > 
> > > 慢慢修吧，升级后可以测试一下14能兼容不，到时候提一个pr
> > 
> > 
> > 等一下那位老兄的结果，哈哈
> 
> 替换了一下，还是错误 make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/python/python3' time: package/feeds/packages/python3/host-compile#0.23#1.88#2.77 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.node-v16.14.0.tar.xz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "node-v16.14.0.tar.xz" "05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623" "" "https://nodejs.org/dist/v16.14.0" ' make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node' time: package/feeds/packages/node/host-compile#0.20#8.61#11.49 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/" '
> 
> * curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/yarn/-/yarn-1.22.17.tgz
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>   Dload  Upload   Total   Spent    Left  Speed
>   100 1216k  100 1216k    0     0   855k      0  0:00:01  0:00:01 --:--:--  855k
>   . /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1
>   [ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17
>   touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepared621e8a302ca41c8c98a0a93cfc5b16c0_6664517399ebbbc92a37c5bb081b5c53
>   (cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
>   touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured
>   cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-EBuSm55VKe npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-EBuSm55VKe npm install --production --global-style --no-save --omit=dev --no-package-lock
>   TypeError: Class extends value undefined is not a constructor or null
>   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
>   at Module._compile (node:internal/modules/cjs/loader:1103:14)
>   at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
>   at Module.load (node:internal/modules/cjs/loader:981:32)
>   at Function.Module._load (node:internal/modules/cjs/loader:822:12)
>   at Module.require (node:internal/modules/cjs/loader:1005:19)
>   at require (node:internal/modules/cjs/helpers:102:18)
>   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
>   at Module._compile (node:internal/modules/cjs/loader:1103:14)
>   at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
>   at Module.load (node:internal/modules/cjs/loader:981:32)
>   at Function.Module._load (node:internal/modules/cjs/loader:822:12)
>   at Module.require (node:internal/modules/cjs/loader:1005:19)
>   at require (node:internal/modules/cjs/helpers:102:18)
>   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
>   at Module._compile (node:internal/modules/cjs/loader:1103:14)
>   make[3]: *** [Makefile:115: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1
[... cut ...]
>   ERROR: package/feeds/packages/node-yarn [host] failed to build.
>   make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
>   make[2]: Leaving directory '/home/hzybb/lede'
>   make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
>   make[1]: Leaving directory '/home/hzybb/lede'
>   make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

哈哈，这个我也不懂什么原因了，我都很少编译yarn，只有一个adguardhome要用到。

Dawneng, 2022-03-08:
你是编译的5.4内核吗？我昨晚编译5.4内时出一样错，刚刚试了试5.10，正常过了。

breakings, 2022-03-08:
> 你是编译的5.4内核吗？我昨晚编译5.4内时出一样错，刚刚试了试5.10，正常过了。

node都能过，你是说的yarn这个包过了吗？

Dawneng, 2022-03-08:
> 

node都能过

Dawneng, 2022-03-08:
> > 你是编译的5.4内核吗？我昨晚编译5.4内时出一样错，刚刚试了试5.10，正常过了。
> 
> node都能过，你是说的yarn这个包过了吗？
刚才diffconfig看了一下
一个用CONFIG_NODEJS_14
另一个用CONFIG_NODEJS_16

netlovehf, 2022-03-08:
> 你是编译的5.4内核吗？我昨晚编译5.4内时出一样错，刚刚试了试5.10，正常过了。

编译的是5.15内核

netlovehf, 2022-03-08:
> 

 make package/feeds/packages/node-yarn/host/{clean,compile} V=s
make[1]: Entering directory '/home/hzybb/lede'
make[2]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
rm -rf /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built
rm -rf /home/hzybb/lede/staging_dir/hostpkg/stamp/.node-yarn_installed
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
time: package/feeds/packages/node-yarn/host-clean#0.12#0.46#0.66
make[1]: Leaving directory '/home/hzybb/lede'
make[1]: Entering directory '/home/hzybb/lede'
make[2]: Entering directory '/home/hzybb/lede/feeds/packages/libs/expat'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.expat-2.4.6.tar.xz.flock -c '        /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "expat-2.4.6.tar.xz" "de55794b7a9bc214852fdc075beaaecd854efe1361597e6268ee87946951289b" "" "@SF/expat"    '
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/libs/expat'
time: package/feeds/packages/expat/host-compile#0.21#0.80#0.98
make[2]: Entering directory '/home/hzybb/lede/feeds/packages/libs/libffi'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.libffi-3.3.tar.gz.flock -c '         /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "libffi-3.3.tar.gz" "72fba7922703ddfa7a028d513ac15a85c8d54c8d67f55fa5a4802885dc652056" "" "https://sourceware.org/pub/libffi/"    '
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/host-compile#0.15#0.34#0.37
make[2]: Entering directory '/home/hzybb/lede/package/utils/bzip2'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.bzip2-1.0.8.tar.gz.flock -c '        /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "bzip2-1.0.8.tar.gz" "ab5a03176ee106d3f0fa90e381da478ddae405918153cca248e682cd0c4a2269" "" "https://sourceware.org/pub/bzip2"    '
make[2]: Leaving directory '/home/hzybb/lede/package/utils/bzip2'
time: package/utils/bzip2/host-compile#0.20#0.10#0.26
make[2]: Entering directory '/home/hzybb/lede/feeds/packages/lang/python/python3'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.Python-3.9.10.tar.xz.flock -c '      /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "Python-3.9.10.tar.xz" "0a8fbfb5287ebc3a13e9baf3d54e08fa06778ffeccf6311aef821bb3a6586cc8" "" "https://www.python.org/ftp/python/3.9.10"    '
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.22#1.17#1.08
make[2]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.node-v16.14.0.tar.xz.flock -c '      /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "node-v16.14.0.tar.xz" "05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623" "" "https://nodejs.org/dist/v16.14.0"    '
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#0.30#3.47#2.32
make[2]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
. /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1
[ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.preparedd9d42c70f4f7bf6f76ac0d2ff2420471_6664517399ebbbc92a37c5bb081b5c53
(cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured
CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_prefix=/home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/host-install/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-gvS8SdEp1R np [... cut ...]
make[2]: *** [Makefile:71: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
time: package/feeds/packages/node-yarn/host-compile#0.48#0.34#0.73
    ERROR: package/feeds/packages/node-yarn [host] failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: package/feeds/packages/node-yarn/host/compile] Error 2

netlovehf, 2022-03-08:
> > > > > > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> > > > > 
> > > > > 
> > > > > 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。
> > > > 
> > > > 
> > > > 慢慢修吧，升级后可以测试一下14能兼容不，到时候提一个pr
> > > 
> > > 
> > > 等一下那位老兄的结果，哈哈
> > 
> > 
> > 替换了一下，还是错误 make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/python/python3' time: package/feeds/packages/python3/host-compile#0.23#1.88#2.77 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.node-v16.14.0.tar.xz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "node-v16.14.0.tar.xz" "05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623" "" "https://nodejs.org/dist/v16.14.0" ' make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node' time: package/feeds/packages/node/host-compile#0.20#8.61#11.49 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/" '
> > 
> > * curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/yarn/-/yarn-1.22.17.tgz
> >   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
> >   Dload  Upload   Total   Spent    Left  Speed
> >   100 1216k  100 1216k    0     0   855k      0  0:00:01  0:00:01 --:--:--  855k
> >   . /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1
> >   [ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17
> >   touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepared621e8a302ca41c8c98a0a93cfc5b16c0_6664517399ebbbc92a37c5bb081b5c53
> >   (cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
> >   touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured
> >   cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-EBuSm55VKe npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-EBuSm55VKe npm install --production --global-style --no-save --omit=dev --no-package-lock
> >   TypeError: Class extends value undefined is not a constructor or null
> >   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
> >   at Module._compile (node:internal/modules/cjs/loader:1103:14)
> >   at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
> >   at Module.load (node:internal/modules/cjs/loader:981:32)
> >   at Function.Module._load (node:internal/modules/cjs/loader:822:12)
> >   at Module.require (node:internal/modules/cjs/loader:1005:19)
> >   at require (node:internal/modules/cjs/helpers:102:18)
> >   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
> >   at Module._compile (node:internal/modules/cjs/loader:1103:14)
> >   at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
> >   at Module.load (node:internal/modules/cjs/loader:981:32)
> >   at Function.Module._load (node:internal/modules/cjs/loader:822:12)
> >   at Module.require (node:internal/modules/cjs/loader:1005:19)
> >   at require (node:internal/modules/cjs/helpers:102:18)
> >   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
> >   at Module._compile (node:internal/modules/cjs/loader:1103:14)
[... cut ...]
make[2]: *** [Makefile:71: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1
make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
time: package/feeds/packages/node-yarn/host-compile#0.48#0.34#0.73
    ERROR: package/feeds/packages/node-yarn [host] failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: package/feeds/packages/node-yarn/host/compile] Error 2

breakings, 2022-03-08:
> > > > > > > 看吧 等会又是一堆翻车的 ，然后又是一堆issue哈哈
> > > > > > 
> > > > > > 
> > > > > > 看他这个情况 node的host-build是过了，只是yarn的host-build没过，用到node的host-build的包本来就才那么几个。用到node-yarn/host的有且仅有一个adguardhome。
> > > > > 
> > > > > 
> > > > > 慢慢修吧，升级后可以测试一下14能兼容不，到时候提一个pr
> > > > 
> > > > 
> > > > 等一下那位老兄的结果，哈哈
> > > 
> > > 
> > > 替换了一下，还是错误 make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/python/python3' time: package/feeds/packages/python3/host-compile#0.23#1.88#2.77 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.node-v16.14.0.tar.xz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "node-v16.14.0.tar.xz" "05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623" "" "https://nodejs.org/dist/v16.14.0" ' make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node' time: package/feeds/packages/node/host-compile#0.20#8.61#11.49 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/" '
> > > 
> > > * curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/yarn/-/yarn-1.22.17.tgz
> > >   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
> > >   Dload  Upload   Total   Spent    Left  Speed
> > >   100 1216k  100 1216k    0     0   855k      0  0:00:01  0:00:01 --:--:--  855k
> > >   . /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1
> > >   [ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17
> > >   touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepared621e8a302ca41c8c98a0a93cfc5b16c0_6664517399ebbbc92a37c5bb081b5c53
> > >   (cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
> > >   touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured
> > >   cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-EBuSm55VKe npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-EBuSm55VKe npm install --production --global-style --no-save --omit=dev --no-package-lock
> > >   TypeError: Class extends value undefined is not a constructor or null
> > >   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
> > >   at Module._compile (node:internal/modules/cjs/loader:1103:14)
> > >   at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
> > >   at Module.load (node:internal/modules/cjs/loader:981:32)
> > >   at Function.Module._load (node:internal/modules/cjs/loader:822:12)
> > >   at Module.require (node:internal/modules/cjs/loader:1005:19)
> > >   at require (node:internal/modules/cjs/helpers:102:18)
> > >   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
> > >   at Module._compile (node:internal/modules/cjs/loader:1103:14)
> > >   at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
> > >   at Module.load (node:internal/modules/cjs/loader:981:32)
> > >   at Function.Module._load (node:internal/modules/cjs/loader:822:12)
> > >   at Module.require (node:internal/modules/cjs/loader:1005:19)
> > >   at require (node:internal/modules/cjs/helpers:102:18)
> > >   at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
[... cut ...]
> > make[2]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
> > time: package/feeds/packages/node-yarn/host-compile#0.48#0.34#0.73
> > ERROR: package/feeds/packages/node-yarn [host] failed to build.
> > make[1]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
> > make[1]: Leaving directory '/home/hzybb/lede'
> > make: *** [/home/hzybb/lede/include/toplevel.mk:230: package/feeds/packages/node-yarn/host/compile] Error 2

感觉像是 npm_modules的问题 这方面我更完全不懂 如果只是想用新版node就选v16编译 要编译yarn的话建议选回v14

WYC-2020, 2022-03-08:
你怕不是用的l大的吧， l大的这个库根本就不会进行安装编译，我亲测即使编译node16 然后选择编译adg，这个库一样不会进行编译的

netlovehf, 2022-03-08:
> 你怕不是用的l大的吧， l大的这个库根本就不会进行安装编译，我亲测即使编译node16 然后选择编译adg，这个库一样不会进行编译的
================================================================
~/lede$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 10 (delta 6), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), 2.36 KiB | 89.00 KiB/s, done.
From https://github.com/coolsnowwolf/packages

netlovehf, 2022-03-08:
9053c0ce..a1389f03  master     -> origin/master
Updating 9053c0ce..a1389f03
Fast-forward
 lang/node-yarn/Makefile | 64 ++++++++++++++++++++++++++++++++++++++++++++++++++++++----------
 lang/node/Makefile      |  2 +-
 2 files changed, 55 insertions(+), 11 deletions(-)

> 你怕不是用的l大的吧， l大的这个库根本就不会进行安装编译，我亲测即使编译node16 然后选择编译adg，这个库一样不会进行编译的
> ================================================================
> ~/lede$ ./scripts/feeds update -a
> Updating feed 'packages' from [...]
 9053c0ce..a1389f03  master     -> origin/master
Updating 9053c0ce..a1389f03
Fast-forward
 lang/node-yarn/Makefile | 64 ++++++++++++++++++++++++++++++++++++++++++++++++++++++----------
 lang/node/Makefile      |  2 +-
 2 files changed, 55 insertions(+), 11 deletions(-)
提交了更新，看看否成功

WYC-2020, 2022-03-08:
切换默认node14 然后加入了编译node-yarn 这个，你可以去测试，原来的版本是不会编译到这个的，我不知道你是怎么操作出来这个 哈哈我14 16 都测试过

netlovehf, 2022-03-08:
> 切换默认node14 然后加入了编译node-yarn 这个，你可以去测试，原来的版本是不会编译到这个的，我不知道你是怎么操作出来这个 哈哈我14 16 都测试过
==============
我为什么有16版本

# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=node

ifeq ($(CONFIG_NODEJS_16),y)
PKG_VERSION:=v16.14.0
PKG_RELEASE:=1
PKG_HASH:=05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623
PATCH_DIR:=./patches/v16.x
else
ifeq ($(CONFIG_NODEJS_10),y)
PKG_VERSION:=v10.24.1
PKG_RELEASE:=3
PKG_HASH:=d72fc2c244603b4668da94081dc4d6067d467fdfa026e06a274012f16600480c
PATCH_DIR:=./patches/v10.x
else
ifeq ($(CONFIG_NODEJS_14),y)
PKG_VERSION:=v14.18.3
PKG_RELEASE:=1
PKG_HASH:=783ac443cd343dd6c68d2abcf7e59e7b978a6a428f6a6025f9b84918b769d608
PATCH_DIR:=./patches/v14.x
else
PKG_VERSION:=v12.22.1
PKG_RELEASE:=4
PKG_HASH:=dd650df7773a6ed3e390320ba51ef33cba6499f0e9397709ea3d1debdcbcb989
PATCH_DIR:=./patches/v12.x
endif
endif
endif

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://nodejs.org/dist/$(PKG_VERSION)

PKG_MAINTAINER:=Hirokazu MORIKAWA <morikw2@gmail.com>
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE
PKG_CPE_ID:=cpe:/a:nodejs:node.js

HOST_BUILD_DEPENDS:=python3/host
HOST_BUILD_PARALLEL:=1

PKG_BUILD_DEPENDS:=python3/host
PKG_BUILD_PARALLEL:=1
PKG_INSTALL:=1
PKG_USE_MIPS16:=0
PKG_ASLR_PIE:=0

include $(INCLUDE_DIR)/host-build.mk
include $(INCLUDE_DIR)/package.mk

define Package/node
  SECTION:=lang
  CATEGORY:=Languages
  SUBMENU:=Node.js
  TITLE:=Node.js is a platform built on Chrome's JavaScript runtime
  URL:=https://nodejs.org/
  DEPENDS:=@(HAS_FPU||KERNEL_MIPS_FPU_EMULATOR||SOFT_FLOAT) @!arc @!armeb @!powerpc \
	   +libstdcpp +libopenssl +zlib +libnghttp2 \
	   +libatomic +NODEJS_ICU_SYSTEM:icu +NODEJS_ICU_SYSTEM:icu-full-data
endef

define Package/node/description
[... 122 lines cut ...]
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/node $(1)/usr/bin/
endef

define Package/node-npm/install
$(INSTALL_DIR) $(1)/usr/lib/node_modules/npm
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/npm/{package.json,LICENSE} \
		$(1)/usr/lib/node_modules/npm/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/npm/{AUTHORS,*.md} \
		$(1)/usr/lib/node_modules/npm/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/npm/{node_modules,bin,lib} \
		$(1)/usr/lib/node_modules/npm/
	$(INSTALL_DIR) $(1)/usr/bin
	$(LN) ../lib/node_modules/npm/bin/npm-cli.js $(1)/usr/bin/npm
	$(LN) ../lib/node_modules/npm/bin/npx-cli.js $(1)/usr/bin/npx
endef

define Package/node-npm/postrm
#!/bin/sh
rm -rf /usr/lib/node_modules/npm || true
endef

$(eval $(call HostBuild))
$(eval $(call BuildPackage,node))
$(eval $(call BuildPackage,node-npm))

WYC-2020, 2022-03-08:
> > 切换默认node14 然后加入了编译node-yarn 这个，你可以去测试，原来的版本是不会编译到这个的，我不知道你是怎么操作出来这个 哈哈我14 16 都测试过
> > ==============
> > 我为什么有16版本
> 
> # This is free software, licensed under the GNU General Public License v2.
> # See /LICENSE for more information.
> include $(TOPDIR)/rules.mk
> 
> PKG_NAME:=node
> 
> ifeq ($(CONFIG_NODEJS_16),y) PKG_VERSION:=v16.14.0 PKG_RELEASE:=1 PKG_HASH:=05eb64193e391fa8a2c159c0f60c171824715165f80c67fcab9dbc944e30c623 PATCH_DIR:=./patches/v16.x else ifeq ($(CONFIG_NODEJS_10),y) PKG_VERSION:=v10.24.1 PKG_RELEASE:=3 PKG_HASH:=d72fc2c244603b4668da94081dc4d6067d467fdfa026e06a274012f16600480c PATCH_DIR:=./patches/v10.x else ifeq ($(CONFIG_NODEJS_14),y) PKG_VERSION:=v14.18.3 PKG_RELEASE:=1 PKG_HASH:=783ac443cd343dd6c68d2abcf7e59e7b978a6a428f6a6025f9b84918b769d608 PATCH_DIR:=./patches/v14.x else PKG_VERSION:=v12.22.1 PKG_RELEASE:=4 PKG_HASH:=dd650df7773a6ed3e390320ba51ef33cba6499f0e9397709ea3d1debdcbcb989 PATCH_DIR:=./patches/v12.x endif endif endif
> 
> PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz PKG_SOURCE_URL:=https://nodejs.org/dist/$(PKG_VERSION)
> 
> PKG_MAINTAINER:=Hirokazu MORIKAWA [morikw2@gmail.com](mailto:morikw2@gmail.com) PKG_LICENSE:=MIT PKG_LICENSE_FILES:=LICENSE PKG_CPE_ID:=cpe:/a:nodejs:node.js
> 
> HOST_BUILD_DEPENDS:=python3/host HOST_BUILD_PARALLEL:=1
> 
> PKG_BUILD_DEPENDS:=python3/host PKG_BUILD_PARALLEL:=1 PKG_INSTALL:=1 PKG_USE_MIPS16:=0 PKG_ASLR_PIE:=0
> 
> include $(INCLUDE_DIR)/host-build.mk include $(INCLUDE_DIR)/package.mk
> 
> define Package/node SECTION:=lang CATEGORY:=Languages SUBMENU:=Node.js TITLE:=Node.js is a platform built on Chrome's JavaScript runtime URL:=https://nodejs.org/ DEPENDS:=@(HAS_FPU||KERNEL_MIPS_FPU_EMULATOR||SOFT_FLOAT) @!arc @!armeb @!powerpc +libstdcpp +libopenssl +zlib +libnghttp2 +libatomic +NODEJS_ICU_SYSTEM:icu +NODEJS_ICU_SYSTEM:icu-full-data endef
> 
> define Package/node/description Node.js® is a JavaScript runtime built on Chrome's V8 JavaScript engine. Node.js uses an event-driven, non-blocking I/O model that makes it lightweight and efficient. Node.js' package ecosystem, npm, is the largest ecosystem of open source libraries in the world.
> 
> *** The following preparations must be made on the host side. *** 1. gcc 8.3 or higher is required. 2. To build a 32-bit target, gcc-multilib, g++-multilib are required. 3. Requires libatomic package. (If necessary, install the 32-bit library at the same time.) ex) sudo apt-get install gcc-multilib g++-multilib endef
> 
> define Package/node-npm SECTION:=lang CATEGORY:=Languages SUBMENU:=Node.js TITLE:=NPM stands for Node Package Manager URL:=https://www.npmjs.com/ DEPENDS:=+node endef
> 
> define Package/node-npm/description NPM is the package manager for NodeJS endef
> 
> define Package/node/config menu "Configuration" depends on PACKAGE_node
> 
> ```
> choice
> 	prompt "Version Selection"
> 	default NODEJS_12 if mips||mipsel||mips64||mips64el
> 	default NODEJS_14
> 	help
> 	 Select node.js version.
> 
> 	config NODEJS_10
> 		bool "10.x End-of-life : NO LONGER SUPPORTED"
> 
> 	config NODEJS_12
> 		bool "12.x Maintenance LTS"
> 
> 	config NODEJS_14
> 		depends on !(mips||mipsel||mips64||mips64el)
> 		bool "14.x Maintenance LTS"
> 
> 	config NODEJS_16
> 		depends on !(mips||mipsel||mips64||mips64el)
> 		bool "16.x Active LTS"
> endchoice
> 
> choice
> 	prompt "i18n features"
> 	default NODEJS_ICU_NONE
> 	help
> 	 Select i18n features
> 
> 	config NODEJS_ICU_NONE
> 		bool "Disable"
> 
> 	config NODEJS_ICU_SMALL
> 		depends on !BIG_ENDIAN
> 		bool "small-icu"
> 
[... 12 lines cut ...]
> ifeq ($(CONFIG_NODEJS_10),y) TARGET_LDFLAGS+=-latomic endif
> 
> ifneq ($(CONFIG_ARCH_64BIT),y) FORCE_32BIT:=-m32 endif
> 
> MAKE_VARS+= DESTCPU=$(NODEJS_CPU) NO_LOAD='cctest.target.mk embedtest.target.mk node_mksnapshot.target.mk' LD_LIBRARY_PATH=$(STAGING_DIR_HOSTPKG)/share/icu/current/lib
> 
> HOST_MAKE_VARS+=NO_LOAD='cctest.target.mk embedtest.target.mk'
> 
> CONFIGURE_VARS:= CC="$(TARGET_CC) $(TARGET_OPTIMIZATION)" CXX="$(TARGET_CXX) $(TARGET_OPTIMIZATION)" CC_host="$(HOSTCC) $(FORCE_32BIT)" CXX_host="$(HOSTCXX) $(FORCE_32BIT)"
> 
> CONFIGURE_ARGS:= --dest-cpu=$(NODEJS_CPU) --dest-os=linux --cross-compiling --shared-zlib --shared-openssl --shared-nghttp2 --with-intl=$(if $(CONFIG_NODEJS_ICU_SMALL),small-icu,$(if $(CONFIG_NODEJS_ICU_SYSTEM),system-icu,none)) $(if $(findstring mips,$(NODEJS_CPU)), $(if $(CONFIG_SOFT_FLOAT),--with-mips-float-abi=soft)) $(if $(findstring +neon,$(CONFIG_CPU_TYPE)),--with-arm-fpu=neon) $(if $(findstring +vfp",$(CONFIG_CPU_TYPE)),--with-arm-fpu=vfp) $(if $(findstring +vfpv3",$(CONFIG_CPU_TYPE)),--with-arm-fpu=vfpv3-d16) $(if $(findstring +vfpv4",$(CONFIG_CPU_TYPE)),--with-arm-fpu=vfpv3) --prefix=/usr
> 
> HOST_CONFIGURE_VARS:= HOST_CONFIGURE_ARGS:= --dest-os=$(if $(findstring Darwin,$(HOST_OS)),mac,linux) --with-intl=small-icu --prefix=$(STAGING_DIR_HOSTPKG)
> 
> define Build/InstallDev $(INSTALL_DIR) $(1)/usr/include $(CP) $(PKG_INSTALL_DIR)/usr/include/* $(1)/usr/include/ endef
> 
> define Package/node/install $(INSTALL_DIR) $(1)/usr/bin $(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/node $(1)/usr/bin/ endef
> 
> define Package/node-npm/install $(INSTALL_DIR) $(1)/usr/lib/node_modules/npm $(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/npm/{package.json,LICENSE} $(1)/usr/lib/node_modules/npm/ $(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/npm/{AUTHORS,*.md} $(1)/usr/lib/node_modules/npm/ $(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/npm/{node_modules,bin,lib} $(1)/usr/lib/node_modules/npm/ $(INSTALL_DIR) $(1)/usr/bin $(LN) ../lib/node_modules/npm/bin/npm-cli.js $(1)/usr/bin/npm $(LN) ../lib/node_modules/npm/bin/npx-cli.js $(1)/usr/bin/npx endef
> 
> define Package/node-npm/postrm #!/bin/sh rm -rf /usr/lib/node_modules/npm || true endef
> 
> $(eval $(call HostBuild)) $(eval $(call BuildPackage,node)) $(eval $(call BuildPackage,node-npm))

原来默认内核是16 你config没删除重新配自然是16了，自己改下就行了

netlovehf, 2022-03-08:
#拉取最新，还是不行
make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0'
mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp
touch /home/hzybb/lede/build_dir/hostpkg/node-v16.14.0/.built
touch /home/hzybb/lede/staging_dir/hostpkg/stamp/.node_installed
make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#5197.23#310.51#5514.01
make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c '          /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/"    '
. /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1
[ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepareddde67abb946bbb04ee50db61d61a453d_6664517399ebbbc92a37c5bb081b5c53
(cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi )
touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured
cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-iQiLTEL0tm npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-iQiLTEL0tm npm install --production --global-style --no-save --omit=dev --no-package-lock
TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
make[3]: *** [Makefile:115: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1
make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn'
time: package/feeds/packages/node-yarn/host-compile#0.35#0.38#0.96
    ERROR: package/feeds/packages/node-yarn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1
make[2]: Leaving directory '/home/hzybb/lede'
make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

breakings, 2022-03-08:
> ============================= 拉取最新，还是不行
> 
> make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0' mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp touch /home/hzybb/lede/build_dir/hostpkg/node-v16.14.0/.built touch /home/hzybb/lede/staging_dir/hostpkg/stamp/.node_installed make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node' time: package/feeds/packages/node/host-compile#5197.23#310.51#5514.01 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/" ' . /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1 [ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepareddde67abb946bbb04ee50db61d61a453d_6664517399ebbbc92a37c5bb081b5c53 (cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash" bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi ) touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-iQiLTEL0tm npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-iQiLTEL0tm npm install --production --global-style --no-save --omit=dev --no-package-lock TypeError: Class extends value undefined is not a constructor or null at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44) at Module._compile (node:internal/modules/cjs/loader:1103:14) at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10) at Module.load (node:internal/modules/cjs/loader:981:32) at Function.Module._load (node:internal/modules/cjs/loader:822:12) at Module.require (node:internal/modules/cjs/loader:1005:19) at require (node:internal/modules/cjs/helpers:102:18) at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33) at Module._compile (node:internal/modules/cjs/loader:1103:14) at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10) at Module.load (node:internal/modules/cjs/loader:981:32) at Function.Module._load (node:internal/modules/cjs/loader:822:12) at Module.require (node:internal/modules/cjs/loader:1005:19) at require (node:internal/modules/cjs/helpers:102:18) at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25) at Module._compile (node:internal/modules/cjs/loader:1103:14) make[3]: *** [Makefile:115: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1 make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' time: package/feeds/packages/node-yarn/host-compile#0.35#0.38#0.96 ERROR: package/feeds/packages/node-yarn [host] failed to build. make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1 make[2]: Leaving directory '/home/hzybb/lede' make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/hzybb/lede' make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

可能yarn的 hostpkg编译不能用v16 或者哪个地方还需要修正

WYC-2020, 2022-03-08:
> > ============================= 拉取最新，还是不行
> > make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0' mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp touch /home/hzybb/lede/build_dir/hostpkg/node-v16.14.0/.built touch /home/hzybb/lede/staging_dir/hostpkg/stamp/.node_installed make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node' time: package/feeds/packages/node/host-compile#5197.23#310.51#5514.01 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/" ' . /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1 [ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepareddde67abb946bbb04ee50db61d61a453d_6664517399ebbbc92a37c5bb081b5c53 (cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash" bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi ) touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-iQiLTEL0tm npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-iQiLTEL0tm npm install --production --global-style --no-save --omit=dev --no-package-lock TypeError: Class extends value undefined is not a constructor or null at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44) at Module._compile (node:internal/modules/cjs/loader:1103:14) at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10) at Module.load (node:internal/modules/cjs/loader:981:32) at Function.Module._load (node:internal/modules/cjs/loader:822:12) at Module.require (node:internal/modules/cjs/loader:1005:19) at require (node:internal/modules/cjs/helpers:102:18) at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33) at Module._compile (node:internal/modules/cjs/loader:1103:14) at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10) at Module.load (node:internal/modules/cjs/loader:981:32) at Function.Module._load (node:internal/modules/cjs/loader:822:12) at Module.require (node:internal/modules/cjs/loader:1005:19) at require (node:internal/modules/cjs/helpers:102:18) at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25) at Module._compile (node:internal/modules/cjs/loader:1103:14) make[3]: *** [Makefile:115: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1 make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' time: package/feeds/packages/node-yarn/host-compile#0.35#0.38#0.96 ERROR: package/feeds/packages/node-yarn [host] failed to build. make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1 make[2]: Leaving directory '/home/hzybb/lede' make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/hzybb/lede' make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2
> 
> 可能yarn的 hostpkg编译不能用v16 或者哪个地方还需要修正 搞不懂是不是这几个模块和它不匹配 socks-proxy-agent make-fetch-happen

不要挣扎放弃16吧，不明白这个升的意义有啥用，没必要用最新的，说不定到时候脚本又有问题

Dawneng, 2022-03-08:
> =================== #拉取最新，还是不行
>
> make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0' mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp touch [...]

重重拉取、重新配置、不会选上16啊   现在默认是14！

Dawneng, 2022-03-08:
> =================== #拉取最新，还是不行
>
> make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0' mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp touch [...]

https://github.com/coolsnowwolf/packages/commit/a1389f036eaffbcd04ea6525a18a527a6fd95a8a

breakings, 2022-03-08:
> 重重拉取、重新配置、不会选上16啊 现在默认是14！

手动14

WYC-2020, 2022-03-09:
> 用v14可以编译成功，不知道用v16为什么不成功

别折腾16

WYC-2020, 2022-03-10:
这个我好想修复了
https://github.com/WYC-2020/packages/commits/master

breakings, 2022-03-10:
> 这个我好想修复了 [WYC-2020/packages@eb4ee43](https://github.com/WYC-2020/packages/commit/eb4ee43a4d7bcbdab3696f76b0089d8666d96c06)

不过我测试 node-yarn/host/compile还是不行，感觉为了一个adguardhome完全不值得，况且官方github上面所有平台都有现编译好的二进制包了。

WYC-2020, 2022-03-10:
> > 这个我好想修复了 [WYC-2020/packages@eb4ee43](https://github.com/WYC-2020/packages/commit/eb4ee43a4d7bcbdab3696f76b0089d8666d96c06)
> 
> 不过我测试 node-yarn/host/compile还是不行，感觉为了一个adguardhome完全不值得，况且官方github上面所有平台都有现编译好的二进制包了。

那暂时不用管了，反正我这边编译没问题，无所谓的，你们用14就行

coolsnowwolf, 2026-09-05:
node-yarn 日志只给出末尾 `Error 1`，没有实际 npm/yarn 失败输出和主机环境，无法确认包源码问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf