coolsnowwolf/lede #9028: clocksource报错

moonerick, 2022-03-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：版本为5.15.26，新编译使用，出现以下日志，每几分出现一条，不知道应该如何处理

### (2) 路由器型号和固件版本

A：QEMU Standard PC (Q35 + ICH9 - Intel(R) Pentium(R) Silver J5005 CPU @ 1.50GHz : 4 Core 4 Thread
unraid下虚拟机使用，cpu为intel J5005

### (3) 详细日志
Tue Mar  8 13:19:34 2022 kern.info kernel: [14669.048261] clocksource: timekeeping watchdog on CPU1: acpi_pm wd-wd read-back delay of 144431ns
Tue Mar  8 13:19:34 2022 kern.info kernel: [14669.056895] clocksource: wd-tsc-wd read-back delay of 124876ns, clock-skew test skipped!
Tue Mar  8 13:25:18 2022 kern.info kernel: [15013.044543] clocksource: timekeeping watchdog on CPU1: acpi_pm wd-wd read-back delay of 371834ns
Tue Mar  8 13:25:18 2022 kern.info kernel: [15013.052421] clocksource: wd-tsc-wd read-back delay of 114539ns, clock-skew test skipped!
Tue Mar  8 13:26:11 2022 kern.info kernel: [15066.040401] clocksource: timekeeping watchdog on CPU3: acpi_pm wd-wd read-back delay of 247796ns
Tue Mar  8 13:26:11 2022 kern.info kernel: [15066.047454] clocksource: wd-tsc-wd read-back delay of 119568ns, clock-skew test skipped!
Tue Mar  8 13:31:37 2022 kern.info kernel: [15392.532233] clocksource: timekeeping watchdog on CPU0: acpi_pm wd-wd read-back delay of 81295ns
Tue Mar  8 13:31:37 2022 kern.info kernel: [15392.539617] clocksource: wd-tsc-wd read-back delay of 549231ns, clock-skew test skipped!
Tue Mar  8 13:32:56 2022 kern.info kernel: [15471.543954] clocksource: timekeeping watchdog on CPU2: acpi_pm wd-wd read-back delay of 88558ns
Tue Mar  8 13:32:56 2022 kern.info kernel: [15471.550963] clocksource: wd-tsc-wd read-back delay of 156165ns, clock-skew test skipped!

coolsnowwolf, 2022-03-08:
虚拟机类型选错了，时钟类型选 HPET，CPU 选原生

moonerick, 2022-03-09:
> 虚拟机类型选错了，时钟类型选 HPET，CPU 选原生

谢谢大神，我研究以下，unraid默认虚拟机

closed 2022-03-09 by moonerick

moonerick, 2022-03-09:
> 虚拟机类型选错了，时钟类型选 HPET，CPU 选原生
处理好了，编译的时候如何配置默认值还得研究一下，再次感谢
bash -c 'echo hpet > /sys/devices/system/clocksource/clocksource0/current_clocksource'