coolsnowwolf/lede #9029: make problem lede

AugustLiu, 2022-03-08:
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
cp -fpR /home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips /home/august/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
/home/august/lede/build_dir/target-mipsel_24kc_musl/root-ramips/lib/functions.sh: line 242: usr/bin/basename: cannot execute binary file: Exec format error
[... cut ...]
find: The relative path 'usr/bin' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/august/lede'
make[1]: *** [package/Makefile:111: /home/august/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/august/lede'
make: *** [/home/august/lede/include/toplevel.mk:230：world] 错误 2

coolsnowwolf, 2022-03-08:
find: The relative path 'usr/bin' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find. Please remove that entry from $PATH

AugustLiu, 2022-03-08:
> find: The relative path 'usr/bin' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find. Please remove that entry from $PATH
thank you ! I know it!

closed 2022-03-18 by AmadeusGhost