coolsnowwolf/lede #9030: 新固件在群辉vmm中安装后不能打开管理页面

itsmine, 2022-03-08:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
我试过许多最近比较新的固件，也包括我自己编译的固件。发现使用群辉的vmm虚拟机安装openwrt，会导致luci管理页面不能打开。网络没有问题，ip地址，ip段，ping，ssh都是可用的。我试用了老固件，发现是可以打开的。我又用新固件在windows下的vm16虚拟机中安装发现是可以正常打开管理页的。我想问下这是什么原因，是因为内核吗？
### (1) 具体问题

A：最近半个月编译的固件在群辉vmm中安装后，不能正常打开管理页。

### (2) 路由器型号和固件版本

A：群辉vmm
固件版本例如：
Linux version 5.15.26 (runner@fv-az152-360) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r0-18abe63) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Mon Mar 7 04:24:40 2022
BusyBox v1.35.0
OpenWrt SNAPSHOT, r0-18abe63
### (3) 详细日志

A：无日志

NagaseKouichi, 2022-03-09:
感觉和我之前在QNAP上用LXD运行openwrt出现的问题一样，最后关掉SYN-flood 防御解决了，你可以试试。SSH运行：

uci rename network.wan=lan
uci rename network.wan6=lan6
uci set "firewall.@defaults[0].syn_flood='0'"
uci commit
/etc/init.d/network restart
/etc/init.d/firewall restart

itsmine, 2022-03-10:
感谢你的帮助，我测试了两个固件，还是老样子

NagaseKouichi, 2022-03-10:
关闭openwrt的防火墙试试

itsmine, 2022-03-11:
关了后更惨，显示拒绝连接。

WYC-2020, 2022-03-18:
luci 管理页面打不开 和内核根本没有任何关系，自己看下http服务这块

itsmine, 2022-03-20:
我放弃了，直接在esxi里面虚拟没问题

closed 2022-03-20 by itsmine