coolsnowwolf/lede #9033: 2022/3/8 xr_usb_serial_common在线编译报错

Cptmacmillan2022007, 2022-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

2022-03-08T13:37:22.8901417Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/xr_usb_serial_common'
2022-03-08T13:37:23.0396992Z mkdir -p /workdir/openwrt/dl
2022-03-08T13:37:23.0435287Z SHELL= flock /workdir/openwrt/tmp/.usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf.tar.xz" "2cf23fec2625480bb4a630f96cd172219d401bd3cf90943394504ea0ba814faf" "" || (  /workdir/openwrt/scripts/dl_github_archive.py --dl-dir="/workdir/openwrt/dl" --url="https://github.com/kasbert/epsolar-tracer" --version="b8dad8cf15de160afbd9989f880dc74b921a857b" --subdir="usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf" --source="usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf.tar.xz" --hash="2cf23fec2625480bb4a630f96cd172219d401bd3cf90943394504ea0ba814faf" || ( 	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf && [ \! -d usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf ] && git clone  https://github.com/kasbert/epsolar-tracer usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf && (cd usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf && git checkout b8dad8cf15de160afbd9989f880dc74b921a857b && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf && git log -1 --format='\''@%ct'\''` && rm -rf usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf | 	xz -zc -7e > /workdir/openwrt/tmp/dl/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf.tar.xz && mv /workdir/openwrt/tmp/dl/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf.tar.xz /workdir/openwrt/dl/ && rm -rf usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf; );  )    '
2022-03-08T13:37:23.0929399Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/.built
2022-03-08T13:37:23.0959165Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/.built_check
2022-03-08T13:37:23.0982014Z make  -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.26" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a" modules
2022-03-08T13:37:23.1025271Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.26'
2022-03-08T13:37:23.4833469Z   CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.o
2022-03-08T13:37:23.8251035Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function 'xr_usb_serial_ctrl_irq':
2022-03-08T13:37:23.8252931Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:261:21: warning: unused variable 'tty' [-Wunused-variable]
2022-03-08T13:37:23.8253574Z   struct tty_struct *tty;
2022-03-08T13:37:23.8303671Z                      ^~~
2022-03-08T13:37:23.8304419Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function 'xr_usb_serial_process_read_urb':
2022-03-08T13:37:23.8305314Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:393:24: warning: unused variable 'tty' [-Wunused-variable]
2022-03-08T13:37:23.8305765Z      struct tty_struct *tty;
2022-03-08T13:37:23.8305986Z                         ^~~
2022-03-08T13:37:23.8306590Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function 'xr_usb_serial_softint':
2022-03-08T13:37:23.8307453Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:468:24: warning: unused variable 'tty' [-Wunused-variable]
2022-03-08T13:37:23.8307888Z      struct tty_struct *tty;
2022-03-08T13:37:23.8308285Z                         ^~~
2022-03-08T13:37:23.8308890Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function 'xr_usb_serial_tty_ioctl':
2022-03-08T13:37:23.8309804Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:878:17: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
2022-03-08T13:37:23.8310345Z                  if (get_user(val, (int __user *)(arg + 2 * sizeof(int))))
2022-03-08T13:37:23.8310584Z                  ^~
2022-03-08T13:37:23.8311306Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:881:4: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
2022-03-08T13:37:23.8311821Z     if (channel == -1)
2022-03-08T13:37:23.8312025Z     ^~
2022-03-08T13:37:23.8350487Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function 'xr_usb_serial_reset_resume':
2022-03-08T13:37:23.8351408Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:1625:24: warning: unused variable 'tty' [-Wunused-variable]
2022-03-08T13:37:23.8351869Z      struct tty_struct *tty;
2022-03-08T13:37:23.8352088Z                         ^~~
2022-03-08T13:37:23.8352668Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c: At top level:
2022-03-08T13:37:23.8353739Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:1690:17: error: initialization of 'unsigned int (*)(struct tty_struct *)' from incompatible pointer type 'int (*)(struct tty_struct *)' [-Werror=incompatible-pointer-types]
2022-03-08T13:37:23.8354421Z   .write_room =  xr_usb_serial_tty_write_room,
2022-03-08T13:37:23.8354669Z                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
2022-03-08T13:37:23.8355365Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:1690:17: note: (near initialization for 'xr_usb_serial_ops.write_room')
2022-03-08T13:37:23.8356675Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:1694:21: error: initialization of 'unsigned int (*)(struct tty_struct *)' from incompatible pointer type 'int (*)(struct tty_struct *)' [-Werror=incompatible-pointer-types]
2022-03-08T13:37:23.8357296Z   .chars_in_buffer = xr_usb_serial_tty_chars_in_buffer,
2022-03-08T13:37:23.8357547Z                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2022-03-08T13:37:23.8358248Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/usb-serial-xr_usb_serial_common-2017-08-01-b8dad8cf/xr_usb_serial_common-1a/xr_usb_serial_common.c:1694:21: note: (near initialization for 'xr_usb_serial_ops.chars_in_buffer')
[... cut ...]
2022-03-08T13:37:24.2682360Z make[2]: *** [package/Makefile:116: package/feeds/packages/xr_usb_serial_common/compile] Error 1
2022-03-08T13:37:24.2682963Z make[2]: Leaving directory '/workdir/openwrt'
2022-03-08T13:37:24.2697946Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2022-03-08T13:37:24.2698483Z make[1]: Leaving directory '/workdir/openwrt'
2022-03-08T13:37:24.2704886Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2022-03-08T13:37:24.2763164Z ##[error]Process completed with exit code 2.

WYC-2020, 2022-03-09:
并不是所有驱动都适配了5.15的，还没来及的修复

WYC-2020, 2022-03-09:
https://github.com/WYC-2020/lede/commit/5e906a1fd70a90bf39ed45612dde5d64b18e7499
https://github.com/WYC-2020/packages/commit/92384aa15259a69ae6f0b8fff06820a04f687f1f
@Cptmacmillan2022007

closed 2022-03-12 by coolsnowwolf