coolsnowwolf/lede #9037: 3月9号编译报错bash: line 6: 8: Bad file descriptor

wokilp, 2022-03-09:
Generating index for package ./luci-lib-nixio_git-22.067.25891-c3b1794-1_mipsel_24kc.ipk
Generating index for package ./luci-mod-admin-full_git-22.067.25891-c3b1794-1_mipsel_24kc.ipk
Generating index for package ./luci-newapi_git-22.067.25891-c3b1794-1_all.ipk
Generating index for package ./luci-proto-ipv6_git-22.067.25891-c3b1794-1_all.ipk
Generating index for package ./luci-proto-ppp_git-22.067.25891-c3b1794-1_all.ipk
Generating index for package ./luci-theme-argon_1.0-2_all.ipk
Generating index for package ./luci_git-22.067.25891-c3b1794-1_all.ipk
Generating index for package ./rpcd-mod-rrdns_20170710_mipsel_24kc.ipk
bash: line 6: 8: Bad file descriptor
Generating index for package ./attr_2.5.1-1_mipsel_24kc.ipk
Generating index for package ./avahi-dbus-daemon_0.8-7_mipsel_24kc.ipk
Generating index for package ./coremark_2022-01-03-b24e397f-1_mipsel_24kc.ipk
Generating index for package ./dawn_2022-01-17-7a726740-1_mipsel_24kc.ipk
Generating index for package ./dbus_1.13.18-1_mipsel_24kc.ipk
Generating index for package ./ddns-scripts_2.7.8-3_all.ipk
Generating index for package ./etherwake_1.09-5_mipsel_24kc.ipk
Generating index for package ./libattr_2.5.1-1_mipsel_24kc.ipk
Generating index for package ./libavahi-client_0.8-7_mipsel_24kc.ipk
Generating index for package ./libavahi-dbus-support_0.8-7_mipsel_24kc.ipk
Generating index for package ./libdaemon_0.14-5_mipsel_24kc.ipk
Generating index for package ./libdbus_1.13.18-1_mipsel_24kc.ipk
Generating index for package ./libexpat_2.4.6-2_mipsel_24kc.ipk
Generating index for package ./libgcrypt_1.9.3-1_mipsel_24kc.ipk
Generating index for package ./libgnutls_3.7.2-1_mipsel_24kc.ipk
Generating index for package ./libgpg-error_1.43-1_mipsel_24kc.ipk
Generating index for package ./libtasn1_4.16.0-1_mipsel_24kc.ipk
Generating index for package ./libtirpc_1.3.1-1_mipsel_24kc.ipk
Generating index for package ./liburing_2.0-1_mipsel_24kc.ipk
Generating index for package ./maccalc_1_mipsel_24kc.ipk
Generating index for package ./miniupnpd_2.0.20170421-3_mipsel_24kc.ipk
Generating index for package ./mwan3_2.8.8-2_all.ipk
Generating index for package ./nlbwmon_2021-09-01-d82c910c-1_mipsel_24kc.ipk
Generating index for package ./p910nd_0.97-9_mipsel_24kc.ipk
Generating index for package ./pdnsd-alt_1.2.9b-par-3_mipsel_24kc.ipk
Generating index for package ./samba4-libs_4.14.12-1_mipsel_24kc.ipk
Generating index for package ./samba4-server_4.14.12-1_mipsel_24kc.ipk
Generating index for package ./vlmcsd_svn1113-3_mipsel_24kc.ipk
Generating index for package ./wget-ssl_1.21.2-1_mipsel_24kc.ipk
Generating index for package ./wsdd2_2021-10-22-9831daf2-1_mipsel_24kc.ipk
Generating index for package ./batctl-default_2022.0-1_mipsel_24kc.ipk
Signing package index...
make[2]: Leaving directory '/home/test/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/test/lede'
make[2]: Nothing to be done for 'json_overview_image_info'.
make[2]: Leaving directory '/home/test/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/test/lede'
make[2]: Leaving directory '/home/test/lede'
make[1]: Leaving directory '/home/test/lede'
r3编译完成
全部编译完成
  adding: openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-phicomm_k2p.manifest (deflated 69%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin (deflated 1%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit.manifest (deflated 71%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-squashfs-kernel1.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-squashfs-rootfs0.bin (deflated 13%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-squashfs-sysupgrade.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-ac2100.manifest (deflated 69%)
  adding: openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-kernel1.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-rootfs0.bin (deflated 13%)
  adding: openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-sysupgrade.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100.manifest (deflated 69%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-v2-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-v2-squashfs-sysupgrade.bin (deflated 1%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-v2.manifest (deflated 71%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-squashfs-kernel1.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-squashfs-rootfs0.bin (deflated 9%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-squashfs-sysupgrade.bin (deflated 0%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g.manifest (deflated 70%)
  adding: openwrt-ramips-mt7621-xiaomi_mi-router-3g-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-mini-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin (deflated 1%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-mini.manifest (deflated 70%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-r3-initramfs-kernel.bin (deflated 0%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-r3-squashfs-kernel1.bin (deflated 0%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-r3-squashfs-rootfs0.bin (deflated 10%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin (deflated 0%)
  adding: openwrt-ramips-mt7620-xiaomi_miwifi-r3.manifest (deflated 70%)
全部打包完成

wokilp, 2022-03-09:
虽然不影响编译，但是这个是哪里出现问题了呢？辛苦大佬指点一下

WYC-2020, 2022-03-09:
这是一个包的描述信息有问题，这个我以前修过，这个找下是那个包 ，然后重新修改一下他的包描述信息即可，最好不要有特殊字符 中文字符 之类的
@wokilp

wokilp, 2022-03-09:
> 这是一个包的描述信息有问题，这个我以前修过，这个找下是那个包 ，然后重新修改一下他的包描述信息即可，最好不要有特殊字符 中文字符 之类的 @wokilp

具体是在哪个位置？lede/build_dir/hostpkg这个目录下吗？

wokilp, 2022-03-09:
感觉像是这个包的问题，rpcd_2021-07-13-1fa35765-1_mipsel_24kc.ipk，但是没有找到这个包的描述文件

wokilp, 2022-03-09:
Package: qos-scripts
Version: 1.3.1-2
Depends: libc, tc, kmod-sched-core, kmod-sched-connmark, kmod-ifb, iptables, iptables-mod-ipopt, iptables-mod-conntrack-extra
Source: package/network/config/qos-scripts
SourceName: qos-scripts
License: GPL-2.0
Section: utils
SourceDateEpoch: 1623666608
Maintainer: Felix Fietkau <nbd@nbd.name>
Architecture: all
Installed-Size: 6375
Filename: qos-scripts_1.3.1-2_all.ipk
Size: 7302
SHA256sum: cd8c8f37975fea37b965532208931042576d6e6659c33d6fc58887ad51a1e46a
Description:  A set of scripts that abstract QoS configuration into a simple
 configuration file supporting stanzas that specify any number of QoS
 entries.

Package: rpcd
Version: 2021-07-13-1fa35765-1
Depends: libc, libubus20210603, libubox20210516, libuci20130104, libblobmsg-json20210516, libjson-c5
Source: package/system/rpcd
SourceName: rpcd
License: ISC
Section: utils
SourceDateEpoch: 1628735881
Maintainer: Jo-Philipp Wich <jo@mein.io>
Architecture: mipsel_24kc
Installed-Size: 22800
Filename: rpcd_2021-07-13-1fa35765-1_mipsel_24kc.ipk
Size: 23606
SHA256sum: 92480f209259763cd78603dd5360e526051fd036f76ad504db5bd795052fbd73
Description:  This package provides the UBUS RPC backend server to expose various
 functionality to frontend programs via JSON-RPC.

Package: shellsync
Version: 0.2-2
Depends: libc, libpthread, kmod-macvlan
Source: package/network/services/shellsync
SourceName: shellsync
Section: opt
SourceDateEpoch: 1643774461
Architecture: mipsel_24kc
Installed-Size: 2023
Filename: shellsync_0.2-2_mipsel_24kc.ipk
Size: 2783
SHA256sum: b921a8ec5ed3d4ac5695a78daae87765c5764ee979155f0562b7006fd6976fbe
Description:  A tool to sync different shell scripts.
 Based on syncppp patch by morfast.

是这个描述文件吗？感觉里面写的没啥问题

WYC-2020, 2022-03-09:
Makefile

closed 2022-03-10 by AmadeusGhost