coolsnowwolf/lede #9038: 请教一下如何在编译时使用命令修改文档内容？

smdx, 2022-03-09:
请教一下如何在编译时使用命令修改文档内容？在PVE LXC安装OP时首页网卡显示不正常
有大佬给出方法修改ethinfo文件获取网卡信息的代码可以解决这个问题
https://github.com/coolsnowwolf/lede/issues/8821#issuecomment-1030261716
源码package/lean/autocore/files/x86/sbin/ethinfo里获取网卡信息的代码修改一下，
a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | grep -v "@" | grep -v "\.")
修改为
a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | awk -F"@" '{print $1}')

在使用云编译时.sh脚本有相关命令修改ip
# Modify default IP
sed -i 's/192.168.1.1/10.0.0.1/g' package/base-files/files/bin/config_generate

请问如何用命令可以把源码package/lean/autocore/files/x86/sbin/ethinfo内容
a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | grep -v "@" | grep -v "\.")
修改为
a=$(ip address | grep ^[0-9] | awk -F: '{print $2}' | sed "s/ //g" | grep '^[e]' | awk -F"@" '{print $1}')

谢谢

Cathgao, 2022-03-11:
你可以另外写这个文件放到仓库里，然后rm cp

smdx, 2022-03-11:
> 你可以另外写这个文件放到仓库里，然后rm cp

谢谢，rm后测试用 cp 仓库源地址 /root
不成功提示cp: cannot stat 'https://raw.githubusercontent.com/xxx/ethinfo': No such file or directory

Cathgao, 2022-03-11:
你为什么不能放到workflow里面跑呢
```
rm package/lean/autocore/files/x86/sbin/ethinfo
cp ./ethinfo package/lean/autocore/files/x86/sbin/ethinfo
```

Cathgao, 2022-03-11:
我的GDOCK项目里面也有个diy.sh，里面就有不少的文件替换
https://github.com/Cathgao/GDOCK-Cath/blob/master/diy.sh

smdx, 2022-03-11:
> 你为什么不能放到workflow里面跑呢
> 
> ```
> rm package/lean/autocore/files/x86/sbin/ethinfo
> cp ./ethinfo package/lean/autocore/files/x86/sbin/ethinfo
> ```

> 我的GDOCK项目里面也有个diy.sh，里面就有不少的文件替换 https://github.com/Cathgao/GDOCK-Cath/blob/master/diy.sh

非常感谢，马上试试

closed 2022-03-12 by smdx