coolsnowwolf/lede #9039: 编译最新源码出错，patching file arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts 失败

ricksuzade-maker, 2022-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译最新固件出现错误，有一个补丁打不上

### (2) 路由器型号和固件版本

A：rockchip rk3328 5.15.27内核版

### (3) 详细日志

A：

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/007-arm64-dts-rockchip-add-EEPROM-node-for-NanoPi-R4S.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/100-rockchip-use-system-LED-for-OpenWrt.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/103-arm64-rockchip-add-OF-node-for-USB-eth-on-NanoPi-R2S.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/105-rockchip-rock-pi-4.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/Makefile

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/106-arm64-rockchip-add-OF-node-for-pcie-eth-on-NanoPi-R4S.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/107-mmc-core-set-initial-signal-voltage-on-power-off.patch using plaintext: 
patching file drivers/mmc/core/core.c

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/201-rockchip-rk3328-add-i2c0-controller-for-nanopi-r2s.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/202-rockchip-rk3328-Add-support-for-OrangePi-R1-Plus.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/Makefile
patching file arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus.dts

Applying /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/203-rockchip-rk3328-Add-support-for-EmbedFire-DoorNet1.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/Makefile
patching file arch/arm64/boot/dts/rockchip/Makefile
Hunk #1 FAILED at 8.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm64/boot/dts/rockchip/Makefile.rej
patching file arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts
Patch failed!  Please fix /home/zhiliang/scripts/neo3/lede/target/linux/rockchip/patches-5.15/203-rockchip-rk3328-Add-support-for-EmbedFire-DoorNet1.patch!
make[4]: *** [Makefile:35: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.27/.prepared_e1be1a19a20354ce21f437f7350a1c33] Error 1
make[4]: Leaving directory '/home/zhiliang/scripts/neo3/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/zhiliang/scripts/neo3/lede/target/linux'
time: target/linux/compile#9.89#5.96#10.60
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/neo3/lede'
make[1]: *** [target/Makefile:18: /home/zhiliang/scripts/neo3/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/neo3/lede'
make: *** [/home/zhiliang/scripts/neo3/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-09:
5.15 升级原因，懒得更新了，等他们修吧

1054009064, 2022-03-09:
貌似重复了

https://github.com/coolsnowwolf/lede/blob/c82ca104504c45f82dbff8a5077410283ad51634/target/linux/rockchip/patches-5.15/203-rockchip-rk3328-Add-support-for-EmbedFire-DoorNet1.patch#L1-L10

https://github.com/coolsnowwolf/lede/blob/c82ca104504c45f82dbff8a5077410283ad51634/target/linux/rockchip/patches-5.15/203-rockchip-rk3328-Add-support-for-EmbedFire-DoorNet1.patch#L11-L20

WYC-2020, 2022-03-09:
> 貌似重复了
>
> https://github.com/coolsnowwolf/lede/blob/c82ca104504c45f82dbff8a5077410283ad51634/target/linux/rockchip/patches-5.15/203-rockchip-rk3328-Add-support-for-EmbedFire-DoorNet1.patch#L1-L10 [...]

那不清楚 没看

1054009064, 2022-03-09:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：编译最新固件出现错误，有一个补丁打不上
>
> ### (2) 路由器型号和固件版本
> A：rockchip rk3328 5.15.27内核版
>
> ### (3) 详细日志
> A： [...]

@ricksuzade-maker  拉取 7af03c8483a135a44fbee34e15011761370c1470 编译后无相关报错

closed 2022-03-10 by AmadeusGhost