coolsnowwolf/lede #9041: x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from

Dawneng, 2022-03-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from  
强迫症看着这个难受，请问如何消除？

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

make[5]: Entering directory '/home/openwrt/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.103'
  DESCEND  objtool
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/kernel/head_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/entry/entry_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/kernel/irqflags.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/hweight.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/clear_page_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/copy_mc_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/copy_page_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/copy_user_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/getuser.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/memcpy_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/memmove_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/memset_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/putuser.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/retpoline.o' being placed in section `__ksymtab_strings'
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/kernel/head_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/entry/entry_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/kernel/irqflags.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/hweight.o' being placed in section `__ksymtab_strings'

WYC-2020, 2022-03-10:
其实这个也没啥影响，我这边修复了，只是好看而已
https://github.com/WYC-2020/lede/commit/97ed560d4df5a283277197885684a16d9d8b9bbb
@Dawneng

Dawneng, 2022-03-10:
> 其实这个也没啥影响，我这边修复了，只是好看而已 [WYC-2020/lede@97ed560](https://github.com/WYC-2020/lede/commit/97ed560d4df5a283277197885684a16d9d8b9bbb) @Dawneng

好的，感谢大佬！  就是看着不舒服   哈哈

closed 2022-03-10 by Dawneng