coolsnowwolf/lede #9043: IPQ806x能不能由5.4内核升级5.10系列

heiger8231, 2022-03-10:
r7800现在稳定版核心还是5.4，而其它的都已经升级到5.10了，请大佬抽空升级一下，谢谢

uniessey, 2022-03-11:
您试着打开test参数试试

1054009064, 2022-03-11:
`Global build settings ---> [y] Use the testing kernel version` 
Please.

heiger8231, 2022-03-11:
Applying /srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/target/linux/ipq806x/patches-5.10/0065-arm-override-compiler-flags.patch using plaintext:
patching file arch/arm/Makefile

Applying /srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/target/linux/ipq806x/patches-5.10/0067-generic-Mangle-bootloader-s-kernel-arguments.patch using plaintext:
patching file arch/arm/Kconfig
patching file arch/arm/boot/compressed/atags_to_fdt.c
patching file init/main.c

Applying /srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/target/linux/ipq806x/patches-5.10/0069-arm-boot-add-dts-files.patch using plaintext:
patching file arch/arm/boot/dts/Makefile
patch: **** malformed patch at line 43: +       qcom-ipq8068-ecw5410.dtb \

Patch failed!  Please fix /srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/target/linux/ipq806x/patches-5.10/0069-arm-boot-add-dts-files.patch!
make[4]: *** [Makefile:38: /srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.10.103/.prepared_e30bc5c9e81abb80dfe90c213c41a356] Error 1
make[4]: Leaving directory '/srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/target/linux/ipq806x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/target/linux'
time: target/linux/compile#16.30#11.14#336.32
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800'
make[1]: *** [target/Makefile:18: /srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800'
make: *** [/srv/dev-disk-by-uuid-ec5758c1-9307-48ba-af14-359aee48245e/openwrt/lean_netgear_r7800/include/toplevel.mk:230：world] 错误 2

coolsnowwolf, 2022-03-11:
5.10 806x 目前无法支持 NSS 加速，性能大降

closed 2022-03-13 by AmadeusGhost