coolsnowwolf/lede #9044: 早上升级后编译错误。

gswangmd-gssi, 2022-03-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
SHELL= flock /home/gswangmd/lede/tmp/.root-copy.flock -c 'cp -fpR /home/gswangmd/lede/build_dir/target-aarch64_cortex-a53_musl/luci-app-mwan3/.pkgdir/luci-i18n-mwan3-zh-cn/. /home/gswangmd/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /home/gswangmd/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.luci-i18n-mwan3-zh-cn_installed
echo "luci-i18n-mwan3-zh-cn" >> /home/gswangmd/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/luci-app-mwan3.default.install
make[3]: Leaving directory '/home/gswangmd/lede/feeds/luci/applications/luci-app-mwan3'
time: package/feeds/luci/luci-app-mwan3/compile#0.32#0.03#0.32
make[3]: Entering directory '/home/gswangmd/lede/feeds/packages/admin/netdata'
mkdir -p /home/gswangmd/lede/dl
SHELL= flock /home/gswangmd/lede/tmp/.netdata-v1.33.1.tar.gz.flock -c '  	/home/gswangmd/lede/scripts/download.pl "/home/gswangmd/lede/dl" "netdata-v1.33.1.tar.gz" "20ba8695d87187787b27128ac3aab9b09aa29ca6b508c48542e0f7d50ec9322b" "" "https://github.com/netdata/netdata/releases/download/v1.33.1"    '
Hash of the local file netdata-v1.33.1.tar.gz does not match (file: fae8ba3d88c555fd7fb3ddf91c34e827163823f269d8469e02627df50f4ba76d, requested: 20ba8695d87187787b27128ac3aab9b09aa29ca6b508c48542e0f7d50ec9322b) - deleting download.
make[3]: *** [Makefile:105: /home/gswangmd/lede/dl/netdata-v1.33.1.tar.gz] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/feeds/packages/admin/netdata'
time: package/feeds/packages/netdata/compile#0.23#0.06#0.23
    ERROR: package/feeds/packages/netdata failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/netdata/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-11:
hash不对，开代理下吧，或者收工下载放dl

lxwq, 2022-03-11:
删除dl文件夹下的netdata重新下载,
然后出下面错误
Package netdata is missing dependencies for the following libraries:
liblz4.so.1

WYC-2020, 2022-03-11:
重新make menuconfig保存，编译，这些嘛自己多试嘛

gswangmd-gssi, 2022-03-11:
这些常规做法，都尝试过了。

WYC-2020, 2022-03-11:
> 这些常规做法，都尝试过了。

我这边一切正常，等会到了我看下那个库在那个仓库，十之八九你源没那个库

breakings, 2022-03-11:
> 重新make menuconfig保存，编译，这些嘛自己多试嘛

> 这些常规做法，都尝试过了。

已经加了liblz4

WYC-2020, 2022-03-11:
> > 重新make menuconfig保存，编译，这些嘛自己多试嘛
> 
> > 这些常规做法，都尝试过了。
> 
> 已经加了liblz4

加了不应该这样才对吧，单独clean后再编译呢，或者你用下我仓库的那个makefile看下，我和l大的有区别，我用我自己的一切正常呢

breakings, 2022-03-11:
> 这些常规做法，都尝试过了。

删除原先下载的文件
Hash of the local file netdata-v1.33.1.tar.gz does not match (file: fae8ba3d88c555fd7fb3ddf91c34e827163823f269d8469e02627df50f4ba76d, requested: 20ba8695d87187787b27128ac3aab9b09aa29ca6b508c48542e0f7d50ec9322b) - deleting download.

closed 2022-03-13 by AmadeusGhost