coolsnowwolf/lede #9046: autocore、autosamba等包编译失败

itsmine, 2022-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

云编译过程中
出现
Unknown package 'autocore-x86'.
Installing automount (1-37) to root...
Installing libuuid1 (2.37.3-1) to root...
Installing libblkid1 (2.37.3-1) to root...
Installing ntfs3-mount (2) to root...
Unknown package 'autosamba'.
Unknown package 'ddns-scripts_aliyun'.
Unknown package 'ddns-scripts_dnspod'.
Unknown package 'default-settings'.
Unknown package 'ipv6helper'.

unknown package会导致下面编译失败。我查看了.config，看到这些包都有。想问问这块怎么弄？

### (3) 详细日志
Configuring kmod-fs-ext4.
Configuring kmod-nls-cp437.
Configuring kmod-nls-iso8859-1.
Configuring kmod-fs-vfat.
Configuring kmod-usb-storage-extras.
Configuring automount.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency bc for autocore-x86
Configuring kmod-crypto-lib-chacha20.
Configuring kmod-crypto-lib-poly1305.
Configuring kmod-crypto-lib-chacha20poly1305.
 * pkg_hash_check_unresolved: cannot find dependency lm-sensors for autocore-x86
Configuring kmod-ipt-iprange.
 * pkg_hash_fetch_best_installation_candidate: Packages for autocore-x86 found, but incompatible with the architectures configured
Configuring iptables-mod-iprange.
 * opkg_install_cmd: Cannot install package autocore-x86.
Configuring kmod-ipt-nat.
 * pkg_hash_check_unresolved: cannot find dependency luci-app-samba for autosamba
Configuring iptables-mod-fullconenat.
 * pkg_hash_check_unresolved: cannot find dependency wsdd2 for autosamba
Configuring firewall.
 * pkg_hash_fetch_best_installation_candidate: Packages for autosamba found, but incompatible with the architectures configured
Configuring grub2-efi.
 * opkg_install_cmd: Cannot install package autosamba.
Configuring libmnl0.
 * pkg_hash_check_unresolved: cannot find dependency ddns-scripts for ddns-scripts_aliyun
Configuring kmod-mmc.
 * pkg_hash_check_unresolved: cannot find dependency wget-ssl for ddns-scripts_aliyun
Configuring odhcp6c.
 * pkg_hash_fetch_best_installation_candidate: Packages for ddns-scripts_aliyun found, but incompatible with the architectures configured
Configuring kmod-udptunnel4.
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
Configuring kmod-udptunnel6.
 * pkg_hash_check_unresolved: cannot find dependency ddns-scripts for ddns-scripts_dnspod
Configuring ethtool.
 * pkg_hash_check_unresolved: cannot find dependency wget-ssl for ddns-scripts_dnspod
Configuring uci.
 * pkg_hash_fetch_best_installation_candidate: Packages for ddns-scripts_dnspod found, but incompatible with the architectures configured
Configuring grub2.
 * opkg_install_cmd: Cannot install package ddns-scripts_dnspod.
Configuring lua.
 * pkg_hash_check_unresolved: cannot find dependency luci-base for default-settings
Configuring dropbear.
 * pkg_hash_check_unresolved: cannot find dependency luci for default-settings
Configuring kmod-crypto-kpp.
 * pkg_hash_fetch_best_installation_candidate: Packages for default-settings found, but incompatible with the architectures configured
Configuring libgmp10.
 * opkg_install_cmd: Cannot install package default-settings.
Configuring kmod-drm-ttm.
 * pkg_hash_check_unresolved: cannot find dependency luci-proto-ipv6 for ipv6helper
Configuring mtd.
 * pkg_hash_fetch_best_installation_candidate: Packages for ipv6helper found, but incompatible with the architectures configured
Configuring vsftpd-alt.
 * opkg_install_cmd: Cannot install package ipv6helper.
Configuring odhcpd-ipv6only.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[1]: *** [package/Makefile:111: /home/runner/work/LedeOpenwrt/LedeOpenwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
Configuring kmod-lib-raid6.
Configuring kmod-lib-lzo.

mitsukileung, 2022-03-11:
你出错的包我也有编译
刚刚编译完X86_64没有这个问题

WYC-2020, 2022-03-11:
十之八九 要更新他的config了，自己本地搭建环境 按你需要的东西选择好，生成config后自己更新

itsmine, 2022-03-11:
我是先在本地搭好环境，运行make defconfig，然后再 make menuconfig，再把.config拿到action上编译。

closed 2022-03-11 by AmadeusGhost