coolsnowwolf/lede #9048: 编译错误

pencail, 2022-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

用虚拟机编译到最后一直报错，网络环境一直挂的全局

### (2) 路由器型号和固件版本

A：编译的是X86，固件是最新下载的

### (3) 详细日志

make[3]: *** [Makefile:86: /home/pencail/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.built] Error 1
make[3]: Leaving directory '/home/pencail/lede/feeds/packages/multimedia/UnblockNeteaseMusic-Go'
time: package/feeds/packages/UnblockNeteaseMusic-Go/compile#0.14#0.45#90.44
    ERROR: package/feeds/packages/UnblockNeteaseMusic-Go failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/UnblockNeteaseMusic-Go/compile] Error 1
make[2]: Leaving directory '/home/pencail/lede'
make[1]: *** [package/Makefile:110: /home/pencail/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pencail/lede'
make: *** [/home/pencail/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-03-11:
单线程 每次有编译错误记得贴单线程日志，每次都在说说累了

WYC-2020, 2022-03-11:
没有单线程，单从这个日志上看 猜测可能是下载失败，最好贴单线程
 @pencail

pencail, 2022-03-11:
> 没有单线程，单从这个日志上看 猜测可能是下载失败，最好贴单线程 @pencail

单线程的日志在哪里找，我去找找，第一次玩编译，好多东西都不知道

WYC-2020, 2022-03-11:
make -j1 V=s @pencail

pencail, 2022-03-11:
> make -j1 V=s @pencail
我就是这么做的，然后就出了上边的错误

time: target/linux/prereq#0.16#0.08#0.33
make[1]: Entering directory '/home/pencail/lede'
make[2]: Entering directory '/home/pencail/lede'
make[3]: Entering directory '/home/pencail/lede/target/linux'
make[4]: Entering directory '/home/pencail/lede/target/linux/x86'
rm -f /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/localversion
/home/pencail/lede/scripts/kconfig.pl  + + /home/pencail/lede/target/linux/generic/config-5.15 /home/pencail/lede/target/linux/x86/config-5.15 /home/pencail/lede/target/linux/x86/64/config-5.15 > /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/pencail/lede/.config >> /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.target
/home/pencail/lede/scripts/package-metadata.pl kconfig /home/pencail/lede/tmp/.packageinfo /home/pencail/lede/.config 5.15 > /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.override
/home/pencail/lede/scripts/kconfig.pl 'm+' '+' /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.target /dev/null /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.override > /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set
mv /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.old
grep -v INITRAMFS /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.old > /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set
rm -rf /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.prev || { cp /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config; cp /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.prev; }
export MAKEFLAGS= ; [ -d /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/user_headers ] || make -C /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27 KCFLAGS="-fmacro-prefix-map=/home/pencail/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/pencail/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Mar 10 00:56:14 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/pencail/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/pencail/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.27 CC="x86_64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/user_headers headers_install
grep '=[ym]' /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.config.set | LC_ALL=C sort | /home/pencail/lede/staging_dir/host/bin/mkhash md5 > /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.vermagic
touch /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/.configured
rm -f /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/vmlinux /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27/System.map
make -C /home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27 KCFLAGS="-fmacro-prefix-map=/home/pencail/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/pencail/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Mar 10 00:56:14 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/pencail/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/pencail/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.27 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/pencail/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.27'
net/sched/Kconfig:45: warning: menuconfig statement without prompt
#
# No change to .config
#
[... cut ...]
po2lmo /home/pencail/lede/feeds/luci/modules/luci-base/po/zh-cn/base.po /home/pencail/lede/build_dir/target-x86_64_musl/luci-base/ipkg-all/luci-i18n-base-zh-cn/usr/lib/lua/luci/i18n/base.zh-cn.lmo;
find /home/pencail/lede/build_dir/target-x86_64_musl/luci-base/ipkg-all/luci-i18n-base-zh-cn -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/pencail/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/pencail/lede/scripts/strip-kmod.sh" PATCHELF="/home/pencail/lede/staging_dir/host/bin/patchelf" /home/pencail/lede/scripts/rstrip.sh /home/p

WYC-2020, 2022-03-11:
那就是下载失败了，取消编译这个看下，这个解锁音乐还有另外一个插件你用那个撒

pencail, 2022-03-11:
> 那就是下载失败了，取消编译这个看下，这个解锁音乐还有另外一个插件你用那个撒
感谢感谢，我一开始把这个插件取消都不管用，后来我删了就编译成功了，谢谢谢谢

bauw2008, 2022-03-13:
这个今天遇到过，用go env -w GOPROXY=https://goproxy.cn,也没效果，最后还是换了节点，重新更新下feeds.就通过了。

closed 2022-10-31 by aiamadeus