coolsnowwolf/lede #9049: macOS下编译，UCL不识别系统类型

AstralHope, 2022-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：
 [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：按照3月8日更新的文档[Build system setup macOS](https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx)编译时，不识别系统类型`aarch64-apple`。看日志以前有给m1打过补丁？但`checking build system type`这里过不去

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

make[3]: Entering directory '/Volumes/OpenWrt/lede/tools/ucl'
. /Volumes/OpenWrt/lede/include/shell.sh; gzip -dc /Volumes/OpenWrt/lede/dl/ucl-1.03.tar.gz | tar -C /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03

Applying ./patches/100-fix-apple-m1-support.patch using plaintext: 
patching file acconfig/config.guess
mkdir -p /Volumes/OpenWrt/lede/staging_dir/host/include/ucl
touch /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.prepared8e56ca839abb48ae5ba758fd357cb845_6664517399ebbbc92a37c5bb081b5c53
(cd /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03; CC="gcc" CFLAGS="-O2 -I/Volumes/OpenWrt/lede/staging_dir/host/include  -std=gnu89" ./configure --prefix=/Volumes/OpenWrt/lede/staging_dir/host );
configure: loading site script /Volumes/OpenWrt/lede/include/site/darwin
configure: Configuring UCL 1.03
checking build system type... Invalid configuration `aarch64-apple-darwin21.3.0': machine `aarch64-apple' not recognized
configure: error: /bin/sh acconfig/config.sub aarch64-apple-darwin21.3.0 failed
make[3]: *** [Makefile:50: /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ucl'
time: tools/ucl/compile#0.10#0.24#0.95
    ERROR: tools/ucl failed to build.
make[2]: *** [tools/Makefile:160: tools/ucl/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-11:
这个无能为力，你可以试下吧哪个patch 删掉后 再编译看下可行不 @AstralHope

WYC-2020, 2022-03-11:
ucl 不打patch 编译 试下 @AstralHope

AstralHope, 2022-03-12:
> 

直接删掉的话检查系统类型就报这个了
checking build system type... configure: error: /bin/sh acconfig/config.sub -apple-darwin21.3.0 failed

patch的代码我看不懂，哪部分是识别系统类型的？我看在ubuntu下正常编译的反馈是
checking build system type... gcc -DHAVE_CONFIG_H -I. -I..   -I/home/astral/lede/staging_dir/host/include   -O2 -I/home/astral/lede/staging_dir/host/include  -MT xalloc-die.o -MD -MP -MF .deps/xalloc-die.Tpo -c -o xalloc-die.o xalloc-die.c @WYC-2020

WYC-2020, 2022-03-15:
@AstralHope 
https://github.com/WYC-2020/lede/commit/576fbe6f9cc7d06aad2fcb6e94775d59ad09855c
盲修没环境测试，你测试一下看下吧

WYC-2020, 2022-03-15:
原来的patch 也需要 @AstralHope

AstralHope, 2022-03-15:
> @AstralHope [WYC-2020/lede@576fbe6](https://github.com/WYC-2020/lede/commit/576fbe6f9cc7d06aad2fcb6e94775d59ad09855c) 盲修没环境测试，你测试一下看下吧

 改报这个错了

@WYC-2020 

```bash
make[3]: Entering directory '/Volumes/OpenWrt/lede/tools/ucl'
. /Volumes/OpenWrt/lede/include/shell.sh; gzip -dc /Volumes/OpenWrt/lede/dl/ucl-1.03.tar.gz | tar -C /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03

Applying ./patches/0001-fix-build-for-macos.patch using plaintext: 
patching file acconfig/config.sub

Applying ./patches/001-autoconf-compat.patch using plaintext: 
patch: **** Only garbage was found in the patch input.
Patch failed!  Please fix ./patches/001-autoconf-compat.patch!
make[3]: *** [Makefile:55: /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.prepared93ed43737dff02ef93ddc5c8b60b6a87_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ucl'
time: tools/ucl/compile#0.03#0.45#0.55
    ERROR: tools/ucl failed to build.
make[2]: *** [tools/Makefile:160: tools/ucl/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-03-15:
@AstralHope 删掉这个patch试下，不行的话明天我看有时间没，安装一个macos环境看下

AstralHope, 2022-03-15:
> @AstralHope 删掉这个patch试下，不行的话明天我看有时间没，安装一个macos环境看下

直接删掉`001-autoconf-compat.patch`后再编译的报错：

```bash
checking whether your compiler passes the ACC conformance test... FAILED
configure: 
configure: Your compiler failed the ACC conformance test - for details see 
configure: `config.log'. Please check that log file and consider sending
configure: a patch or bug-report to <markus@oberhumer.com>.
configure: Thanks for your support.
configure: 
configure: error: ACC conformance test failed. Stop.
make[3]: *** [Makefile:50: /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ucl'
time: tools/ucl/compile#5.23#5.59#18.92
    ERROR: tools/ucl failed to build.
make[2]: *** [tools/Makefile:160: tools/ucl/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
```

对Mac下编译倒是没有刚需，看见支持就试了下，没想到有坑:joy:，辛苦了

WYC-2020, 2022-03-15:
> > @AstralHope 删掉这个patch试下，不行的话明天我看有时间没，安装一个macos环境看下
> 
> @WYC-2020 直接删掉`001-autoconf-compat.patch`后再编译的报错：
> 
> ```shell
> checking whether your compiler passes the ACC conformance test... FAILED
> configure: 
> configure: Your compiler failed the ACC conformance test - for details see 
> configure: `config.log'. Please check that log file and consider sending
> configure: a patch or bug-report to <markus@oberhumer.com>.
> configure: Thanks for your support.
> configure: 
> configure: error: ACC conformance test failed. Stop.
> make[3]: *** [Makefile:50: /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.configured] Error 1
> make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ucl'
> time: tools/ucl/compile#5.23#5.59#18.92
>     ERROR: tools/ucl failed to build.
> make[2]: *** [tools/Makefile:160: tools/ucl/compile] Error 1
> make[2]: Leaving directory '/Volumes/OpenWrt/lede'
> make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
> make[1]: Leaving directory '/Volumes/OpenWrt/lede'
> make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
> ```
> 
> 对Mac下编译倒是没有刚需，看见支持就试了下，没想到有坑😂，辛苦了

编译的什么类型，发出来，有空的时候我看下

xiufeiwu, 2022-03-16:
这个我翻别人的文章解决了，原版patch基础上先补个patch（见附件），Makefile里这行改成这样：HOST_CFLAGS +=-std=gnu89 -std=c90 -fPIC -Wno-error=implicit-function-declaration，没猜错你是M1版本吧；这样之后能编译通过，后面你会卡在zerotierone的编译过不去的，哈哈，目前没找到解决方案
[ucl-apple-m1-configsub-fix.zip](https://github.com/coolsnowwolf/lede/files/8259876/ucl-apple-m1-configsub-fix.zip)

WYC-2020, 2022-03-16:
@xiufeiwu 我这边虚拟机mac cpu平台不是你哪个所以没法测试了，https://github.com/WYC-2020/lede/commit/56fc735b121c1a7c27b7bf2613099d06fa5e181f 
你这样改下看行不，还不行我也无能为力了，没环境无法测试

WYC-2020, 2022-03-16:
> 这个我翻别人的文章解决了，原版patch基础上先补个patch（见附件），Makefile里这行改成这样：HOST_CFLAGS +=-std=gnu89 -std=c90 -fPIC -Wno-error=implicit-function-declaration，没猜错你是M1版本吧；这样之后能编译通过，后面你会卡在zerotierone的编译过不去的，哈哈，目前没找到解决方案 [...]

这个补丁l大本身就有

xiufeiwu, 2022-03-16:
> 这个补丁l大本身就有

l大补的是config.guess，m1上虚拟机ubuntu环境用这个补丁就可以成功了；但是m1原生用homebrew编译的话config.sub也要补的，然后加上编译的参数才能编译通过

WYC-2020, 2022-03-16:
> > 这个补丁l大本身就有
> 
> l大补的是config.guess，m1上虚拟机ubuntu环境用这个补丁就可以成功了；但是m1原生用homebrew编译的话config.sub也要补的，然后加上编译的参数才能编译通过

这个可以让有环境的测试下，然后加入

AstralHope, 2022-03-16:
> @xiufeiwu 我这边虚拟机mac cpu平台不是你哪个所以没法测试了，[WYC-2020/lede@56fc735](https://github.com/WYC-2020/lede/commit/56fc735b121c1a7c27b7bf2613099d06fa5e181f) 你这样改下看行不，还不行我也无能为力了，没环境无法测试
https://github.com/WYC-2020/lede/commit/56fc735b121c1a7c27b7bf2613099d06fa5e181f 这个我编译没过，
```bash
checking how to run the C++ preprocessor... /lib/cpp
configure: error: C++ preprocessor "/lib/cpp" fails sanity check
See `config.log' for more details.
make[3]: *** [Makefile:50: /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ucl'
time: tools/ucl/compile#0.92#1.01#3.65
    ERROR: tools/ucl failed to build.
make[2]: *** [tools/Makefile:160: tools/ucl/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
```

Makefile修改成`HOST_CFLAGS +=-std=gnu89 -std=c90 -fPIC -Wno-error=implicit-function-declaration`，添加ucl-apple-m1-configsub-fix这个patch的没选zerotier还是没过，不过报错好像和这个无关了？

```bash
make[1]: *** [toolchain/Makefile:93: /Volumes/OpenWrt/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-03-16:
> > @xiufeiwu 我这边虚拟机mac cpu平台不是你哪个所以没法测试了，[WYC-2020/lede@56fc735](https://github.com/WYC-2020/lede/commit/56fc735b121c1a7c27b7bf2613099d06fa5e181f) 你这样改下看行不，还不行我也无能为力了，没环境无法测试
> > [WYC-2020/lede@56fc735](https://github.com/WYC-2020/lede/commit/56fc735b121c1a7c27b7bf2613099d06fa5e181f) 这个我编译没过，
> 
> ```shell
> checking how to run the C++ preprocessor... /lib/cpp
> configure: error: C++ preprocessor "/lib/cpp" fails sanity check
> See `config.log' for more details.
> make[3]: *** [Makefile:50: /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.configured] Error 1
> make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ucl'
> time: tools/ucl/compile#0.92#1.01#3.65
>     ERROR: tools/ucl failed to build.
> make[2]: *** [tools/Makefile:160: tools/ucl/compile] Error 1
> make[2]: Leaving directory '/Volumes/OpenWrt/lede'
> make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
> make[1]: Leaving directory '/Volumes/OpenWrt/lede'
> make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
> ```
> 
> Makefile修改成`HOST_CFLAGS +=-std=gnu89 -std=c90 -fPIC -Wno-error=implicit-function-declaration`，添加ucl-apple-m1-configsub-fix这个patch的没选zerotier还是没过，不过报错好像和这个无关了？
> 
> 测试环境我按[Build system setup macOS](https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx)配的，M1max的机器，12.3系统，homebrew装的各种包
> 
> ```shell
> make[5]: Leaving directory '/Volumes/OpenWrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial/gcc'
> make[4]: *** [Makefile:4279: all-gcc] Error 2
> make[4]: Leaving directory '/Volumes/OpenWrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial'
> make[3]: *** [Makefile:37: /Volumes/OpenWrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Error 2
> make[3]: Leaving directory '/Volumes/OpenWrt/lede/toolchain/gcc/initial'
> time: toolchain/gcc/initial/compile#4.54#0.76#2.58
>     ERROR: toolchain/gcc/initial failed to build.
> make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
> make[2]: Leaving directory '/Volumes/OpenWrt/lede'
> make[1]: *** [toolchain/Makefile:93: /Volumes/OpenWrt/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
> make[1]: Leaving directory '/Volumes/OpenWrt/lede'
> make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
> ```

那就是说另外一个兄弟的patch可以解决这个编译问题嘛

xiufeiwu, 2022-03-18:
> ```shell
> checking how to run the C++ preprocessor... /lib/cpp
> configure: error: C++ preprocessor "/lib/cpp" fails sanity check
> See `config.log' for more details.
> make[3]: *** [Makefile:50: [...]

哈哈，我又来了，我踩过的坑，继续帮你；附件patch放到 toolchain-gcc-patches，8.4下面，然后就能解决了，再往后应该就到我上面说的zerotierone的包没法编译过了，我从去年7月卡到现在，至今没有人解决，翻遍了所有网站
[gcc8.4_apple_silicon.patch.zip](https://github.com/coolsnowwolf/lede/files/8296715/gcc8.4_apple_silicon.patch.zip)

WYC-2020, 2022-03-18:
> 哈哈，我又来了，我踩过的坑，继续帮你；附件patch放到 toolchain-gcc-patches，8.4下面，然后就能解决了，再往后应该就到我上面说的zerotierone的包没法编译过了，我从去年7月卡到现在，至今没有人解决，翻遍了所有网站 [...]

放弃吧 还是用ubuntu编译吧

xiufeiwu, 2022-03-18:
ubuntu没啥难度，嘿嘿

AstralHope, 2022-03-19:
> 哈哈，我又来了，我踩过的坑，继续帮你；附件patch放到 toolchain-gcc-patches，8.4下面，然后就能解决了，再往后应该就到我上面说的zerotierone的包没法编译过了，我从去年7月卡到现在，至今没有人解决，翻遍了所有网站 [...]

还是没过，放弃，还是老老实实Ubuntu编译吧，哈哈

AstralHope, 2022-03-23:
更新了这个https://github.com/coolsnowwolf/lede/commit/04a6c950bea0c1beb14a2ee1261a3f1b9b826839

然后再做这部，ucl里
_Makefile修改成HOST_CFLAGS +=-std=gnu89 -std=c90 -fPIC -Wno-error=implicit-function-declaration，添加ucl-apple-m1-configsub-fix这个patch的，config没选zerotier_，卡在了别的地方，但ucl部分编译通过了，没测试这样改了会不会影响Ubuntu下正常编译，Ubuntu也过了的话倒是可以加入

ucl编译通过部分：
```bash
make[3]: Entering directory '/Volumes/OpenWrt/lede/tools/ucl'
. /Volumes/OpenWrt/lede/include/shell.sh; gzip -dc /Volumes/OpenWrt/lede/dl/ucl-1.03.tar.gz | tar -C /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03

Applying ./patches/100-fix-apple-m1-support.patch using plaintext: 
patching file acconfig/config.guess

Applying ./patches/ucl-apple-m1-configsub-fix.patch using plaintext: 
patching file acconfig/config.sub
patch unexpectedly ends in middle of line
Hunk #7 succeeded at 1503 with fuzz 1.
mkdir -p /Volumes/OpenWrt/lede/staging_dir/host/include/ucl
touch /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03/.prepared32cc975d59de25e4e80f23e3c5d35f59_6664517399ebbbc92a37c5bb081b5c53
(cd /Volumes/OpenWrt/lede/build_dir/host/ucl-1.03; CC="gcc" CFLAGS="-O2 -I/Volumes/OpenWrt/lede/staging_dir/host/include  -std=gnu89 -std=c90 -fPIC -Wno-error=implicit-function-declaration" ./configure --prefix=/Volumes/OpenWrt/lede/staging_dir/host );
configure: loading site script /Volumes/OpenWrt/lede/include/site/darwin
configure: Configuring UCL 1.03
checking build system type... aarch64-apple-darwin21.4.0
checking host system type... aarch64-apple-darwin21.4.0
checking target system type... aarch64-apple-darwin21.4.0
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ANSI C... none needed
checking how to run the C preprocessor... gcc -E
checking whether the C preprocessor needs special flags... none needed
checking for an ANSI C-conforming const... yes
checking whether byte ordering is bigendian... no
```

新问题：
```bash
make[5]: Entering directory '/Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.30'
make[6]: *** No rule to make target '/Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.12/ioctl.o', needed by '/Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.12/cryptodev.o'.  Stop.
make[5]: *** [Makefile:1868: /Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.12] Error 2
make[5]: Leaving directory '/Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.30'
make[4]: *** [Makefile:27: build] Error 2
make[4]: Leaving directory '/Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.12'
make[3]: *** [Makefile:61: /Volumes/OpenWrt/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.12/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/lede/package/kernel/cryptodev-linux'
time: package/kernel/cryptodev-linux/compile#0.25#0.40#0.76
    ERROR: package/kernel/cryptodev-linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/cryptodev-linux/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/OpenWrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
```

closed 2022-04-24 by AstralHope