coolsnowwolf/lede #9050: aarch64_cortex-a53+crypto 编译 mbedtls-2.28.0报错

sKyissKy, 2022-03-11:
aarch64_cortex-a53+crypto 编译 mbedtls-2.28.0报错

MAKEFLAGS="" /home/ubuntu/lede/staging_dir/host/bin/ninja  -j1 -C /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0 
ninja: Entering directory `/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0'
[1/299] Building C object library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o
FAILED: library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o 
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc  -I/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/include -I/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0=mbedtls-2.28.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -O2 -MD -MT library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -MF library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o.d -o library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -c /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c: In function 'mbedtls_armv8ce_aes_crypt_ecb':
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12440:1: error: inlining failed in call to always_inline 'vaesmcq_u8': target specific option mismatch
 vaesmcq_u8 (uint8x16_t data)
 ^~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:62:17: note: called from here
             x = vaesmcq_u8( x );
                 ^~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:61:17: note: called from here
             x = vaeseq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12440:1: error: inlining failed in call to always_inline 'vaesmcq_u8': target specific option mismatch
 vaesmcq_u8 (uint8x16_t data)
 ^~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:62:17: note: called from here
             x = vaesmcq_u8( x );
                 ^~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:61:17: note: called from here
             x = vaeseq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:66:13: note: called from here
         x = vaeseq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:66:13: note: called from here
         x = vaeseq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12447:1: error: inlining failed in call to always_inline 'vaesimcq_u8': target specific option mismatch
 vaesimcq_u8 (uint8x16_t data)
 ^~~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:75:17: note: called from here
             x = vaesimcq_u8( x );
                 ^~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:74:17: note: called from here
             x = vaesdq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12447:1: error: inlining failed in call to always_inline 'vaesimcq_u8': target specific option mismatch
 vaesimcq_u8 (uint8x16_t data)
 ^~~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:75:17: note: called from here
             x = vaesimcq_u8( x );
[... 7 lines cut ...]
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:79:13: note: called from here
         x = vaesdq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
[... cut ...]
make[3]: Leaving directory '/home/ubuntu/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.18#0.07#0.24
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53+crypto_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-03-11:
https://github.com/coolsnowwolf/lede/issues/8922 相同issue不搜索一下

sKyissKy, 2022-03-11:
> #8922 相同issue不搜索一下

问题不完全一样……因为架构不加crypto直接就能编译过去…我试试这个办法看能不能过

WYC-2020, 2022-03-11:
> > #8922 相同issue不搜索一下
> 
> 问题不完全一样……因为架构不加crypto直接就能编译过去…我试试这个办法看能不能过

咋不一样 都是一个函数 相同的提示，相同的平台

WYC-2020, 2022-03-11:
有时间我编一个这个看下，没时间编译各种类型的

WYC-2020, 2022-03-11:
[image]
@sKyissKy  我这边没问题呢，你把你的config 文件发给我看看

sKyissKy, 2022-03-11:
> [image] @sKyissKy 我这边没问题呢，你把你的config 文件发给我看看

你的架构不对。。。。。选ipq60XX，就有crypto，你的架构后面都没有crypto

sKyissKy, 2022-03-11:
> [image] @sKyissKy 我这边没问题呢，你把你的config 文件发给我看看

你应该选的是ipq807x的，后面没有crypto，这个我编译也是没问题的

WYC-2020, 2022-03-11:
config 发给我 我懒得找 @sKyissKy

WYC-2020, 2022-03-11:
@sKyissKy 
[image]
没问题自己找原因吧 溜了

sKyissKy, 2022-03-11:
> @sKyissKy
> [image]
> 没问题自己找原因吧 溜了

不是没问题，我现在发不了啊，晚上给你发

WYC-2020, 2022-03-11:
> > @sKyissKy
> > [image]
> > 没问题自己找原因吧 溜了
> 
> 不是没问题，我现在发不了啊，晚上给你发

我说这边编译 了 没问题 带crypto的，让你自己找下原因了

sKyissKy, 2022-03-11:
> > > @sKyissKy
> > > [image]
> > > 没问题自己找原因吧 溜了
> > 
> > 
> > 不是没问题，我现在发不了啊，晚上给你发
> 
> 我说这边编译 了 没问题 带crypto的，让你自己找下原因了

sKyissKy, 2022-03-11:
> > > > @sKyissKy
> > > > [image]
> > > > 没问题自己找原因吧 溜了
> > > 
> > > 
> > > 不是没问题，我现在发不了啊，晚上给你发
> > 
> > 
> > 我说这边编译 了 没问题 带crypto的，让你自己找下原因了
[config.txt](https://github.com/coolsnowwolf/lede/files/8232085/config.txt)

sKyissKy, 2022-03-11:
> > > @sKyissKy
> > > [image]
> > > 没问题自己找原因吧 溜了
> > 
> > 
> > 不是没问题，我现在发不了啊，晚上给你发
> 
> 我说这边编译 了 没问题 带crypto的，让你自己找下原因了

我现在重新pull和update还是不行。。。。给你看看config
[config.txt](https://github.com/coolsnowwolf/lede/files/8232326/config.txt)
吧

littleBanshee, 2022-03-14:
built last night,all ok
[image]

closed 2022-03-17 by sKyissKy