coolsnowwolf/lede #9054: 一个5.15的patch坏掉了,求修复

realJiaoKan, 2022-03-11:
日志:
`Applying /home/parallels/Workshop/lede/target/linux/generic/hack-5.15/220-arm-gc_sections.patch using plaintext: 
patching file arch/arm/Kconfig
patching file arch/arm/boot/compressed/Makefile
patching file arch/arm/kernel/vmlinux.lds.S
patching file arch/arm/include/asm/vmlinux.lds.h
Hunk #1 succeeded at 42 (offset 13 lines).
Hunk #2 succeeded at 109 (offset 13 lines).
Hunk #3 FAILED at 112.
1 out of 3 hunks FAILED -- saving rejects to file arch/arm/include/asm/vmlinux.lds.h.rej
Patch failed!  Please fix /home/parallels/Workshop/lede/target/linux/generic/hack-5.15/220-arm-gc_sections.patch!`

编译环境:
Ubuntu 20.04 (Parallels on macOS Monterey)

minlang112, 2022-03-12:
[kernel: fix 5.15.28 patch error][220-arm-gc_sections.patch]

你是指这个？

closed 2022-03-12 by coolsnowwolf

realJiaoKan, 2022-03-12:
可以用了,谢谢