coolsnowwolf/lede #9061: download hash对不上

SkyTeamWater, 2022-03-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

download 哈希对不上，使用了pr #6526 

### (2) 路由器型号和固件版本

x86 R22.3.13

### (3) 详细日志

```
SHELL= flock /home/ubuntu/lede/tmp/.netdata-v1.33.1.tar.gz.flock -c '   /home/ubuntu/lede/scripts/download.pl "/home/ubuntu/lede/dl" "netdata-v1.33.1.tar.gz" "20ba8695d87187787b27128ac3aab9b09aa29ca6b508c48542e0f7d50ec9322b" "" "https://github.com/netdata/netdata/releases/download/v1.33.1"    '
Hash of the local file netdata-v1.33.1.tar.gz does not match (file: fae8ba3d88c555fd7fb3ddf91c34e827163823f269d8469e02627df50f4ba76d, requested: 20ba8695d87187787b27128ac3aab9b09aa29ca6b508c48542e0f7d50ec9322b) - deleting download.
make[2]: *** [Makefile:105: /home/ubuntu/lede/dl/netdata-v1.33.1.tar.gz] Error 2
make[2]: Leaving directory '/home/ubuntu/lede/feeds/packages/admin/netdata'
time: package/feeds/packages/netdata/download#0.21#0.06#0.25
    ERROR: package/feeds/packages/netdata failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/netdata/download] Error 1
```

nitsort, 2022-03-13:
把`lede/dl`里的`netdata-v1.33.1.tar`删了再编译

SkyTeamWater, 2022-03-13:
ok解决了谢谢大佬

closed 2022-03-13 by SkyTeamWater