coolsnowwolf/lede #9062: 哪位大神可以帮忙，固件编译出来的固件开机后无法使用!

shedkiss, 2022-03-13:
[image]

之前用的5.4内核 可以正常使用，更新固件后如图所示 网口无法正常识别，请大神帮忙。退回5.4又可以正常识别。配置：软路由x86，esxi7.0 Update 3里设置了网卡直通。

noyjcl, 2022-03-30:
编译成功，R2S的SYS一直慢闪。但是LAN等不亮，无法获取IP~~~~~3月30日编译的

szeller1111, 2022-04-01:
> [image]
> 
> 之前用的5.4内核 可以正常使用，更新固件后如图所示 网口无法正常识别，请大神帮忙。退回5.4又可以正常识别。配置：软路由x86，esxi7.0 Update 3里设置了网卡直通。

我和你环境一样，也有类似的问题

pplloo, 2022-04-20:
我也碰到一样的问题了。

pplloo, 2022-04-20:
对了，我用可用的旧版本内的刷新固件的功能升级新固件是可以的。

szeller1111, 2022-04-23:
> 我也碰到一样的问题了。
> 
> 不过，我用可用的旧版本内的刷新固件的功能升级新固件是可以的。

手动切换内核到5.10，并加入英特尔网卡官方驱动。可以运行了 。https://github.com/coolsnowwolf/lede/issues/8537#issuecomment-1002744412

skaterjuice, 2022-05-22:
> > I also encountered the same problem.
> > However, it is OK for me to upgrade the new firmware with the flash firmware function in the old version available.
> 
> Manually switch the kernel to 5.10 and add the official Intel NIC driver. ready to run. [#8537 (comment)](https://github.com/coolsnowwolf/lede/issues/8537#issuecomment-1002744412)

How? You have a tutorial?

MIBZORE, 2022-08-04:
> 编译成功，R2S的SYS一直慢闪。但是LAN等不亮，无法获取IP~~~~~3月30日编译的

同样的问题

coolsnowwolf, 2026-09-05:
已按当前源码核验。该 issue 最初是 2022 年 x86/ESXi 直通环境的旧内核切换报告，后续评论又混入 R2S、树莓派等不同平台，但始终没有提供 PCI ID、驱动绑定、dmesg/串口日志，因此无法对应到一个可复现的统一源码问题。当前 x86_64 默认镜像已包含 e1000e、igb、igc、ixgbe、ixgbevf，VMware vmxnet3 也已内建。若当前版本仍有问题，请按具体设备另开 issue，并附 lspci -nnk、dmesg、ethtool -i、所用内核和虚拟化直通配置。此旧 issue 先关闭。

closed 2026-09-05 by coolsnowwolf