coolsnowwolf/lede #9063: 新手编译L大源码出错，请大大们指导一下

irvineyang, 2022-03-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
编译出现了错误。错误的详细日志在下方。
然后我还运行过
make -j8 download V=s
这条命令运行时候，出现过错误：
can't open file '/home/vagrant/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.28/.config' at /home/vagrant/lede/scripts/kconfig.pl line 32
grep: /home/vagrant/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.28/.config: No such file or directory

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：if [ -f ../stage_final ] \
   && cmp -s ../stage_current ../stage_final; then \
   cp ../prev-gcc/cc1plus-checksum.c cc1plus-checksum.c; \
else \
  build/genchecksum cp/cp-lang.o c-family/stub-objc.o cp/call.o cp/class.o cp/constexpr.o cp/constraint.o cp/cp-gimplify.o cp/cp-objcp-common.o cp/cp-ubsan.o cp/cvt.o cp/cxx-pretty-print.o cp/decl.o cp/decl2.o cp/dump.o cp/error.o cp/except.o cp/expr.o cp/friend.o cp/init.o cp/lambda.o cp/lex.o cp/logic.o cp/mangle.o cp/method.o cp/name-lookup.o cp/optimize.o cp/parser.o cp/pt.o cp/ptree.o cp/repo.o cp/rtti.o cp/search.o cp/semantics.o cp/tree.o cp/typeck.o cp/typeck2.o cp/vtable-class-hierarchy.o attribs.o incpath.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o i386-c.o glibc-c.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a ../libiberty/libiberty.a   ../libdecnumber/libdecnumber.a ../libbacktrace/.libs/libbacktrace.a \
                     checksum-options > cc1plus-checksum.c.tmp &&	   \
  /home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change cc1plus-checksum.c.tmp cc1plus-checksum.c; \
fi
g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/home/vagrant/lede/staging_dir/host/include -I/home/vagrant/lede/staging_dir/host/include -I/home/vagrant/lede/staging_dir/host/include  -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o cc1plus-checksum.o -MT cc1plus-checksum.o -MMD -MP -MF ./.deps/cc1plus-checksum.TPo cc1plus-checksum.c
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1plus \
      cp/cp-lang.o c-family/stub-objc.o cp/call.o cp/class.o cp/constexpr.o cp/constraint.o cp/cp-gimplify.o cp/cp-objcp-common.o cp/cp-ubsan.o cp/cvt.o cp/cxx-pretty-print.o cp/decl.o cp/decl2.o cp/dump.o cp/error.o cp/except.o cp/expr.o cp/friend.o cp/init.o cp/lambda.o cp/lex.o cp/logic.o cp/mangle.o cp/method.o cp/name-lookup.o cp/optimize.o cp/parser.o cp/pt.o cp/ptree.o cp/repo.o cp/rtti.o cp/search.o cp/semantics.o cp/tree.o cp/typeck.o cp/typeck2.o cp/vtable-class-hierarchy.o attribs.o incpath.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o i386-c.o glibc-c.o cc1plus-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/home/vagrant/lede/staging_dir/host/lib -L/home/vagrant/lede/staging_dir/host/lib -L/home/vagrant/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz
collect2: error: ld returned 1 exit status
make[6]: *** [/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/cp/Make-lang.in:120: cc1plus] Error 1
make[6]: Leaving directory '/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/gcc'
make[5]: *** [Makefile:4270: all-gcc] Error 2
make[5]: Leaving directory '/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
make[4]: *** [Makefile:906: all] Error 2
make[4]: Leaving directory '/home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
make[3]: *** [Makefile:91: /home/vagrant/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built] Error 2
make[3]: Leaving directory '/home/vagrant/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#533.50#43.29#582.98
    ERROR: toolchain/gcc/final failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/home/vagrant/lede'
make[1]: *** [toolchain/Makefile:93: /home/vagrant/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/vagrant/lede'
make: *** [/home/vagrant/lede/include/toplevel.mk:230: world] Error 2

psonys, 2022-03-13:
同样的错误，求解

irvineyang, 2022-03-13:
@WYC-2020 大佬，如果你有空上来看到又有空的话，请指教一下，谢谢

irvineyang, 2022-03-13:
Vmware 虚拟机中安装ubuntu-20.04.4-desktop
刚才删了所有代码，重新来过，在运行
./scripts/feeds install -a
的时候，发现一些提醒

Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'luci-lib-jsonc', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'strongswan', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'strongswan-minimal', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'strongswan-mod-kernel-libipsec', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'strongswan-mod-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'strongswan-mod-xauth-generic', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a dependency on 'xl2tpd', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-server/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/mt/luci-app-mtwifi/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/mt/drivers/mt7615d/Makefile' has a dependency on 'maccalc', which does not exist
WARNING: Makefile 'package/lean/n2n_v2/Makefile' has a dependency on 'libzstd', which does not exist
WARNING: Makefile 'package/lean/n2n_v2/Makefile' has a dependency on 'libzstd', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libtins', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libyaml-cpp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/firmware/quantenna/Makefile' has a dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/firmware/quantenna/Makefile' has a build dependency on 'libtirpc', which does not exist
Installing all packages from feed packages.

请问是这些依赖引起的问题吗？如果是，这个该如何处理？

irvineyang, 2022-03-13:
再运行一次
./scripts/feeds update -a

日志是：
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
Already up to date.
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
Already up to date.
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
Already up to date.
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Already up to date.
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index'

irvineyang, 2022-03-13:
再运行一次
./scripts/feeds install -a
日志是：
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

psonys, 2022-03-13:
> 再运行一次 ./scripts/feeds install -a 日志是： Installing all packages from feed packages. Installing all packages from feed luci. Installing all packages from feed routing. Installing all packages from feed telephony.

这个是正常的

WYC-2020, 2022-03-13:
太基础，建议去看点编译的教程再说

closed 2022-03-18 by AmadeusGhost