coolsnowwolf/lede #9064: dd img出错

SkyTeamWater, 2022-03-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

最后输出img和img.gz文件出错
求助大佬们
之前几天编译都正常，config没改动过

### (2) 路由器型号和固件版本

x86 最新源码

### (3) 详细日志

```
dd if=/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new bs=2147483648 conv=sync
make[5]: *** [Makefile:164: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz] Killed
make[5]: *** Deleting file '/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz'
make[5]: Leaving directory '/home/ubuntu/lede/target/linux/x86/image'
make[4]: *** [Makefile:27: install] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/target/linux/x86'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/target/linux'
time: target/linux/install#472.62#7.08#177.83
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [target/Makefile:19: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230：world] 错误 2
```

WYC-2020, 2022-03-13:
自己看空间是否足够，这种问题十之八九都是自己问题，要是真有问题早就翻车炸了

SkyTeamWater, 2022-03-13:
空间肯定是够的

GreatMichaelLee, 2022-03-13:
> 空间肯定是够的
> […](#)

1. 用单线程编译
2. 检查你的kernel/root分区大小设置是否合理。

SkyTeamWater, 2022-03-13:
我clean了一下，把dl删了重新编译，通过了，谢谢各位

closed 2022-03-13 by SkyTeamWater