coolsnowwolf/lede #9065: X86 最新版编译报错。

jlzsss, 2022-03-14:
使用的是5.15内核，报错内容如下：
/media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c: In function 'rt2x00lib_set_mac_address':
/media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:1000:13: error: too few arguments to function 'of_get_mac_address'
  mac_addr = of_get_mac_address(rt2x00dev->dev->of_node);
             ^~~~~~~~~~~~~~~~~~
In file included from /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/backport-include/linux/of_net.h:3,
                 from /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:19:
./include/linux/of_net.h:25:19: note: declared here
 static inline int of_get_mac_address(struct device_node *np, u8 *mac)
                   ^~~~~~~~~~~~~~~~~~
make[10]: *** [scripts/Makefile.build:277: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o] Error 1
make[9]: *** [scripts/Makefile.build:540: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00] Error 2
make[8]: *** [scripts/Makefile.build:540: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink] Error 2
make[7]: *** [scripts/Makefile.build:540: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[6]: *** [Makefile:1868: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[5]: *** [Makefile.build:13: modules] Error 2
make[4]: *** [Makefile.real:93: modules] Error 2
make[3]: *** [Makefile:121: modules] Error 2
make[3]: Leaving directory '/media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[2]: *** [Makefile:584: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
make[2]: Leaving directory '/media/jlz/FIRMWARE/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#38.62#4.23#42.72
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/media/jlz/FIRMWARE/lede'
make: *** [/media/jlz/FIRMWARE/lede/include/toplevel.mk:230：package/mac80211/compile] 错误 2
请大神看看是怎么回事？

malash, 2022-03-14:
修改 `build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c` 第1000到1002行。

从

```c
    mac_addr = of_get_mac_address(rt2x00dev->dev->of_node);
    if (!IS_ERR(mac_addr))
        ether_addr_copy(eeprom_mac_addr, mac_addr);
```

改为

```c
    mac_addr = of_get_mac_address(rt2x00dev->dev->of_node, eeprom_mac_addr);
```

cc @coolsnowwolf 建议更新下 [master分支的patch](https://github.com/coolsnowwolf/lede/blob/c9da824ac8029eb0b7d0bca836532667acda1ca6/package/kernel/mac80211/patches/rt2x00/607-rt2x00-add_platform_data_mac_addr.patch) 哈，谢谢～

jlzsss, 2022-03-14:
谢谢

NeedforGit, 2022-03-15:
> 修改 `build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c` 第1000到1002行。
>
> 从
>
> ```c
>     mac_addr = [...]

我用云编译也是一直出现这个错误，请问云编译的话要改什么？ 感谢！

jewforice-openwrt, 2022-03-20:
云编译报错+1，求大佬更新下

closed 2022-03-29 by AmadeusGhost