coolsnowwolf/lede #9068: Bug report

micron10, 2022-03-14:
Hi 
after build fast-classiffier and enable qdisc in kernel when try to login on machine server crash with this bug report : 

Mar 14 07:14:06  [   18.759785][ T1003] BUG: unable to handle page fault for address: 00000001000002c9
Mar 14 07:14:06  [   18.769469][ T1003] #PF: supervisor read access in kernel mode
Mar 14 07:14:06  [   18.769469][ T1003] #PF: error_code(0x0000) - not-present page
Mar 14 07:14:06  [   18.769469][ T1003] PGD 0 P4D 0
Mar 14 07:14:06  [   18.769469][ T1003] Oops: 0000 [#1] SMP NOPTI
Mar 14 07:14:06  [   18.769469][ T1003] CPU: 1 PID: 1003 Comm: sshd Tainted: G           O      5.16.14 #1
Mar 14 07:14:06  [   18.769469][ T1003] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.14.0-0-g155821a1990b-prebuilt.qemu.org 04/01/2014
Mar 14 07:14:06  [   18.769469][ T1003] RIP: 0010:fast_classifier_post_routing+0x1a7/0xc00 [fast_classifier]
Mar 14 07:14:06  [   18.769469][ T1003] Code: 00 00 00 81 ff e0 00 00 00 0f 84 02 04 00 00 44 89 74 24 5c 44 89 54 24 7c 41 0f b7 b4 24 ac 00 00 00 49 8b bc 24 b8 00 00 00 <0f> b6 74 37 01 40 c0 ee 02 0f 85 f6 02 00 00 3c 06 0f 84 10 03 00
Mar 14 07:14:06  [   18.769469][ T1003] RSP: 0018:ffffbb8a810b79b8 EFLAGS: 00010287
Mar 14 07:14:06  [   18.769469][ T1003] RAX: 0000000000000006 RBX: ffff9d112247f900 RCX: 0000000000001600
Mar 14 07:14:06  [   18.769469][ T1003] RDX: 0000000000001600 RSI: 0000000000000008 RDI: 00000001000002c0
Mar 14 07:14:06  [   18.769469][ T1003] RBP: ffffbb8a810b7b58 R08: 000000000000feed R09: 000000000000feed
Mar 14 07:14:06  [   18.769469][ T1003] R10: 0000000046208e4e R11: ffffbb8a810b79e0 R12: ffff9d110301ece0
Mar 14 07:14:06  [   18.769469][ T1003] R13: 0000000000000001 R14: 0000000014edfeb2 R15: 0000000000000006
Mar 14 07:14:07  [   18.769469][ T1003] FS:  00007f2bc0835740(0000) GS:ffff9d1237c80000(0000) knlGS:0000000000000000
Mar 14 07:14:07  [   18.769469][ T1003] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar 14 07:14:07  [   18.769469][ T1003] CR2: 00000001000002c9 CR3: 0000000122472000 CR4: 00000000001506e0
Mar 14 07:14:07  [   18.769469][ T1003] Call Trace:
Mar 14 07:14:07  [   18.769469][ T1003]  <TASK>
Mar 14 07:14:07  [   18.769469][ T1003]  ? __flush_work.isra.0+0x30/0x200
Mar 14 07:14:07  [   18.769469][ T1003]  nf_hook_slow+0x36/0xa0
Mar 14 07:14:07  [   18.769469][ T1003]  ip_output+0xb0/0xd0
Mar 14 07:14:07  [   18.769469][ T1003]  ? ip_output+0xd0/0xd0
Mar 14 07:14:07  [   18.769469][ T1003]  __ip_queue_xmit+0x188/0x400
Mar 14 07:14:07  [   18.769469][ T1003]  __tcp_transmit_skb+0x4a3/0x8a0
Mar 14 07:14:07  [   18.769469][ T1003]  tcp_write_xmit+0x3a7/0xa60
Mar 14 07:14:07  [   18.769469][ T1003]  __tcp_push_pending_frames+0x2d/0xb0
Mar 14 07:14:07  [   18.769469][ T1003]  tcp_sendmsg_locked+0x299/0xb40
Mar 14 07:14:07  [   18.769469][ T1003]  tcp_sendmsg+0x23/0x40
Mar 14 07:14:07  [   18.769469][ T1003]  sock_write_iter+0xb2/0x100
Mar 14 07:14:07  [   18.769469][ T1003]  ? intel_pmu_store_lbr+0x20/0x240
Mar 14 07:14:07  [   18.769469][ T1003]  new_sync_write+0x164/0x170
Mar 14 07:14:07  [   18.769469][ T1003]  ? intel_pmu_store_lbr+0x20/0x240
Mar 14 07:14:07  [   18.769469][ T1003]  vfs_write+0x1bf/0x260
Mar 14 07:14:07  [   18.769469][ T1003]  ksys_write+0x82/0xc0
Mar 14 07:14:07  [   18.769469][ T1003]  do_syscall_64+0x35/0x80
Mar 14 07:14:07  [   18.769469][ T1003]  entry_SYSCALL_64_after_hwframe+0x44/0xae
Mar 14 07:14:07  [   18.769469][ T1003] RIP: 0033:0x7f2bc0945c93
Mar 14 07:14:07  [   18.769469][ T1003] Code: 8b 15 99 d1 0f 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 64 8b 04 25 18 00 00 00 85 c0 75 14 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 55 c3 0f 1f 40 00 48 83 ec 28 48 89 54 24 18
Mar 14 07:14:07  [   18.769469][ T1003] RSP: 002b:00007ffd4f3fedd8 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
Mar 14 07:14:07  [   18.769469][ T1003] RAX: ffffffffffffffda RBX: 000000000000006c RCX: 00007f2bc0945c93
Mar 14 07:14:07  [   18.769469][ T1003] RDX: 000000000000006c RSI: 00005600817846b0 RDI: 0000000000000004
Mar 14 07:14:07  [   18.769469][ T1003] RBP: 0000560081789a90 R08: 0000000000000000 R09: 0000000000000000
Mar 14 07:14:07  [   18.769469][ T1003] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
Mar 14 07:14:07  [   19.348300][ T1003] R13: 00007ffd4f3fee60 R14: 00007ffd4f3feee0 R15: 0000000000000001
Mar 14 07:14:07  [   19.366341][ T1003]  </TASK>
Mar 14 07:14:07  [   19.375493][ T1003] Modules linked in: xt_nat iptable_nat xt_TCPMSS xt_comment iptable_mangle ip_tables netconsole e1000 ixgbe mdio mdio_devres libphy i40e nf_nat_sip nf_conntrack_sip nf_nat_pptp nf_conntrack_pptp nf_nat_tftp nf_conntrack_tftp nf_nat_ftp nf_conntrack_ftp nf_nat nf_xnatlog(O) fast_classifier(O) shortcut_fe_cm(O) nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 shortcut_fe_ipv6(O) shortcut_fe(O) ipmi_devintf ipmi_msghandler rtc_cmos§
Mar 14 07:14:07  [   19.447723][ T1003] CR2: 00000001000002c9
Mar 14 07:14:07  [   19.458901][ T1003] ---[ end trace f4a988c1ddecbc6b ]---
Mar 14 07:14:07  [   19.470129][ T1003] RIP: 0010:fast_classifier_post_routing+0x1a7/0xc00 [fast_classifier]
Mar 14 07:14:07  [   19.493053][ T1003] Code: 00 00 00 81 ff e0 00 00 00 0f 84 02 04 00 00 44 89 74 24 5c 44 89 54 24 7c 41 0f b7 b4 24 ac 00 00 00 49 8b bc 24 b8 00 00 00 <0f> b6 74 37 01 40 c0 ee 02 0f 85 f6 02 00 00 3c 06 0f 84 10 03 00
Mar 14 07:14:07  [   19.529010][ T1003] RSP: 0018:ffffbb8a810b79b8 EFLAGS: 00010287
Mar 14 07:14:07  [   19.541021][ T1003] RAX: 0000000000000006 RBX: ffff9d112247f900 RCX: 0000000000001600
Mar 14 07:14:07  [   19.564919][ T1003] RDX: 0000000000001600 RSI: 0000000000000008 RDI: 00000001000002c0
Mar 14 07:14:07  [   19.588715][ T1003] RBP: ffffbb8a810b7b58 R08: 000000000000feed R09: 000000000000feed
Mar 14 07:14:07  [   19.612498][ T1003] R10: 0000000046208e4e R11: ffffbb8a810b79e0 R12: ffff9d110301ece0
Mar 14 07:14:07  [   19.636324][ T1003] R13: 0000000000000001 R14: 0000000014edfeb2 R15: 0000000000000006
Mar 14 07:14:07  [   19.659996][ T1003] FS:  00007f2bc0835740(0000) GS:ffff9d1237c80000(0000) knlGS:0000000000000000
Mar 14 07:14:07  [   19.683747][ T1003] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar 14 07:14:07  [   19.695720][ T1003] CR2: 00000001000002c9 CR3: 0000000122472000 CR4: 00000000001506e0
Mar 14 07:14:07  [   19.719804][ T1003] Kernel panic - not syncing: Fatal exception
Mar 14 07:14:07  [   19.731780][ T1003] Kernel Offset: 0x1b000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
Mar 14 07:14:07  [   19.731780][ T1003] Rebooting in 10 seconds..

micron10, 2022-03-15:
@coolsnowwolf 
Hi fix this after clean rebuild of source all is fine.
but need your help

after enable fast-classifier module
if customer use  NAT ip shaper (qdisc) is work in both direction 300/300 down/up
but if user use Real ip (Not NAT ip) with same qdisc rulls not work for upload , download is fine. -  300/700 down/up

May be fast-classifier check only NAT rulls and if user use Real ip fast classifer miss  any check.

If you have time can you check it. and possibly fix it.

m

micron10, 2022-03-16:
Hi @coolsnowwolf 
One more is there options to sync code with : https://source.codeaurora.org/quic/qsdk/oss/lklm/shortcut-fe/tree/?h=NHSS.QSDK.12.1.r2

to add PPPoe support

coolsnowwolf, 2026-09-05:
fast-classifier 在 Linux 5.16 上的 post-routing 内核崩溃是旧厂商加速模块缺陷，报告没有可安全移植的上游修复；关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf