coolsnowwolf/lede #9073: ARRIS SBR-AC1750固件编译成功，WIFI正常，LAN口无法上网但可PING通外网，DNS解析正常

zbluekuma, 2022-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

**Q：是否搜索了issue (使用 "x" 选择)**
* [ X] 没有类似的issue

**## 2. 详细叙述**

A：小众设备，原来第三方固件内核太老，网上也没什么好用的固件。自己按大神的教程，学着编译了固件。
主要是自用，也在论坛分享给需要的网友。
按您的编译教程，使用menuconfig默认设置，成功编译了ARRIS SBR-AC1750固件。
路由器通过mtd write 命令，刷入kernel1.bin kernel，刷入rootfs1.bin ubi，刷机成功。

**### (1) 具体问题**

A：刷机后重启路由器，WIFI无线正常使用，但是通过LAN口有线连接的设备，设备无法正常上网。
主要是浏览器不能浏览网页，系统显示有线连接成功，网络诊断正常。原本登录的QQ，微信可以发送文字信息。
通过ping命令ping IP或者域名都正常

### (2) 路由器型号和固件版本

A：路由器型号 Arris SBR-AC1750 (CpuMark : 1858.581322 Scores)  
      固件版本OpenWrt R22.3.13 / LuCI Master (git-22.068.45502-a50e601) （注：上一个版本同样问题）

### (3) 详细日志

A：[    0.000000] Linux version 5.4.175 (cnz@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4325-7586b4b30)) #0 Mon Mar 14 06:45:28 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is Arris SBR-AC1750
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 251176K/262144K available (5562K kernel code, 174K rwdata, 1204K rodata, 1240K init, 201K bss, 10968K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from start_kernel+0x358/0x54c with crng_init=0
[    0.000000] CPU clock: 720.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000007] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008291] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.074976] pid_max: default: 32768 minimum: 301
[    0.080025] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087800] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.100049] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.110543] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.117871] pinctrl core: initialized pinctrl subsystem
[    0.124270] NET: Registered protocol family 16
[    0.157900] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[... 244 lines cut ...]
[  122.223652] ath: regdomain 0x8348 dynamically updated by user
[  127.425985] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[  127.443301] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[  127.449610] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[  127.457786] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 24984 iram: 38672'
[  127.796105] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[  127.821355] ath10k_pci 0000:00:00.0: rts threshold -1
[  127.868161] br-lan: port 2(wlan0) entered blocking state
[  127.873608] br-lan: port 2(wlan0) entered disabled state
[  127.879261] device wlan0 entered promiscuous mode
[  128.406247] br-lan: port 3(wlan1) entered blocking state
[  128.411819] br-lan: port 3(wlan1) entered disabled state
[  128.417499] device wlan1 entered promiscuous mode
[  128.422527] br-lan: port 3(wlan1) entered blocking state
[  128.427949] br-lan: port 3(wlan1) entered forwarding state
[  129.017018] br-lan: port 3(wlan1) entered disabled state
[  129.580329] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  129.587089] br-lan: port 2(wlan0) entered blocking state
[  129.592531] br-lan: port 2(wlan0) entered forwarding state
[  138.183491] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  138.190531] br-lan: port 3(wlan1) entered blocking state
[  138.195990] br-lan: port 3(wlan1) entered forwarding state
[  200.524216] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  200.532413] NFSD: Using legacy client tracking operations.
[  200.537984] NFSD: starting 10-second grace period (net f0000029)

coolsnowwolf, 2022-03-25:
看起来是 MDIO 注册有线的问题

zbluekuma, 2022-04-03:
谢谢Lean大，今天版本的固件编译，LAN口上网已正常。issue关闭。

zbluekuma, 2022-04-06:
5.4.188编辑失败日志
Collecting package info: done
make[1]: Entering directory '/home/cnz/lede'
make[2]: Entering directory '/home/cnz/lede'
make[3]: Entering directory '/home/cnz/lede/target/linux'
make[4]: Entering directory '/home/cnz/lede/target/linux/ath79'
rm -f /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/localversion
/home/cnz/lede/scripts/kconfig.pl  + + /home/cnz/lede/target/linux/generic/config-5.4 /home/cnz/lede/target/linux/ath79/config-5.4 /home/cnz/lede/target/linux/ath79/nand/config-default > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/cnz/lede/.config >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
/home/cnz/lede/scripts/package-metadata.pl kconfig /home/cnz/lede/tmp/.packageinfo /home/cnz/lede/.config 5.4 > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.override
/home/cnz/lede/scripts/kconfig.pl 'm+' '+' /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target /dev/null /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.override > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
mv /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.old
grep -v INITRAMFS /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.old > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
rm -rf /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/modules
cmp -s /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.prev || { cp /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config; cp /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.prev; }
export MAKEFLAGS= ; [ -d /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/user_headers ] || make -C /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188 KCFLAGS="-fmacro-prefix-map=/home/cnz/lede/build_dir/target-mips_24kc_musl=target-mips_24kc_musl" HOSTCFLAGS="-O2 -I/home/cnz/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr  4 16:58:43 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/cnz/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/cnz/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.188 CC="mips-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/user_headers headers_install
grep '=[ym]' /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set | LC_ALL=C sort | /home/cnz/lede/staging_dir/host/bin/mkhash md5 > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.vermagic
touch /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.configured
rm -f /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/vmlinux /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/System.map
make -C /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188 KCFLAGS="-fmacro-prefix-map=/home/cnz/lede/build_dir/target-mips_24kc_musl=target-mips_24kc_musl" HOSTCFLAGS="-O2 -I/home/cnz/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr  4 16:58:43 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/cnz/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/cnz/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.188 CC="mips-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188'
scripts/kconfig/conf  --olddefconfig Kconfig
net/sched/Kconfig:45: warning: menuconfig statement without prompt
[... cut ...]
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/cnz/lede/target/linux'
time: target/linux/compile#8.14#4.79#12.92
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/cnz/lede'
make[1]: *** [target/Makefile:18: /home/cnz/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/cnz/lede'
make: *** [/home/cnz/lede/include/toplevel.mk:230: world] Error 2

zbluekuma, 2022-04-06:
5.15.32内核，内核日志提示NAND分区信息如下：
[    0.450495] 7 fixed-partitions partitions found on MTD device ar934x-nand
[    0.457391] Creating 7 MTD partitions on "ar934x-nand":
[    0.462720] 0x000000000000-0x000000100000 : "u-boot"
[    0.468081] mtdblock: MTD device 'u-boot' is NAND, please consider using UBI block devices instead.
[    0.478540] 0x000000100000-0x000000200000 : "boot-flag"
[    0.484165] mtdblock: MTD device 'boot-flag' is NAND, please consider using UBI block devices instead.
[    0.494818] 0x000000200000-0x000000600000 : "kernel"
[    0.500245] mtdblock: MTD device 'kernel' is NAND, please consider using UBI block devices instead.
[    0.510632] 0x000000600000-0x000006a00000 : "ubi"
[    0.516137] mtdblock: MTD device 'ubi' is NAND, please consider using UBI block devices instead.
[    0.526231] 0x000006a00000-0x000006b00000 : "ft"
[    0.531343] mtdblock: MTD device 'ft' is NAND, please consider using UBI block devices instead.
[    0.541412] 0x000006b00000-0x000006d00000 : "PKI"
[    0.546508] mtdblock: MTD device 'PKI' is NAND, please consider using UBI block devices instead.
[    0.556611] 0x000006d00000-0x000006e00000 : "caldata"
[    0.562098] mtdblock: MTD device 'caldata' is NAND, please consider using UBI block devices instead.
[    0.923373] switch0: Atheros AR8337 rev. 2 switch registered on mdio.0
[    1.577661] ag71xx 19000000.eth: connected to PHY at mdio.0:00 [uid=004dd036, driver=Atheros AR8216/AR8236/AR8316]
[    1.588763] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: rgmii
[    1.930149] ag71xx 1a000000.eth: connected to PHY at fixed-0:00 [uid=00000000, driver=Generic PHY]
[    1.939827] eth1: Atheros AG71xx at 0xba000000, irq 5, mode: sgmii
[    1.946520] ath79-wdt 18060008.wdt: unable to register misc device, err=-16
[    1.953652] ath79-wdt: probe of 18060008.wdt failed with error -16
[    1.961520] NET: Registered PF_INET6 protocol family
[    1.972655] Segment Routing with IPv6
[    1.976419] In-situ OAM (IOAM) with IPv6
[    1.980558] NET: Registered PF_PACKET protocol family

上面日志提示两个问题：
1、[    0.468081] mtdblock: MTD device 'u-boot' is NAND, please consider using UBI block devices instead.
2、[    1.946520] ath79-wdt 18060008.wdt: unable to register misc device, err=-16
还请大神们给予帮助，谢谢。

closed 2022-05-14 by zbluekuma