coolsnowwolf/lede #9074: op云编译files大法文件权限求助

smdx, 2022-03-16:
各位大佬好，在github op云编译项目根目录建立了files/sbin/文件夹放置了ethinfo文件，编译完刷了固件首页显示信息不正常发现文件权限不对为644改成755重启后就正常了。请问各位大佬如何在云编译时修改这个文件的权限为755？谢谢

[image]
[image]

复刻自 jingleijack/X86_64-TEST的项目，P3TERX/Actions-OpenWrt的模板。

WYC-2020, 2022-03-16:
还真是纯小白，你自己放置的ethinfo文件 本身权限就不对，自己改了，放好不就行了

Dawneng, 2022-03-16:
在云编译工作流配置中找到操作files文件夹的地方加一句赋权指令不就行了？

smdx, 2022-03-16:
> 还真是纯小白，你自己放置的ethinfo文件 本身权限就不对，自己改了，放好不就行了

哎..本身就是对编程完全不懂啊....问题是在windows怎么修改权限？ethinfo文件在win修改好内容后上传权限就这样了。

smdx, 2022-03-16:
> 在云编译工作流配置中找到操作files文件夹的地方加一句赋权指令不就行了？

大佬，这个是具体流程配置，可以帮忙看看吗？谢谢

```
#
# Copyright (c) 2019-2020 P3TERX <https://p3terx.com>
#
# This is free software, licensed under the MIT License.
# See /LICENSE for more information.
#
# https://github.com/P3TERX/Actions-OpenWrt
# Description: Build OpenWrt using GitHub Actions
#

name: Build x86_64-Use

on:
  repository_dispatch:
  workflow_dispatch:
    inputs:
      ssh:
        description: 'SSH connection to Actions'
        required: false
        default: 'false'

env:
  REPO_URL: https://github.com/coolsnowwolf/lede
  REPO_BRANCH: master
  FEEDS_CONF: feeds.conf.default
  CONFIG_FILE: lean-x86_64-Use.config
  DIY_P1_SH: diy-part1.sh
  DIY_P2_SH: diy-part2.sh
  UPLOAD_BIN_DIR: false
  UPLOAD_FIRMWARE: true
  UPLOAD_COWTRANSFER: false
  UPLOAD_WETRANSFER: false
  UPLOAD_RELEASE: true
  TZ: Asia/Shanghai

jobs:
  build:
    runs-on: ubuntu-20.04

    steps:
    - name: Checkout
      uses: actions/checkout@main

    - name: Initialization environment
      env:
        DEBIAN_FRONTEND: noninteractive
      run: |
        sudo rm -rf /etc/apt/sources.list.d/* /usr/share/dotnet /usr/local/lib/android /opt/ghc
        sudo -E apt-get -qq update
        sudo -E apt-get -qq install $(curl -fsSL git.io/depends-ubuntu-2004)
        sudo -E apt-get -qq autoremove --purge
        sudo -E apt-get -qq clean
        sudo timedatectl set-timezone "$TZ"
        sudo mkdir -p /workdir
        sudo chown $USER:$GROUPS /workdir

    - name: Clone source code
      working-directory: /workdir
      run: |
        df -hT $PWD
        git clone $REPO_URL -b $REPO_BRANCH openwrt
        ln -sf /workdir/openwrt $GITHUB_WORKSPACE/openwrt

    - name: Load custom feeds
      run: |
[... 101 lines cut ...]
    - name: Upload firmware to release
      uses: softprops/action-gh-release@v1
      if: steps.tag.outputs.status == 'success' && !cancelled()
      env:
        GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
      with:
        tag_name: ${{ steps.tag.outputs.release_tag }}
        body_path: release.txt
        files: ${{ env.FIRMWARE }}/*

    - name: Delete workflow runs
      uses: GitRML/delete-workflow-runs@main
      with:
        retain_days: 1
        keep_minimum_runs: 1

    - name: Remove old Releases
      uses: dev-drprasad/delete-older-releases@v0.1.0
      if: env.UPLOAD_RELEASE == 'true' && !cancelled()
      with:
        keep_latest: 5
        delete_tags: true
      env:
        GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
```

Dawneng, 2022-03-16:
> 

应该是这样吧  
    - name: Load custom configuration
      run: |
        [ -e files ] && mv files openwrt/files
        [ -e $CONFIG_FILE ] && mv $CONFIG_FILE openwrt/.config
        chmod +x $DIY_P2_SH
        cd openwrt
        chmod -R 755 files
        $GITHUB_WORKSPACE/$DIY_P2_SH

smdx, 2022-03-16:
> > 
> 
> 应该是这样吧 - name: Load custom configuration run: | [ -e files ] && mv files openwrt/files [ -e $CONFIG_FILE ] && mv $CONFIG_FILE openwrt/.config chmod +x $DIY_P2_SH cd openwrt chmod -R 755 files $GITHUB_WORKSPACE/$DIY_P2_SH

非常感谢，我试试。

closed 2022-03-16 by smdx