coolsnowwolf/lede #9075: ERROR: package/kernel/mac80211 failed to build (build variant: regular).

dzhl, 2022-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 查找过，但是没看懂怎么解决

## 2. 详细叙述

### (1) 具体问题

A：使用同一个config文件，年前可以正常编译，本月想重新编译一版，就编译失败，查了issue，也没看懂怎么解决，感觉好像跟内核5.15有关，我把内核改为5.4，也没有成功。有人说重置config，是删除config，再重新生成吗？如果这样操作，那怎么保留以前的选择呢？折腾了许久，也没搞定，只好厚着脸皮来请教大家

### (2) 路由器型号和固件版本

A：
X86内核
### (3) 详细日志

A：
```
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.wolfssl-5.1.1-stable.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "wolfssl-5.1.1-stable.tar.gz" "d3e0544dbe7e9587c0f6538cdc671b6492663bb7a4281819538abe6c99cdbd92" "" "https://github.com/wolfSSL/wolfssl/archive/v5.1.1-stable"    '
make[3]: Leaving directory '/workdir/openwrt/package/libs/wolfssl'
time: package/libs/wolfssl/compile#0.26#0.11#0.31
make[3]: Entering directory '/workdir/openwrt/package/network/services/hostapd'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.hostapd-2021-05-22-b102f19b.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "hostapd-2021-05-22-b102f19b.tar.xz" "cb3cb968883042fc582752be1607586696c18e6ecf9808c9a8ac50e204584367" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf hostapd-2021-05-22-b102f19b && [ \! -d hostapd-2021-05-22-b102f19b ] && git clone  http://w1.fi/hostap.git hostapd-2021-05-22-b102f19b && (cd hostapd-2021-05-22-b102f19b && git checkout b102f19bcc53c7f7db3951424d4d46709b4f1986 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd hostapd-2021-05-22-b102f19b && git log -1 --format='\''@%ct'\''` && rm -rf hostapd-2021-05-22-b102f19b/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c hostapd-2021-05-22-b102f19b | 	xz -zc -7e > /workdir/openwrt/tmp/dl/hostapd-2021-05-22-b102f19b.tar.xz && mv /workdir/openwrt/tmp/dl/hostapd-2021-05-22-b102f19b.tar.xz /workdir/openwrt/dl/ && rm -rf hostapd-2021-05-22-b102f19b;  )    '
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/hostapd-wpad-full-internal/hostapd-2021-05-22-b102f19b/.pkgdir/hostapd-common/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.hostapd-common_installed
echo "hostapd-common" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/hostapd.wpad-full-internal.install
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/hostapd-wpad-full-internal/hostapd-2021-05-22-b102f19b/.pkgdir/wpad/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.wpad_installed
echo "wpad" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/hostapd.wpad-full-internal.install
make[3]: Leaving directory '/workdir/openwrt/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-internal/compile#0.37#0.12#0.45
make[3]: Entering directory '/workdir/openwrt/package/network/utils/iw'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.iw-5.9-8fab0c9e.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "iw-5.9-8fab0c9e.tar.xz" "797b322bc03952f3127ae0a7da476c14ada1bbe9a9ae234a56dd6f864c568e16" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf iw-5.9-8fab0c9e && [ \! -d iw-5.9-8fab0c9e ] && git clone  https://git.kernel.org/pub/scm/linux/kernel/git/jberg/iw.git iw-5.9-8fab0c9e && (cd iw-5.9-8fab0c9e && git checkout 8fab0c9ee9db217587a58efcc37421c86edcb638 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd iw-5.9-8fab0c9e && git log -1 --format='\''@%ct'\''` && rm -rf iw-5.9-8fab0c9e/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c iw-5.9-8fab0c9e | 	xz -zc -7e > /workdir/openwrt/tmp/dl/iw-5.9-8fab0c9e.tar.xz && mv /workdir/openwrt/tmp/dl/iw-5.9-8fab0c9e.tar.xz /workdir/openwrt/dl/ && rm -rf iw-5.9-8fab0c9e;  )    '
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/iw-tiny/iw-5.9-8fab0c9e/.pkgdir/iw/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.iw_installed
echo "iw" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/iw.tiny.install
make[3]: Leaving directory '/workdir/openwrt/package/network/utils/iw'
time: package/network/utils/iw/tiny/compile#0.28#0.07#0.33
make[3]: Entering directory '/workdir/openwrt/package/kernel/mac80211'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.backports-5.10.42-1.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "backports-5.10.42-1.tar.xz" "6876520105240844fdb32d1dcdf2bfdea291a37a96f16c892fda3776ba714fcb" "" "@KERNEL/linux/kernel/projects/backports/stable/v5.10.42/"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built_check
. /workdir/openwrt/include/shell.sh; var2file "V_mac80211_config" /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.config
make -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Mar 15 17:50:37 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/fast-classifier.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/oaf.symvers" KERNELRELEASE=5.15.28 EXTRA_CFLAGS="-I/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1=backports-5.10.42-1  -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include/qca-nss-drv" KLIB_BUILD="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.28" MODPROBE=true KLIB=/lib/modules/5.15.28 KERNEL_SUBLEVEL=15 KBUILD_LDFLAGS_MODULE_PREREQ= allnoconfig
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[5]: 'Kconfig.versions' is up to date.
make[6]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/modules
make  -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/h [... cut ...]
time: package/kernel/mac80211/regular/compile#1.25#0.31#1.50
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.
```

WYC-2020, 2022-03-16:
这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl

breakings, 2022-03-16:
> 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl

好家伙，今天的701补丁也炸了。

WYC-2020, 2022-03-16:
> > 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl
> 
> 好家伙，今天的701补丁也炸了。

那不清楚哦 我还没更新

WYC-2020, 2022-03-16:
> > 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl
> 
> 好家伙，今天的701补丁也炸了。

具体是哪个补丁 发出来 有大佬看到了 会修复

breakings, 2022-03-16:
> > > 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl
> > 
> > 
> > 好家伙，今天的701补丁也炸了。
> 
> 具体是哪个补丁 发出来 有大佬看到了 会修复

https://github.com/coolsnowwolf/lede/commit/a5d4d89136fd9078ceee6c378f84c65351c95197

WYC-2020, 2022-03-16:
> > > > 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl
> > > 
> > > 
> > > 好家伙，今天的701补丁也炸了。
> > 
> > 
> > 具体是哪个补丁 发出来 有大佬看到了 会修复
> 
> [a5d4d89](https://github.com/coolsnowwolf/lede/commit/a5d4d89136fd9078ceee6c378f84c65351c95197)

这补丁会失败？等会我看下

WYC-2020, 2022-03-16:
> > > > 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl
> > > 
> > > 
> > > 好家伙，今天的701补丁也炸了。
> > 
> > 
> > 具体是哪个补丁 发出来 有大佬看到了 会修复
> 
> [a5d4d89](https://github.com/coolsnowwolf/lede/commit/a5d4d89136fd9078ceee6c378f84c65351c95197)

https://github.com/WYC-2020/lede/commit/19c98781da559d1ad333ec5085333b8fa8f4ec9b
修复

WYC-2020, 2022-03-16:
> > > > 这个问题我以前就修复了，l大这边没修复，自己去我仓库拉去651补丁 @dzhl
> > > 
> > > 
> > > 好家伙，今天的701补丁也炸了。
> > 
> > 
> > 具体是哪个补丁 发出来 有大佬看到了 会修复
> 
> [a5d4d89](https://github.com/coolsnowwolf/lede/commit/a5d4d89136fd9078ceee6c378f84c65351c95197)

好家伙看了这个补丁 和另外一个补丁很多是一样的，并且被包含再这个补丁（610-v5.13-54-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support），直接revert了，不需要这个补丁

closed 2022-03-25 by coolsnowwolf