coolsnowwolf/lede #9076: 究竟要如何开启ipv6？光猫桥接

HanzChen, 2022-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：按网上的教程开启IPV6，wan6是获取到了240e开头的地址，但是lan口和里面的设备都没有获取到。

### (2) 路由器型号和固件版本

A：x86，自行编译

### (3) 详细日志

光猫桥接模式，使用华硕ax86u用native是可以完美获取ipv6地址的。
但是用软路由就死活不行
设置如下

[image]
[image]
[image]
[image]
[image]
[image]

WYC-2020, 2022-03-16:
软路由自身拨号，wan6 口没用吧，wan 就能获取到ipv6

frely, 2022-03-16:
Base system  ---> 
    Build with DHCPv6 support.
Extra packages  ---> 
    ipv6helper

HanzChen, 2022-03-16:
> Base system ---> Build with DHCPv6 support. Extra packages ---> ipv6helper

Already done this. The problem here is that WAN6 shows a valid IPV6, but all LAN devices have no IPV6...
已经这么做了，现在的主要是lan设备无法获取有效的ipv6...

WYC-2020, 2022-03-16:
> > Base system ---> Build with DHCPv6 support. Extra packages ---> ipv6helper
> 
> Already done this. The problem here is that WAN6 shows a valid IPV6, but all LAN devices have no IPV6... 已经这么做了，现在的主要是lan设备无法获取有效的ipv6...

确定你的宽带运营商提供了ipv6 支持

frely, 2022-03-16:
> > 基础系统 ---> 使用 DHCPv6 支持构建。额外的软件包---> ipv6helper
> 
> 已经这样做了。这里的问题是，WAN6显示了有效的IPV6，但是所有的LAN设备都没有IPV6…… 已经完成了，现在的主要是设备无法获取有效的ipv6……

openwrt的默认配置不要动，获取不到ipv6的话检车下光猫的ipv6配置。

HanzChen, 2022-03-16:
> > > Base system ---> Build with DHCPv6 support. Extra packages ---> ipv6helper
> > 
> > 
> > Already done this. The problem here is that WAN6 shows a valid IPV6, but all LAN devices have no IPV6... 已经这么做了，现在的主要是lan设备无法获取有效的ipv6...
> 
> 确定你的宽带运营商提供了ipv6 支持

确定的，使用华硕路由器是完美支持IPV6，所有子设备都能正确获取。

HanzChen, 2022-03-16:
> 使用华硕路由器是完美支持IPV6，所有子设备都能正确获取。

所以其实不需要“DHCPv6 ”和“ ipv6helper”？

WYC-2020, 2022-03-16:
> > 使用华硕路由器是完美支持IPV6，所有子设备都能正确获取。
> 
> 所以其实不需要“DHCPv6 ”和“ ipv6helper”？

https://www.youtube.com/watch?v=aVv64eXk50s 自己参考一下 这个ytb 很多教程的，我家里运营商不支持 所以就没搞ipv6

WYC-2020, 2022-03-16:
> > 使用华硕路由器是完美支持IPV6，所有子设备都能正确获取。
> 
> 所以其实不需要“DHCPv6 ”和“ ipv6helper”？

必须要要的

frely, 2022-03-16:
昨天最新编译的固件，使用默认配置拨号后lan设备就有ipv6了。
Base system ---> Build with DHCPv6 support.
Extra packages ---> ipv6helper

[image]

HanzChen, 2022-03-16:
> 昨天最新编译的固件，使用默认配置拨号后lan设备就有ipv6了。 Base system ---> Build with DHCPv6 support. Extra packages ---> ipv6helper
> 
> [image]

好的我再试试看

breakings, 2022-03-16:
我是光猫桥接，硬路由拨号。
[image]

HanzChen, 2022-03-16:
重新编译后解决问题。。
Base system ---> Build with DHCPv6 support. 
Extra packages ---> ipv6helper
[image]

closed 2022-03-16 by HanzChen