coolsnowwolf/lede #9077: 本地编译总出这个错误

qqsir-dev, 2022-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
Configuring luci-i18n-base-en.
Configuring kmod-button-hotplug.
Configuring luci-i18n-bonding-en.
Configuring kmod-igb.
Configuring kmod-igc.
Configuring kmod-bnx2.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/ipsec
        But that file is already provided by package  * strongswan-ipsec
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/controller/ipsec-server.lua
        But that file is already provided by package  * luci-app-ipsec-server
 * opkg_install_cmd: Cannot install package luci-app-ipsec-vpnd.
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/ipsec
        But that file is already provided by package  * strongswan-ipsec
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/controller/ipsec-server.lua
        But that file is already provided by package  * luci-app-ipsec-server
 * opkg_install_cmd: Cannot install package luci-i18n-ipsec-vpnd-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/user/openwrt'
make[1]: *** [package/Makefile:111: /home/user/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:230: world] Error 2

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

WYC-2020, 2022-03-16:
这个别人说过，只编译一个，我仓库是删除了vpnd，同时把sever移动到了luci，并修复一个bug，你可以用我那个luci @qqsir-dev

qqsir-dev, 2022-03-16:
好的，多谢

closed 2022-03-16 by qqsir-dev