coolsnowwolf/lede #9082: 5.10编译报错X86

rongcenter, 2022-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

5.10编译报错X86

### (2) 路由器型号和固件版本

X86

### (3) 详细日志

Applying /home/xr/lede/target/linux/generic/pending-5.10/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch using plaintext:
patching file drivers/net/ethernet/mediatek/mtk_ppe_offload.c
Hunk #1 FAILED at 7.
Hunk #2 FAILED at 20.
Hunk #3 succeeded at 78 with fuzz 1 (offset 14 lines).
Hunk #4 succeeded at 302 with fuzz 2 (offset 40 lines).
Hunk #5 succeeded at 351 with fuzz 1 (offset 51 lines).
2 out of 5 hunks FAILED -- saving rejects to file drivers/net/ethernet/mediatek/mtk_ppe_offload.c.rej
Patch failed!  Please fix /home/xr/lede/target/linux/generic/pending-5.10/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch!
make[4]: *** [Makefile:35: /home/xr/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.104/.prepared_92d58548cc8c1e7f395767b1d696ff59] Error 1
make[4]: Leaving directory '/home/xr/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/xr/lede/target/linux'
time: target/linux/compile#11.56#27.46#36.75
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/xr/lede'
make[1]: *** [target/Makefile:18: /home/xr/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/xr/lede'
make: *** [/home/xr/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-17:
701补丁没必要要

WYC-2020, 2022-03-17:
@rongcenter https://github.com/coolsnowwolf/lede/issues/9075#issue-1170613868
看这个

liuyanantj, 2022-03-17:
大雕后续会修复的吧 @WYC-2020 我还是等着他更新吧  或者是顺着你提那个自己在本地改了？

WYC-2020, 2022-03-17:
> 大雕后续会修复的吧 @WYC-2020 我还是等着他更新吧 或者是顺着你提那个自己在本地改了？

那我就不清楚了，自己改肯定没问题，大雕啥时候更新我就不清楚了

cyzzc, 2022-03-17:
> 701补丁没必要要

大佬，您说的没必要要的意思是不是找到这个文件直接删除掉就行是吗，盼能得到回复！

WYC-2020, 2022-03-17:
> > 701补丁没必要要
> 
> 大佬，您说的没必要要的意思是不是找到这个文件直接删除掉就行是吗，盼能得到回复！

是的

WYC-2020, 2022-03-17:
> > 701补丁没必要要
> 
> 大佬，您说的没必要要的意思是不是找到这个文件直接删除掉就行是吗，盼能得到回复！

自己去我仓库看嘛也明白嘛

rongcenter, 2022-03-17:
> > > 701补丁没必要要
> > 
> > 
> > 大佬，您说的没必要要的意思是不是找到这个文件直接删除掉就行是吗，盼能得到回复！
> 
> 自己去我仓库看嘛也明白嘛

谢谢大佬

dsyo2008, 2022-03-17:
一样编译5.10报错，老实用默认5.15编译成功，暂时用5.15没啥问题

rongcenter, 2022-03-17:
> 一样编译5.10报错，老实用默认5.15编译成功，暂时用5.15没啥问题

5.15igb新驱动编译不过去，所以没用，用自带的驱动我是虚拟机intel pci卡又不行，很烦。不行到时候换换螃蟹卡

WYC-2020, 2022-03-18:
> > 一样编译5.10报错，老实用默认5.15编译成功，暂时用5.15没啥问题
> 
> 5.15igb新驱动编译不过去，所以没用，用自带的驱动我是虚拟机intel pci卡又不行，很烦。不行到时候换换螃蟹卡

你是用了直通不，你不用直通都不行吗

cyzzc, 2022-03-18:
> > > 一样编译5.10报错，老实用默认5.15编译成功，暂时用5.15没啥问题
> 
> > 
> 
> > 5.15igb新驱动编译不过去，所以没用，用自带的驱动我是虚拟机intel pci卡又不行，很烦。不行到时候换换螃蟹卡
> 
> 
> 
> 你是用了直通不，你不用直通都不行吗

- [ ] 一位大佬跟我说把701这个补丁删除掉就可以了，我也试过确实可以，要不你也试试。

closed 2022-03-23 by coolsnowwolf