coolsnowwolf/lede #9083: 最新版云编译X86报错

NeedforGit, 2022-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译报错

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

make[11]: *** [scripts/Makefile.build:277: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o] Error 1
make[10]: *** [scripts/Makefile.build:540: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00] Error 2
make[9]: *** [scripts/Makefile.build:540: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink] Error 2
make[8]: *** [scripts/Makefile.build:540: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
make[7]: *** [Makefile:1868: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[3]: *** [Makefile:584: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
time: package/kernel/mac80211/regular/compile#1.47#0.48#1.88
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

最近几天云编译X86一直提示这个错误，请问怎样解决？感谢！

WYC-2020, 2022-03-17:
不知道看下是否有类似的issue嘛，不是有一个和你这个一样嘛

dzhl, 2022-03-18:
这个问题能不能从根上去除啊，这么多人遇到这个问题，关键的是对于我们这些紧紧会编译不会代码的来改代码太难了啊

WYC-2020, 2022-03-18:
> 这个问题能不能从根上去除啊，这么多人遇到这个问题，关键的是对于我们这些紧紧会编译不会代码的来改代码太难了啊

早就有根本上去除的方法了 ，l大没更新到而已

dzhl, 2022-03-18:
> > 这个问题能不能从根上去除啊，这么多人遇到这个问题，关键的是对于我们这些紧紧会编译不会代码的来改代码太难了啊
> 
> 早就有根本上去除的方法了 ，l大没更新到而已

我意思是L大咋就不更新了呢

WYC-2020, 2022-03-18:
> > > 这个问题能不能从根上去除啊，这么多人遇到这个问题，关键的是对于我们这些紧紧会编译不会代码的来改代码太难了啊
> > 
> > 
> > 早就有根本上去除的方法了 ，l大没更新到而已
> 
> 我意思是L大咋就不更新了呢

自己去我仓库拉去，然后提取651补丁，放到对应目录就可以了(/package/kernel/mac80211/patches)/rt2x00/)，然后自己提pr

luxiaojiong, 2022-03-20:
> > > > 这个问题能不能从根上去除啊，这么多人遇到这个问题，关键的是对于我们这些紧紧会编译不会代码的来改代码太难了啊
> > > 
> > > 
> > > 早就有根本上去除的方法了 ，l大没更新到而已
> > 
> > 
> > 我意思是L大咋就不更新了呢
> 
> 自己去我仓库拉去，然后提取651补丁，放到对应目录就可以了(/package/kernel/mac80211/patches)/rt2x00/)，然后自己提pr

新手请教一个问题，如何提取651补丁呢，有没有网上教程，谢谢！

closed 2022-03-29 by AmadeusGhost