coolsnowwolf/lede #9087: package/utils/lua failed to build.

jonlubas, 2022-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
package/utils/lua failed to build.
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
下载DL出错 make -j8 download V=s  确定已经翻墙了，虽然是免费节点 但是curl -i google.com 反馈都正常 百度也搜索了 编译这个固件 我已经花费了10多天了，还没有成功。
### (1) 具体问题

A：package/utils/lua failed to build.

### (2) 路由器型号和固件版本

A：软路由

### (3) 详细日志

A：mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.libnl-tiny-2020-08-05-c291088f.tar.xz.flock -c '          /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "libnl-tiny-2020-08-05-c291088f.tar.xz" "99bcce12701bb34dadb39689d95c2c5cf1e27719d0ecfd645d3957a8947025ac" "" || (       echo "Checking out files from the git repository..."; mkdir -p /home/jonlubas/lede/tmp/dl && cd /home/jonlubas/lede/tmp/dl && rm -rf libnl-tiny-2020-08-05-c291088f && [ \! -d libnl-tiny-2020-08-05-c291088f ] && git clone  https://git.openwrt.org/project/libnl-tiny.git libnl-tiny-2020-08-05-c291088f && (cd libnl-tiny-2020-08-05-c291088f && git checkout c291088f631d1694f7ba0444b59677b194348da8 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd libnl-tiny-2020-08-05-c291088f && git log -1 --format='\''@%ct'\''` && rm -rf libnl-tiny-2020-08-05-c291088f/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libnl-tiny-2020-08-05-c291088f |    xz -zc -7e > /home/jonlubas/lede/tmp/dl/libnl-tiny-2020-08-05-c291088f.tar.xz && mv /home/jonlubas/lede/tmp/dl/libnl-tiny-2020-08-05-c291088f.tar.xz /home/jonlubas/lede/dl/ && rm -rf libnl-tiny-2020-08-05-c291088f;  )    '
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/libjson-c'
time: package/libs/libjson-c/download#0.09#1.57#2.28
make[2]: Entering directory '/home/jonlubas/lede/package/libs/pcre'
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/libmnl'
time: package/libs/libmnl/download#0.12#1.34#1.82
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/libnl-tiny'
time: package/libs/libnl-tiny/download#0.07#1.42#1.80
make[2]: Entering directory '/home/jonlubas/lede/package/libs/readline'
make[2]: Entering directory '/home/jonlubas/lede/package/libs/sysfsutils'
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.libubox-2021-05-16-b14c4688.tar.xz.flock -c '     /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "libubox-2021-05-16-b14c4688.tar.xz" "7dd1db1e0074a9c7c722db654cce3111b3bd3cff0bfd791c4497cb0f6c22d3ca" "" || (          echo "Checking out files from the git repository..."; mkdir -p /home/jonlubas/lede/tmp/dl && cd /home/jonlubas/lede/tmp/dl && rm -rf libubox-2021-05-16-b14c4688 && [ \! -d libubox-2021-05-16-b14c4688 ] && git clone  https://git.openwrt.org/project/libubox.git libubox-2021-05-16-b14c4688 && (cd libubox-2021-05-16-b14c4688 && git checkout b14c4688612c05c78ce984d7bde633bce8703b1e && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd libubox-2021-05-16-b14c4688 && git log -1 --format='\''@%ct'\''` && rm -rf libubox-2021-05-16-b14c4688/.git &&      tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libubox-2021-05-16-b14c4688 |       xz -zc -7e > /home/jonlubas/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz && mv /home/jonlubas/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz /home/jonlubas/lede/dl/ && rm -rf libubox-2021-05-16-b14c4688;  )    '
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/libubox'
time: package/libs/libubox/download#0.12#1.64#2.38
make[2]: Entering directory '/home/jonlubas/lede/package/libs/toolchain'
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.openssl-1.1.1n.tar.gz.flock -c '          /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "openssl-1.1.1n.tar.gz" "40dceb51a4f6a5275bde0e6bf20ef4b91bfc32ed57c0552e2e8e15463372b17a" "" "https://mirrors.tencent.com/openssl/source/" "https://mirrors.tencent.com/openssl/source/old/1.1.1/" "https://www.openssl.org/source/" "https://www.openssl.org/source/old/1.1.1/" "https://ftp.fi.muni.cz/pub/openssl/source/" "https://ftp.fi.muni.cz/pub/openssl/source/old/1.1.1/" "ftp://ftp.pca.dfn.de/pub/tools/net/openssl/source/" "ftp://ftp.pca.dfn.de/pub/tools/net/openssl/source/old/1.1.1/"    '
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.mbedtls-2.28.0.tar.gz.flock -c '          /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "mbedtls-2.28.0.tar.gz" "6519579b836ed78cc549375c7c18b111df5717e86ca0eeff4cb64b2674f424cc" "" "https://codeload.github.com/ARMmbed/mbedtls/tar.gz/v2.28.0?"    '
mkdir -p /home/jonlubas/lede/dl
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/openssl'
time: package/libs/openssl/download#0.23#1.06#1.69
SHELL= flock /home/jonlubas/lede/tmp/.libusb-1.0.24.tar.bz2.flock -c '          /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "libusb-1.0.24.tar.bz2" "7efd2685f7b327326dcfb85cee426d9b871fd70e22caa15bb68d595ce2a2b12a" "" "https://github.com/libusb/libusb/releases/download/v1.0.24" "@SF/libusb"    '
make[2]: Entering directory '/home/jonlubas/lede/package/libs/uclient'
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/mbedtls'
time: package/libs/mbedtls/download#0.12#1.48#1.99
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/libusb'
time: package/libs/libusb/download#0.28#1.67#2.12
make[2]: Entering directory '/home/jonlubas/lede/package/libs/ustream-ssl'
make[2]: Entering directory '/home/jonlubas/lede/package/libs/zlib'
mkdir -p /home/jonlubas/lede/dl
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.ncurses-6.2.tar.gz.flock -c '     /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "ncurses-6.2.tar.gz" "30306e0c76e0f9f1f0de987cf1c82a5c21e1ce6568b9227f7da5b71cbea86c9d" "" "@GNU/ncurses"    '
SHELL= flock /home/jonlubas/lede/tmp/.sysfsutils-2.1.0.tar.gz.flock -c '        /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "sysfsutils-2.1.0.tar.gz" "e865de2c1f559fff0d3fc936e660c0efaf7afe662064f2fb97ccad1ec28d208a" "" "@SF/linux-diag"    '
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/sysfsutils'
time: package/libs/sysfsutils/download#0.17#1.32#1.80
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/ncurses'
time: package/libs/ncurses/download#0.14#1.79#2.40
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.pcre-8.44.tar.bz2.flock -c '      /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "pcre-8.44.tar.bz2" "19108658b23b3ec5058edc9f66ac545ea19f9537234be1ec62b714c84399366d" "" "@SF/pcre"    '
make[2]: Entering directory '/home/jonlubas/lede/package/network/config/firewall'
make[2]: Entering directory '/home/jonlubas/lede/package/network/config/netifd'
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.readline-8.1.tar.gz.flock -c '    /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "readline-8.1.tar.gz" "f8ceb4ee131e3232226a17f51b164afc46cd0b9e6cef344be87c65962cb82b02" "" "@GNU/readline"    '
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/toolchain'
time: package/libs/toolchain/download#0.10#1.18#1.49
make[2]: Leaving directory '/home/jonlubas/lede/package/libs/pcre'
time: package/libs/pcre/download#0.23#1.71#2.30
make[2]: Entering directory '/home/jonlubas/lede/package/network/services/dnsmasq'
[... 170 lines cut ...]
time: package/utils/jsonfilter/download#0.12#1.23#1.64
SHELL= flock /home/jonlubas/lede/tmp/.e2fsprogs-1.46.5.tar.xz.flock -c '        /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "e2fsprogs-1.46.5.tar.xz" "2f16c9176704cf645dc69d5b15ff704ae722d665df38b2ed3cfc249757d8d81e" "" "@KERNEL/linux/kernel/people/tytso/e2fsprogs/v1.46.5/"    '
make[2]: Leaving directory '/home/jonlubas/lede/package/utils/e2fsprogs'
time: package/utils/e2fsprogs/download#0.15#1.70#2.03
mkdir -p /home/jonlubas/lede/dl
SHELL= flock /home/jonlubas/lede/tmp/.util-linux-2.37.3.tar.xz.flock -c '       /home/jonlubas/lede/scripts/download.pl "/home/jonlubas/lede/dl" "util-linux-2.37.3.tar.xz" "590c592e58cd6bf38519cb467af05ce6a1ab18040e3e3418f24bcfb2f55f9776" "" "@KERNEL/linux/utils/util-linux/v2.37"    '
mkdir -p /home/jonlubas/lede/dl
[... cut ...]
make[1]: *** [package/Makefile:116: package/utils/lua/download] Error 1
make[1]: Leaving directory '/home/jonlubas/lede'
make[1]: Entering directory '/home/jonlubas/lede'
make[2]: Entering directory '/home/jonlubas/lede/target/linux'
make[3]: Entering directory '/home/jonlubas/lede/target/linux/x86'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/jonlubas/lede/target/linux/x86'
make[2]: Leaving directory '/home/jonlubas/lede/target/linux'
time: target/linux/download#0.03#0.32#0.34
make[1]: Leaving directory '/home/jonlubas/lede'

jonlubas, 2022-03-17:
软路由里面虚拟了ARCHLINUX 跑DOCKER UBUNTU编译不成功，找了台老机器装了ARCHLINUX 继续DOCKER UBUNTU编译不成功，最后用WINDOWS10 子系统UBUNTU编译 还是不成功。折腾了快半个月了，实在是没辙。LEDE源码有问题？

WYC-2020, 2022-03-17:
真是不怕死，明显就是自己代理不行，然后下载失败了，就到备用节点下载了，结果hash不对而已。自己把hash跳过就行的，你还是不要编译了，看你折腾的这么造孽

jonlubas, 2022-03-17:
> 真是不怕死，明显就是自己代理不行，然后下载失败了，就到备用节点下载了，结果hash不对而已。自己把hash跳过就行的，你还是不要编译了，看你折腾的这么造孽

关键是这个lua的包 我已经去网站下载好了丢进dl文件夹里了。

WYC-2020, 2022-03-18:
> > 真是不怕死，明显就是自己代理不行，然后下载失败了，就到备用节点下载了，结果hash不对而已。自己把hash跳过就行的，你还是不要编译了，看你折腾的这么造孽
> 
> 关键是这个lua的包 我已经去网站下载好了丢进dl文件夹里了。

那么多个备用网址不是随便下就可以的，下的包必须hash和make里面一致才行，自己下载下来用工具算下，或者直接跳过hash值的校验

WYC-2020, 2022-03-18:
[lua-5.1.5.tar.gz](https://github.com/coolsnowwolf/lede/files/8300533/lua-5.1.5.tar.gz)

jonlubas, 2022-03-18:
> > > 真是不怕死，明显就是自己代理不行，然后下载失败了，就到备用节点下载了，结果hash不对而已。自己把hash跳过就行的，你还是不要编译了，看你折腾的这么造孽
> > 
> > 
> > 关键是这个lua的包 我已经去网站下载好了丢进dl文件夹里了。
> 
> 那么多个备用网址不是随便下就可以的，下的包必须hash和make里面一致才行，自己下载下来用工具算下，或者直接跳过hash值的校验

谢谢你 我已经编译过去了
我是一名卡车司机 玩成这样我觉得不错了
你帮我省了好多电费 支付宝发过来我发个红包
另外 怎么跳过HASH验证。

closed 2022-03-23 by coolsnowwolf