coolsnowwolf/lede #9088: mtk_ppe_offload.c 补丁报错

sKyissKy, 2022-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mtk_ppe_offload.c 补丁报错 

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
Applying /home/ubuntu/lede/target/linux/generic/pending-5.10/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch using plaintext: 
patching file drivers/net/ethernet/mediatek/mtk_ppe_offload.c
Hunk #1 FAILED at 7.
Hunk #2 FAILED at 20.
Hunk #3 succeeded at 78 with fuzz 1 (offset 14 lines).
Hunk #4 succeeded at 302 with fuzz 2 (offset 40 lines).
Hunk #5 succeeded at 351 with fuzz 1 (offset 51 lines).
2 out of 5 hunks FAILED -- saving rejects to file drivers/net/ethernet/mediatek/mtk_ppe_offload.c.rej
Patch failed!  Please fix /home/ubuntu/lede/target/linux/generic/pending-5.10/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch!
make[4]: *** [Makefile:32: /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/linux-ipq807x_generic/linux-5.10.104/.prepared_39c75a14d89526dae1da6da9d32474f5] Error 1
make[4]: Leaving directory '/home/ubuntu/lede/target/linux/ipq807x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/target/linux'
time: target/linux/compile#1.57#10.08#9.35
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [target/Makefile:18: /home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53+crypto_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230：world] 错误 2
[config.txt](https://github.com/coolsnowwolf/lede/files/8288568/config.txt)

sKyissKy, 2022-03-17:
X86编译没问题了。但是ipq807x编译会报错，希望大雕修复，谢谢。

WYC-2020, 2022-03-17:
自己去找相关issue说了解决方法，稍微翻下issue嘛

aaronchin, 2022-03-18:
我遭遇的是701有问题拿掉701以后920有问题，拿掉920以后出现以下
At top level:
drivers/net/ethernet/mediatek/mtk_ppe_offload.c:74:1: warning: 'mtk_flow_set_ipv6_addr' defined but not used [-Wunused-function]
 mtk_flow_set_ipv6_addr(struct mtk_foe_entry *foe, struct mtk_flow_data *data)
 ^~~~~~~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:280: drivers/net/ethernet/mediatek/mtk_ppe_offload.o] Error 1
make[8]: *** [scripts/Makefile.build:497: drivers/net/ethernet/mediatek] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/net/ethernet] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: *** [Makefile:1822: drivers] Error 2
make[5]: Leaving directory '/home/aaronchin/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.104'
make[4]: *** [Makefile:19: /home/aaronchin/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.104/.modules] Error 2
make[4]: Leaving directory '/home/aaronchin/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/aaronchin/lede/target/linux'
time: target/linux/compile#8.59#4.45#12.72
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/aaronchin/lede'
make[1]: *** [target/Makefile:18: /home/aaronchin/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/aaronchin/lede'
make: *** [/home/aaronchin/lede/include/toplevel.mk:230: world] Error 2

closed 2022-03-18 by coolsnowwolf