coolsnowwolf/lede #9090: 编译时使用GZIP img镜像文件大小不会压缩

TorBoxCode, 2022-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ *] 没有类似的issue

## 2. 详细叙述
使用CONFIG_TARGET_IMAGES_GZIP=y编译时CONFIG_TARGET_ROOTFS_PARTSIZE指定多大最后编译出的img镜像就有多大
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
路由器：nanopi R4s
固件为squashfs版本
版本为3月18日最新代码编译，5.4；5.10；5.15版本内核都试过了。

A：

### (3) 详细日志

A：

WYC-2020, 2022-03-18:
这个gzip 是最后不会产生img后缀的 会被压缩进。img.gz 这样的后缀文件中，大小难道没有变化？我x86 指定2G压缩后 就才170M

TorBoxCode, 2022-03-18:
> 这个gzip是最后不会产生img后缀的会被压缩进。img.gz这样的后缀文件中，大小有没有变化？我x86指定2G压缩后才170M

gz压缩文件是会变小，但解压后img文件不会变小，去年的版本img文件的大小是实际文件的大小，比如说编译了刷32GB卡的实际大小就1-2百M左右，现在编译出来的就是32GB，这会导致一个问题，升级的时候上传不了，只能重新写卡。

WYC-2020, 2022-03-18:
> > 这个gzip是最后不会产生img后缀的会被压缩进。img.gz这样的后缀文件中，大小有没有变化？我x86指定2G压缩后才170M
> 
> gz压缩文件是会变小，但解压后img文件不会变小，去年的版本img文件的大小是实际文件的大小，比如说编译了刷32GB卡的实际大小就1-2百M左右，现在编译出来的就是32GB，这会导致一个问题，升级的时候上传不了，只能重新写卡。

那是填充问题而已，这个几万年前就说了，这个我一直用x86 一直都是这样，你可以去看lienol的，他的好像没填充

WYC-2020, 2022-03-18:
你把PADDING=1 去掉试下

> > 这个gzip是最后不会产生img后缀的会被压缩进。img.gz这样的后缀文件中，大小有没有变化？我x86指定2G压缩后才170M
> 
> gz压缩文件是会变小，但解压后img文件不会变小，去年的版本img文件的大小是实际文件的大小，比如说编译了刷32GB卡的实际大小就1-2百M左右，现在编译出来的就是32GB，这会导致一个问题，升级的时候上传不了，只能重新写卡。

TorBoxCode, 2022-03-18:
> 你把PADDING=1 去掉试下
> 
> > > 这个gzip是最后不会产生img后缀的会被压缩进。img.gz这样的后缀文件中，大小有没有变化？我x86指定2G压缩后才170M
> > 
> > 
> > gz压缩文件是会变小，但解压后img文件不会变小，去年的版本img文件的大小是实际文件的大小，比如说编译了刷32GB卡的实际大小就1-2百M左右，现在编译出来的就是32GB，这会导致一个问题，升级的时候上传不了，只能重新写卡。

按照你的指导今天编译了一下已经解决了这个问题，谢谢！

liwenjie119, 2022-03-18:
> > 你把PADDING=1 去掉试下
> > > > 这个gzip是最后不会产生img后缀的会被压缩进。img.gz这样的后缀文件中，大小有没有变化？我x86指定2G压缩后才170M
> > > 
> > > 
> > > gz压缩文件是会变小，但解压后img文件不会变小，去年的版本img文件的大小是实际文件的大小，比如说编译了刷32GB卡的实际大小就1-2百M左右，现在编译出来的就是32GB，这会导致一个问题，升级的时候上传不了，只能重新写卡。
> 
> 按照你的指导今天编译了一下已经解决了这个问题，谢谢！

这个是修改哪个配置啊，我没在config中找到此项

TorBoxCode, 2022-03-18:
> > > 你把PADDING=1 去掉试下
> > > > > 这个gzip是最后不会产生img后缀的会被压缩进。img.gz这样的后缀文件中，大小有没有变化？我x86指定2G压缩后才170M
> > > > 
> > > > 
> > > > gz压缩文件是会变小，但解压后img文件不会变小，去年的版本img文件的大小是实际文件的大小，比如说编译了刷32GB卡的实际大小就1-2百M左右，现在编译出来的就是32GB，这会导致一个问题，升级的时候上传不了，只能重新写卡。
> > 
> > 
> > 按照你的指导今天编译了一下已经解决了这个问题，谢谢！
> 
> 这个是修改哪个配置啊，我没在config中找到此项

sed -i 's/PADDING=1//g' lede/target/linux/rockchip/image/Makefile

TorBoxCode, 2022-03-18:
> sed -i 's/PADDING=1//g' lede/target/linux/rockchip/image/Makefile

根据你的CPU架构在lede/target/linux/对应文件夹中找

closed 2022-03-18 by TorBoxCode