coolsnowwolf/lede #9093: 内网IP登录web页面自动跳转到 https

deplives, 2022-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

几天前刚编译的版本，内网 IP 登录总是跳转到 https
web 管理中 没有启用 重定向HTTP到HTTPS
[image]
[image]

### (2) 路由器型号和固件版本

x86

### (3) 详细日志

A：

deplives, 2022-03-18:
补充一下新的发现：
在luci登陆界面未登录的时候，一定会自动跳转到https上。即使是手动输入http:// 也会被重定向到https，但是输入密码登录之后，就可以手动换成http，不会被重定向了

deplives, 2022-03-18:
> 清空浏览器所有记录，或者用无痕模式

[image]

WYC-2020, 2022-03-18:
我是不清楚你是怎么搞的，反正我用x86这么久没遇到过这个问题，你web服务是用的uhttp还是nginx

WYC-2020, 2022-03-18:
换个浏览器再试下

Beginner-Go, 2022-03-18:
他应该用的是nginx

deplives, 2022-03-18:
web 服务用的是 uhttp
[image]
尝试在`/etc/config/uhttpd` 中删除 
```
option cert '/etc/uhttpd.crt'
option key '/etc/uhttpd.key'
```
然后重启之后web界面就起不来了，而且我在虚拟机完全重新安装了一下我现在用的这个固件，发现默认是有这个配置的。所以我理解应该不是这个的问题。

deplives, 2022-03-18:
解决了
删掉了`/etc/config/uhttpd`中的
```
list listen_https '0.0.0.0:443'
list listen_https '[::]:443'
option cert '/etc/uhttpd.crt'
option key '/etc/uhttpd.key'
```
重启 uhttpd 就可以了
然后发现了个问题，我猜测貌似是之前只要在 `Web 管理页面设置` 中打开过 `重定向HTTP到HTTPS` 就会增加上面的配置。但是关闭这选项后https相关的配置没有被去掉导致的。
这个只是我猜测的，没有验证

WYC-2020, 2022-03-18:
我的一直就有，也不会跳转呢

WYC-2020, 2022-03-18:
刚测试了，可以重定向可以取消，没任何问题

closed 2022-03-25 by coolsnowwolf

xingyunjohn1, 2022-08-09:
> 补充一下新的发现： 在luci登陆界面未登录的时候，一定会自动跳转到https上。即使是手动输入http:// 也会被重定向到https，但是输入密码登录之后，就可以手动换成http，不会被重定向了

我有同样的问题。我在某些情况需要用到https，不能像你一样直接把他删了。目前还没找到好的办法。

miludeerforest, 2023-02-07:
> 解决了 删掉了`/etc/config/uhttpd`中的
> 
> ```
> list listen_https '0.0.0.0:443'
> list listen_https '[::]:443'
> option cert '/etc/uhttpd.crt'
> option key '/etc/uhttpd.key'
> ```
> 
> 重启 uhttpd 就可以了 然后发现了个问题，我猜测貌似是之前只要在 `Web 管理页面设置` 中打开过 `重定向HTTP到HTTPS` 就会增加上面的配置。但是关闭这选项后https相关的配置没有被去掉导致的。 这个只是我猜测的，没有验证

不用删除的，破案了，你点击取消uhttpd重定向之后，右上角显示的代码 uhttpd.main.redirect_https='0' 才是你需要添加到/etc/config/uhttpd里的，原先的option redirect_https '0'是不生效的。添加uhttpd.main.redirect_https='0'进去之后退出管理账户，清cookie,重登录就正常了

fenfenyangyangmate, 2023-08-30:
> > 解决了 删掉了`/etc/config/uhttpd`的
> > ```
> > list listen_https '0.0.0.0:443'
> > list listen_https '[::]:443'
> > option cert '/etc/uhttpd.crt'
> > option key '/etc/uhttpd.key'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 重启uhttpd就可以了然后发现了问题，我猜测好像是之前只要在中`Web 管理页面设置`打开`重定向HTTP到HTTPS`就会增加上面的配置。但是这个选项后https关闭相关的配置没有被去掉导致的。只是这个我猜测的，没有验证
> 
> 不用删除的，破案了，你点击取消uhttpd重定向之后右上角，显示的代码 uhttpd.main.redirect_https='0' 那么你需要添加到/etc/config/uhttpd里的，这样的选项redirect_https '0 '是不生效的。添加uhttpd.main.redirect_https='0'进去之后退出管理账户，清cookie，重新登录就正常了

这是正解

xuhuanzy, 2023-09-02:
> 不用删除的，破案了，你点击取消uhttpd重定向之后，右上角显示的代码 uhttpd.main.redirect_https='0' 才是你需要添加到/etc/config/uhttpd里的，原先的option redirect_https '0'是不生效的。添加uhttpd.main.redirect_https='0'进去之后退出管理账户，清cookie,重登录就正常了

这加在哪里?

xianren78, 2024-06-06:
简单的办法是添加一个单独的https配置监听443，原来的main配置只监听80端口即可