coolsnowwolf/lede #9094: 编译错误，大致意思是提示在patch一个驱动时失败，具体原因未知，请大神们帮忙看看怎么解决

cfj521, 2022-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
编译错误，大致意思是提示在patch一个驱动时失败，具体原因未知，请大神们帮忙看看怎么解决

### (1) 具体问题
编译失败，日志如下
A：

### (2) 路由器型号和固件版本
Redmi AX6 R21.11.11。个人revert了那个drop

A：

### (3) 详细日志

Applying /home/colex/lede/target/linux/generic/pending-5.10/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch using plaintext:
patching file drivers/net/ethernet/mediatek/mtk_ppe_offload.c
Hunk #1 FAILED at 7.
Hunk #2 FAILED at 20.
Hunk #3 succeeded at 78 with fuzz 1 (offset 14 lines).
Hunk #4 succeeded at 302 with fuzz 2 (offset 40 lines).
Hunk #5 succeeded at 351 with fuzz 1 (offset 51 lines).
2 out of 5 hunks FAILED -- saving rejects to file drivers/net/ethernet/mediatek/mtk_ppe_offload.c.rej
Patch failed!  Please fix /home/colex/lede/target/linux/generic/pending-5.10/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch!
make[3]: *** [Makefile:122: /home/colex/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/linux-5.10.104/.prepared] Error 1
make[3]: Leaving directory '/home/colex/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#3.50#11.46#10.35
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/colex/lede'
make[1]: *** [toolchain/Makefile:93: /home/colex/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/colex/lede'
make: *** [/home/colex/lede/include/toplevel.mk:230: world] Error 2

A：

WYC-2020, 2022-03-18:
翻下相同的patch嘛，都不嫌烦，这个又说怎么解决啊

cfj521, 2022-03-19:
> 翻下相同的patch嘛，都不嫌烦，这个又说怎么解决啊

抱歉，我是个新手，不太懂你说的看相同的patch是什么意思，我看它上面一个applying patch没什么问题，以下是 .patch 原文件，我大概看了下语法也没什么问题，真不知道问题出在哪？

[701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch.txt](https://github.com/coolsnowwolf/lede/files/8308830/701-net-ethernet-mtk_eth_soc-add-ipv6-flow-offloading-support.patch.txt)

cfj521, 2022-03-19:
the problem is gone, by cut off the duplicated item, as https://github.com/coolsnowwolf/lede/pull/9098.

closed 2022-03-19 by cfj521