coolsnowwolf/lede #9102: zerotier疯狂输出日志

zjy2200656, 2022-03-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
内核日志疯狂报错，大概20秒一次
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：重启无效，关闭zerotier不在输出错误日志，不影响路由器使用

### (2) 路由器型号和固件版本

A：P&W R619AC 128M
      OpenWrt R22.3.13 / LuCI Master (git-22.068.45502-a50e601)

### (3) 详细日志

A：
Sun Mar 20 07:38:40 2022 kern.warn kernel: [50463.270524] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:39:01 2022 kern.warn kernel: [50483.604237] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:39:10 2022 kern.warn kernel: [50493.304385] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:39:40 2022 kern.warn kernel: [50523.340233] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:40:06 2022 kern.warn kernel: [50548.664569] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:40:10 2022 kern.warn kernel: [50553.370209] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:40:40 2022 kern.warn kernel: [50583.405808] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:41:11 2022 kern.warn kernel: [50613.441337] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:41:11 2022 kern.warn kernel: [50613.755183] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:41:41 2022 kern.warn kernel: [50643.477447] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:42:11 2022 kern.warn kernel: [50673.511105] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:42:16 2022 kern.warn kernel: [50678.839201] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:42:41 2022 kern.warn kernel: [50703.543684] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:43:11 2022 kern.warn kernel: [50733.577253] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:43:21 2022 kern.warn kernel: [50743.901877] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:43:41 2022 kern.warn kernel: [50763.610398] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:44:11 2022 kern.warn kernel: [50793.643936] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:44:26 2022 kern.warn kernel: [50808.962750] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:44:41 2022 kern.warn kernel: [50823.678982] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:45:11 2022 kern.warn kernel: [50853.713192] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:45:31 2022 kern.warn kernel: [50874.034289] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:45:41 2022 kern.warn kernel: [50883.745536] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:46:11 2022 kern.warn kernel: [50913.780804] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:46:36 2022 kern.warn kernel: [50939.121270] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:46:41 2022 kern.warn kernel: [50943.816566] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:47:11 2022 kern.warn kernel: [50973.846744] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:47:41 2022 kern.warn kernel: [51003.880446] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:47:41 2022 kern.warn kernel: [51004.205638] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:48:11 2022 kern.warn kernel: [51033.914757] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:48:41 2022 kern.warn kernel: [51063.945090] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:48:46 2022 kern.warn kernel: [51069.258164] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:49:11 2022 kern.warn kernel: [51093.978644] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:49:41 2022 kern.warn kernel: [51124.010544] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:49:51 2022 kern.warn kernel: [51134.324117] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:50:11 2022 kern.warn kernel: [51154.042687] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:50:41 2022 kern.warn kernel: [51184.080610] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:50:56 2022 kern.warn kernel: [51199.409865] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:51:11 2022 kern.warn kernel: [51214.114481] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:51:41 2022 kern.warn kernel: [51244.148037] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:52:02 2022 kern.warn kernel: [51264.493368] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:52:11 2022 kern.warn kernel: [51274.183824] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:52:41 2022 kern.warn kernel: [51304.217674] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:53:07 2022 kern.warn kernel: [51329.546205] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:53:11 2022 kern.warn kernel: [51334.252137] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:53:41 2022 kern.warn kernel: [51364.286319] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:54:11 2022 kern.warn kernel: [51394.319553] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:54:12 2022 kern.warn kernel: [51394.633885] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:54:41 2022 kern.warn kernel: [51424.356166] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 07:55:11 2022 kern.warn kernel: [51454.389978] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
[... 385 lines cut ...]
Sun Mar 20 10:07:21 2022 kern.warn kernel: [59383.659552] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:07:36 2022 kern.warn kernel: [59398.978906] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:07:51 2022 kern.warn kernel: [59413.695002] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:08:21 2022 kern.warn kernel: [59443.746418] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:08:41 2022 kern.warn kernel: [59464.061951] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:08:51 2022 kern.warn kernel: [59473.772179] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:09:21 2022 kern.warn kernel: [59503.806508] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:09:46 2022 kern.warn kernel: [59529.148668] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:09:51 2022 kern.warn kernel: [59533.843239] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:10:21 2022 kern.warn kernel: [59563.877008] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:10:51 2022 kern.warn kernel: [59593.910898] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
[... cut ...]
Sun Mar 20 10:14:21 2022 kern.warn kernel: [59804.146079] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:14:51 2022 kern.warn kernel: [59834.179472] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:15:12 2022 kern.warn kernel: [59854.512220] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.
Sun Mar 20 10:15:21 2022 kern.warn kernel: [59864.220265] netlink: 8 bytes leftover after parsing attributes in process `zerotier-one'.

WYC-2020, 2022-03-20:
这玩意上游的问题，https://github.com/zerotier/ZeroTierOne/issues/1380

WYC-2020, 2022-03-21:
[image]
根据这个人的测试你可以用1.4.6试下

zjy2200656, 2022-03-21:
> [image] 根据这个人的测试你可以用1.4.6试下

好的，但是同步最新源码直接不启动了，还需要回退吗
显示	
sh -c config() { echo "You cannot use UCI in firewall includes!" >&2; exit 1; }; . /etc/zerotier.start

closed 2022-03-21 by zjy2200656