coolsnowwolf/lede #9106: x86 5.10内核编译失败

liuyanantj, 2022-03-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86 改5.10内核后编译失败

### (2) 路由器型号和固件版本

A：x86 同步近日最新代码

### (3) 详细日志

Applying /home/tc/lede/target/linux/generic/hack-5.10/220-gc_sections.patch using plaintext: 
patching file arch/arm/Kconfig
patching file arch/arm/boot/compressed/Makefile
patching file arch/arm/kernel/vmlinux.lds.S
patching file arch/arm/include/asm/vmlinux.lds.h
Hunk #1 FAILED at 23.
Hunk #2 succeeded at 109 (offset 13 lines).
Hunk #3 FAILED at 112.
2 out of 3 hunks FAILED -- saving rejects to file arch/arm/include/asm/vmlinux.lds.h.rej
Patch failed!  Please fix /home/tc/lede/target/linux/generic/hack-5.10/220-gc_sections.patch!
Makefile:112: recipe for target '/home/tc/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/linux-5.10.106/.prepared' failed
make[3]: *** [/home/tc/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/linux-5.10.106/.prepared] Error 1
make[3]: Leaving directory '/home/tc/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#1.55#9.01#8.03
    ERROR: toolchain/kernel-headers failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/tc/lede'
toolchain/Makefile:93: recipe for target '/home/tc/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/tc/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
/home/tc/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

minfengyk, 2022-03-20:
全新编译，同样的问题

liuyanantj, 2022-03-20:
收到  我试试看

Beginner-Go, 2022-03-20:
can you test?
https://github.com/coolsnowwolf/lede/pull/9105

alan21j, 2022-03-21:
同樣的問題 mips mt7621 編譯 5.4 沒問題， 編譯 5.10 不成功出現如下情況：

Applying /workdir/openwrt/target/linux/generic/hack-5.10/214-spidev_h_portability.patch using plaintext: 
patching file include/uapi/linux/spi/spidev.h

Applying /workdir/openwrt/target/linux/generic/hack-5.10/220-gc_sections.patch using plaintext: 
patching file arch/arm/Kconfig
patching file arch/arm/boot/compressed/Makefile
patching file arch/arm/kernel/vmlinux.lds.S
patching file arch/arm/include/asm/vmlinux.lds.h
Hunk #1 FAILED at 23.
Hunk #2 succeeded at 109 (offset 13 lines).
Hunk #3 FAILED at 112.
2 out of 3 hunks FAILED -- saving rejects to file arch/arm/include/asm/vmlinux.lds.h.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/generic/hack-5.10/220-gc_sections.patch!
make[3]: *** [Makefile:123: /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/linux-5.10.106/.prepared] Error 1
make[3]: Leaving directory '/workdir/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#12.64#6.41#15.21
    ERROR: toolchain/kernel-headers failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-03-21:
patch更新不完整

liuyanantj, 2022-03-21:
>

liuyanantj, 2022-03-21:
> can you test? #9105

不太会整  看着文件不是一个文件  还是等等修复吧

liuyanantj, 2022-03-21:
> patch更新不完整

怎么更新 后来又全新编译了  还是过不去

JoseCoW, 2022-03-21:
先拉 Beginner-Go 仓库里最新的改动吧，在 lede 下执行
```
git remote add 'k5.10' -t 'k5.10' https://github.com/Beginner-Go/lede.git
```
```
git fetch 'k5.10'
```
```
git cherry-pick 536e1e27da074d9cacc4cff76fc1b59a7efca495
```

WYC-2020, 2022-03-21:
https://github.com/WYC-2020/lede/commit/85d58b252ac5560beb9dd5a2198acd9330518da9
简单修复了一下
@liuyanantj @minfengyk @alan21j

dot-x-line, 2022-03-21:
同样情况，刚试了，lede文件夹git pull过了，还是失败。

Applying /home/dxl/lede/target/linux/generic/hack-5.10/220-gc_sections.patch using plaintext:
patching file arch/arm/Kconfig
patching file arch/arm/boot/compressed/Makefile
patching file arch/arm/kernel/vmlinux.lds.S
patching file arch/arm/include/asm/vmlinux.lds.h
Hunk #1 FAILED at 23.
Hunk #2 succeeded at 109 (offset 13 lines).
Hunk #3 FAILED at 112.
2 out of 3 hunks FAILED -- saving rejects to file arch/arm/include/asm/vmlinux.lds.h.rej
Patch failed!  Please fix /home/dxl/lede/target/linux/generic/hack-5.10/220-gc_sections.patch!
make[3]: *** [Makefile:123: /home/dxl/lede/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/linux-5.10.106/.prepared] Error 1
make[3]: Leaving directory '/home/dxl/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#23.12#128.26#152.85
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/dxl/lede'
make[1]: *** [toolchain/Makefile:93: /home/dxl/lede/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/dxl/lede'
make: *** [/home/dxl/lede/include/toplevel.mk:230: world] Error 2

Beginner-Go, 2022-03-21:
> 同样情况，刚试了，lede文件夹git pull过了，还是失败。
>
> Applying /home/dxl/lede/target/linux/generic/hack-5.10/220-gc_sections.patch using plaintext: patching file arch/arm/Kconfig patching file [...]

删除lede 重新clone

liuyanantj, 2022-03-21:
> [WYC-2020/lede@85d58b2](https://github.com/WYC-2020/lede/commit/85d58b252ac5560beb9dd5a2198acd9330518da9) 简单修复了一下 @liuyanantj @minfengyk @alan21j

感谢！！

closed 2022-03-21 by liuyanantj

WYC-2020, 2022-03-21:
> > [WYC-2020/lede@85d58b2](https://github.com/WYC-2020/lede/commit/85d58b252ac5560beb9dd5a2198acd9330518da9) 简单修复了一下 @liuyanantj @minfengyk @alan21j
> 
> 感谢！！

暂时别用我说的这个，目前还有一点问题

WYC-2020, 2022-03-21:
> > [WYC-2020/lede@85d58b2](https://github.com/WYC-2020/lede/commit/85d58b252ac5560beb9dd5a2198acd9330518da9) 简单修复了一下 @liuyanantj @minfengyk @alan21j
> 
> 感谢！！

对比5.4 .5.10 5.15  这三个版本的patch 发现有的地方有漏掉 不知是有意去掉还是什么，具体不清楚，我这边正在修改，但是具体是否可以用我这边没机器验证，所以暂时不要用我这个提交哈，万一砖块就悲剧了，我先自己测试能编译过再说,还是以大雕的为准吧，我自己仓库无所谓

WYC-2020, 2022-03-21:
> > [WYC-2020/lede@85d58b2](https://github.com/WYC-2020/lede/commit/85d58b252ac5560beb9dd5a2198acd9330518da9) 简单修复了一下 @liuyanantj @minfengyk @alan21j
> 
> 感谢！！

https://github.com/WYC-2020/lede/commit/1a0ea1ea8b0921ec1d48eec66958882ec1388a1a
好了 5.4 5.10 5.15  关于这个补丁现在统一了一下，并且5.4有一个地方新增了一处，欢迎入坑测试哈哈

dot-x-line, 2022-03-21:
> 

大神，要用你的lede库吗？还是已经更新到lean的库了。

dot-x-line, 2022-03-21:
> > > [WYC-2020/lede@85d58b2](https://github.com/WYC-2020/lede/commit/85d58b252ac5560beb9dd5a2198acd9330518da9) 简单修复了一下 @liuyanantj @minfengyk @alan21j
> > 
> > 
> > 感谢！！
> 
> [WYC-2020/lede@1a0ea1e](https://github.com/WYC-2020/lede/commit/1a0ea1ea8b0921ec1d48eec66958882ec1388a1a) 好了 5.4 5.10 5.15 关于这个补丁现在统一了一下，并且5.4有一个地方新增了一处，欢迎入坑测试哈哈

大神，要用你的lede库吗？还是已经更新到lean的库了。

WYC-2020, 2022-03-21:
> 大神，要用你的lede库吗？还是已经更新到lean的库了。

大雕是已经修复由于这个问题导致的编译问题，只是觉得那个patch 不完整并且不统一，我这边完整了一下，没有更新到大雕的库