coolsnowwolf/lede #9108: x86最新编译报错，恳请高手帮忙看是什么原因

zhengyiquan, 2022-03-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
已经全新重新编译了一次。5.15内核的，没有问题。但是我按照贴友提供的信息修改了5.4内核编译就出现报错。恳请高手帮忙看下是那里出错了

### (1)
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
<pre>Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-fs-ntfs3 for ntfs3-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for ntfs3-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 * 	kmod-fs-ntfs3
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory &apos;/home/z/lede&apos;
make[1]: *** [package/Makefile:111: /home/z/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory &apos;/home/z/lede&apos;
make: *** [/home/z/lede/include/toplevel.mk:230：world] 错误 2
</pre>

A：

zhengyiquan, 2022-03-20:
就只修改了KERNEL_PATCHVER:=5.15 改成 5.4

WYC-2020, 2022-03-20:
那肯定失败了，自己删除config重新配置，原因懒得说，和自动挂载依赖有关系，以前说过

closed 2022-03-23 by coolsnowwolf

zhengyiquan, 2022-03-25:
> 那肯定失败了，自己删除config重新配置，原因懒得说，和自动挂载依赖有关系，以前说过

又重新拉取全新编译了一次，可以了。谢谢