coolsnowwolf/lede #9112: 编绎ipq60xx,报错

lgh200, 2022-03-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Makefile:17: recipe for target '/home/lgh/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/linux-ipq60xx/linux-5.10.106/.prepared_e0a3b657bf26eb2afbe37a73b27cda5c' failed
make[4]: *** [/home/lgh/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/linux-ipq60xx/linux-5.10.106/.prepared_e0a3b657bf26eb2afbe37a73b27cda5c] Error 1
make[4]: Leaving directory '/home/lgh/lede/target/linux/ipq60xx'
Makefile:11: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/lgh/lede/target/linux'
time: target/linux/compile#8.93#66.20#74.54
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 1
make[2]: Leaving directory '/home/lgh/lede'
target/Makefile:18: recipe for target '/home/lgh/lede/staging_dir/target-aarch64_cortex-a53+crypto_musl/stamp/.target_compile' failed
make[1]: *** [/home/lgh/lede/staging_dir/target-aarch64_cortex-a53+crypto_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lgh/lede'
/home/lgh/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
lgh@DESKTOP-1GU35TK:~/lede$

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

1054009064, 2022-03-21:
昨天更新的内核翻车车了，等修复吧

kairlychen, 2022-03-21:
昨天早上七八点编译还好好的，下午再编译一下就翻车车了。今天看有更新了，中午再编译了一下，还是报错。看着好像还是内核的编译报错

closed 2022-03-23 by lgh200