coolsnowwolf/lede #9113: zerotier无法启动

zjy2200656, 2022-03-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
zerotier无法启动
Q：是否搜索了issue (使用 "x" 选择)x
* [x] 没有类似的issue

## 2. 详细叙述
同步到最新源码编译之后，zerotier无法启动
### (1) 具体问题

A：路由器系统界面显示

sh -c config() { echo "You cannot use UCI in firewall includes!" >&2; exit 1; }; . /etc/zerotier.start

### (2) 路由器型号和固件版本

A：
P&W R619AC 128M
OpenWrt R22.3.13 / LuCI Master (git-22.068.45502-a50e601)

### (3) 详细日志

A：Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.829148] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.840346] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.844013] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.847743] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.851490] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.855275] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.858963] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.862676] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.866474] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.870324] warning: do not keep alive
Mon Mar 21 10:00:14 2022 kern.warn kernel: [34636.873880] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34636.877685] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34636.881355] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34636.890705] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34636.931492] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.016874] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.017053] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.021467] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.024073] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.027058] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.030787] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.035897] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.039301] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.059368] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.068114] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.074462] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.076285] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.143890] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.144161] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.145648] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.148105] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.150329] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.168729] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.169968] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.171152] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.171401] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.175161] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.178846] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.186333] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.190044] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.190852] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.193789] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.198610] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.201235] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.205018] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.208768] warning: do not keep alive
Mon Mar 21 10:00:15 2022 kern.warn kernel: [34637.216220] warning: do not keep alive
[... 659 lines cut ...]
Mon Mar 21 10:28:40 2022 kern.warn kernel: [36342.366887] warning: do not keep alive
Mon Mar 21 10:28:53 2022 kern.warn kernel: [36354.910499] warning: do not keep alive
Mon Mar 21 10:28:53 2022 kern.warn kernel: [36354.962485] warning: do not keep alive
Mon Mar 21 10:28:55 2022 kern.warn kernel: [36357.159832] warning: do not keep alive
Mon Mar 21 10:28:55 2022 kern.warn kernel: [36357.173692] warning: do not keep alive
Mon Mar 21 10:30:01 2022 kern.warn kernel: [36423.276723] warning: do not keep alive
Mon Mar 21 10:30:01 2022 kern.warn kernel: [36423.602368] warning: do not keep alive
Mon Mar 21 10:31:55 2022 kern.warn kernel: [36537.194457] warning: do not keep alive
Mon Mar 21 10:31:55 2022 kern.warn kernel: [36537.208606] warning: do not keep alive
Mon Mar 21 10:32:00 2022 kern.warn kernel: [36542.593500] warning: do not keep alive
Mon Mar 21 10:32:01 2022 kern.warn kernel: [36543.603127] warning: do not keep alive
Mon Mar 21 10:32:03 2022 kern.warn kernel: [36545.611647] warning: do not keep alive
Mon Mar 21 10:33:05 2022 kern.warn kernel: [36607.607654] warning: do not keep alive
Mon Mar 21 10:33:58 2022 kern.warn kernel: [36659.877471] warning: do not keep alive
Mon Mar 21 10:34:40 2022 kern.warn kernel: [36702.647024] warning: do not keep alive
Mon Mar 21 10:34:41 2022 kern.warn kernel: [36702.879469] warning: do not keep alive
Mon Mar 21 10:34:41 2022 kern.warn kernel: [36702.959892] warning: do not keep alive
Mon Mar 21 10:34:41 2022 kern.warn kernel: [36703.601047] warning: do not keep alive
Mon Mar 21 10:34:55 2022 kern.warn kernel: [36717.227625] warning: do not keep alive
Mon Mar 21 10:34:55 2022 kern.warn kernel: [36717.241750] warning: do not keep alive
Mon Mar 21 10:35:20 2022 kern.warn kernel: [36742.769709] warning: do not keep alive
Mon Mar 21 10:35:21 2022 kern.warn kernel: [36743.604580] warning: do not keep alive
Mon Mar 21 10:35:45 2022 kern.warn kernel: [36767.613845] warning: do not keep alive
Mon Mar 21 10:36:01 2022 kern.warn kernel: [36783.608421] warning: do not keep alive
Mon Mar 21 10:36:25 2022 kern.warn kernel: [36807.621287] warning: do not keep alive

WYC-2020, 2022-03-21:
刚还试了一切正常，刚编译的

zjy2200656, 2022-03-21:
> 刚还试了一切正常，刚编译的

https://github.com/coolsnowwolf/packages/commit/7ad235a77cf89c6f9241f82e1c6a5fc3a108f081#commitcomment-69068522
应该是armv7上都不工作，回退1.8.4正常

WYC-2020, 2022-03-21:
> > 刚还试了一切正常，刚编译的
> 
> [coolsnowwolf/packages@7ad235a#commitcomment-69068522](https://github.com/coolsnowwolf/packages/commit/7ad235a77cf89c6f9241f82e1c6a5fc3a108f081#commitcomment-69068522) 应该是armv7上都不工作，回退1.8.4正常

那就不清楚了，没这个环境，没法测试

breakings, 2022-03-22:
> > 刚还试了一切正常，刚编译的
> 
> [coolsnowwolf/packages@7ad235a#commitcomment-69068522](https://github.com/coolsnowwolf/packages/commit/7ad235a77cf89c6f9241f82e1c6a5fc3a108f081#commitcomment-69068522) 应该是armv7上都不工作，回退1.8.4正常

去掉`-fPIE` 这个编译看看

Lionzj, 2022-03-23:
K3编译完也是一样的问题

closed 2022-03-23 by zjy2200656