coolsnowwolf/lede #9116: ipv6 udp连接会在10分钟后出现问题

evfun, 2022-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
使用ipv6 udp连接，udp连接会在10分钟后出现问题，服务器发送的数据客户端不能收到，但客户端发送的数据服务器能收到

### (1) 具体问题
路由器上没有运行ssrp，多线多拨，qos等网络插件。
路由器下面一个openvpn客户端，使用ipv6 udp协议连接至服务端。开始的10分钟内一切良好，但是10分钟以后连接断开。openvpn开始重新连接。
尝试在其它网络下使用ipv6 udp连接openvpn服务器，没有问题。
尝试使用ipv6 tcp进行ssh连接至服务器，10分钟后没有掉线，说明ipv6 tcp长连接没有问题。

最后使用netcat对udp测试
找一台有公网ipv6的服务器，在服务器上运行 nc -6 -l -u -p 2001
在客户端运行while :; do date ; sleep 2; done | nc -6 -u  [服务器的公网v6地址] 2001
可以看到服务器上每隔两秒收到客户端发来的时间。
在服务器端输入一些任意内容后按回车，输入的内容会被发送到客户端并显示在终端上。
过了10分钟以后，服务端依然会显示客户端发来的内容，但是服务器无法向客户端发送数据。
此时ipv6 udp连接出现问题

### (2) 路由器型号和固件版本
型号  HiWiFi HC5962
固件版本 OpenWrt R22.3.13 / LuCI Master (git-22.068.45502-a50e601)
内核版本  5.4.175
A：

### (3) 详细日志

A：
系统日志：https://pastebin.com/cikwhHiJ
内核日志：https://pastebin.com/mf1D0QN5

evfun, 2022-03-23:
在关闭硬件流量分载和软件流量分载后，所有ipv6 udp连接又能够恢复正常

yanxin152133, 2022-03-24:
为什么我用的R21.11.11在重启之后前几分钟网络有问题，你这是过几分钟有问题？

evfun, 2022-03-26:
> 为什么我用的R21.11.11在重启之后前几分钟网络有问题，你这是过几分钟有问题？

我的可能是硬件nat加速和ipv6 udp不兼容，关掉就好了。但是还是希望大佬修一下，mt7621纯cpu跑，负载会变得很高

coolsnowwolf, 2026-09-05:
IPv6 UDP 十分钟后单向失效基于旧 5.4.175/HC5962，缺少 conntrack、超时参数和当前内核复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf