coolsnowwolf/lede #9131: SSRP产生大量连入本地1234端口的连接

wolf001100, 2022-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
已经以关键字“1234”搜索，没有发现类似的issue

## 2. 详细叙述

### (1) 具体问题

使用SSRP，Trojan协议，每隔几个小时，便会有大量连接从本地WAN1口连入到本地WAN2：1234端口，一秒钟数十个，每个发送24字节，追踪此连接，会发现内容如下：

User-Agent: NetSystemsResearch studies the availability of various services across the internet. Our website is netsystemsresearch.com

### (2) 路由器型号和固件版本

A：路由器为J1900软路由，移动电信联通三网多播，负载均衡，VPS出口已指定本地WAN1
固件版本：尝试多个作者的5.4和5.10内核版本固件，跨度从2021年1月到今年3月最新，均有此问题

### (3) 详细日志

Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36270 disconnected, 0 bytes received, 24 bytes sent, lasted for 10 seconds
Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36272 disconnected, 0 bytes received, 24 bytes sent, lasted for 9 seconds
Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36273 disconnected, 0 bytes received, 24 bytes sent, lasted for 9 seconds
Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36274 disconnected, 0 bytes received, 24 bytes sent, lasted for 8 seconds
Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36305 authenticated by authenticator (5fe7591)
Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36305 requested connection to 60.4.9.*:1234
Mar 24 03:24:02 debian trojan[3311]: [2022-03-24 03:24:02] [INFO] 183.197.2.*:36275 disconnected, 0 bytes received, 24 bytes sent, lasted for 8 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36279 disconnected, 0 bytes received, 24 bytes sent, lasted for 9 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36280 disconnected, 0 bytes received, 24 bytes sent, lasted for 8 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36281 disconnected, 0 bytes received, 24 bytes sent, lasted for 8 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36306 authenticated by authenticator (5fe7591)
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36306 requested connection to 60.4.9.*:1234
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36282 disconnected, 0 bytes received, 24 bytes sent, lasted for 8 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36283 disconnected, 0 bytes received, 24 bytes sent, lasted for 7 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36284 disconnected, 0 bytes received, 24 bytes sent, lasted for 7 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36286 disconnected, 0 bytes received, 24 bytes sent, lasted for 7 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36307 authenticated by authenticator (5fe7591)
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36307 requested connection to 60.4.9.*:1234
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36287 disconnected, 0 bytes received, 24 bytes sent, lasted for 6 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36288 disconnected, 0 bytes received, 24 bytes sent, lasted for 6 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36289 disconnected, 0 bytes received, 24 bytes sent, lasted for 6 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36290 disconnected, 0 bytes received, 24 bytes sent, lasted for 5 seconds
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36309 authenticated by authenticator (5fe7591)
Mar 24 03:24:03 debian trojan[3311]: [2022-03-24 03:24:03] [INFO] 183.197.2.*:36309 requested connection to 60.4.9.*:1234
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36291 disconnected, 0 bytes received, 24 bytes sent, lasted for 6 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36292 disconnected, 0 bytes received, 24 bytes sent, lasted for 6 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36293 disconnected, 0 bytes received, 24 bytes sent, lasted for 5 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36310 authenticated by authenticator (5fe7591)
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36310 requested connection to 60.4.9.*:1234
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36295 disconnected, 0 bytes received, 24 bytes sent, lasted for 5 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36296 disconnected, 0 bytes received, 24 bytes sent, lasted for 4 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36297 disconnected, 0 bytes received, 24 bytes sent, lasted for 4 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36299 disconnected, 0 bytes received, 24 bytes sent, lasted for 4 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36311 authenticated by authenticator (5fe7591)
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36311 requested connection to 60.4.9.*:1234
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36300 disconnected, 0 bytes received, 24 bytes sent, lasted for 3 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36301 disconnected, 0 bytes received, 24 bytes sent, lasted for 3 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36302 disconnected, 0 bytes received, 24 bytes sent, lasted for 3 seconds
Mar 24 03:24:04 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36303 disconnected, 0 bytes received, 24 bytes sent, lasted for 2 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36312 authenticated by authenticator (5fe7591)
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:04] [INFO] 183.197.2.*:36312 requested connection to 60.4.9.*:1234
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36304 disconnected, 0 bytes received, 24 bytes sent, lasted for 3 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36305 disconnected, 0 bytes received, 24 bytes sent, lasted for 3 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36306 disconnected, 0 bytes received, 24 bytes sent, lasted for 2 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36307 disconnected, 0 bytes received, 24 bytes sent, lasted for 2 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36313 authenticated by authenticator (5fe7591)
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36313 requested connection to 60.4.9.*:1234
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36309 disconnected, 0 bytes received, 24 bytes sent, lasted for 2 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36310 disconnected, 0 bytes received, 24 bytes sent, lasted for 1 seconds
Mar 24 03:24:05 debian trojan[3311]: [2022-03-24 03:24:05] [INFO] 183.197.2.*:36311 disconnected, 0 bytes received, 24 bytes sent, lasted for 1 seconds
[... 137 lines cut ...]
Mar 24 03:24:22 debian trojan[3311]: [2022-03-24 03:24:22] [INFO] 183.197.2.*:36504 authenticated by authenticator (5fe7591)
Mar 24 03:24:22 debian trojan[3311]: [2022-03-24 03:24:22] [INFO] 183.197.2.*:36504 requested connection to 60.4.9.*:1234
Mar 24 03:24:22 debian trojan[3311]: [2022-03-24 03:24:22] [INFO] 183.197.2.*:36504 disconnected, 0 bytes received, 24 bytes sent, lasted for 0 seconds
Mar 24 03:24:22 debian trojan[3311]: [2022-03-24 03:24:22] [INFO] 183.197.2.*:36505 authenticated by authenticator (5fe7591)
Mar 24 03:24:22 debian trojan[3311]: [2022-03-24 03:24:22] [INFO] 183.197.2.*:36505 requested connection to 60.4.9.*:1234
Mar 24 03:24:22 debian trojan[3311]: [2022-03-24 03:24:22] [INFO] 183.197.2.*:36505 disconnected, 0 bytes received, 24 bytes sent, lasted for 0 seconds
[... cut ...]
Mar 24 03:24:23 debian trojan[3311]: [2022-03-24 03:24:23] [INFO] 183.197.2.*:36508 authenticated by authenticator (5fe7591)
Mar 24 03:24:23 debian trojan[3311]: [2022-03-24 03:24:23] [INFO] 183.197.2.*:36508 requested connection to 60.4.9.*:1234

日志IP已做隐藏处理。

其中，INFO后的183.197.2.*为WAN1，60.4.9.*为WAN2

由于此问题已导致多个VPS服务商以怀疑散播spam为由导致suspend，故不得已上来求助。

此现象貌似与VPS端服务器版本无关，协议无关，尝试过Vless, Trojan, SS，只要是客户端为shadowsocks皆有此现象

Passwall，Helloworld，Clash没有此现象

尝试删除节点中的“本地端口”参数，UI会提示“一些项目无效，无法保存”，直接删除配置文件的此行参数，会导致服务无法启动。

盼复，谢谢。

deplives, 2022-03-27:
所以这和 lede 有啥关系么？

minlang112, 2022-03-27:
看上去就是VPS那边被爬虫扫了……

wolf001100, 2022-03-27:
回复一下上面两位：
1.,确实关系不大。。。我上面说了，搜索ssrp，给了两个连接，都是指向了此repo，所以就写在这儿了。后来觉得不合适，就close了。。。。但是转念一想，万一这里有高人呢？所以就reopen了。。

2.这个现象只出现在SSRP这个插件上，同样的环境，使用其他的插件就无此现象。

closed 2022-04-09 by wolf001100