coolsnowwolf/lede #9132: GITHUB ACTIONS 云编译失败

thomaswcy, 2022-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：GitHub Actions 云编译失败

### (2) 路由器型号和固件版本

A：R2S ARM架构

### (3) 详细日志

A：
```
ake[2]: Entering directory '/home/runner/work/R2S/R2S/openwrt/package/kernel/linux'
Package kmod-fs-virtiofs is missing dependencies for the following libraries:
virtio.ko
virtio_ring.ko
make[2]: *** [modules/fs.mk:631: /home/runner/work/R2S/R2S/openwrt/bin/targets/rockchip/armv8/packages/kmod-fs-virtiofs_5.15.31-1_aarch64_generic.ipk] Error 1
make[2]: Leaving directory '/home/runner/work/R2S/R2S/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#3.44#0.30#3.72
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/R2S/R2S/openwrt'
make: *** [/home/runner/work/R2S/R2S/openwrt/include/toplevel.mk:230: package/compile] Error 2
Error: Process completed with exit code 2.
```

WYC-2020, 2022-03-27:
内核那个版本，等空了有时间修复一下

thomaswcy, 2022-03-27:
谢谢大佬！

helmiau, 2022-04-01:
same on me, anyone have the fix?
@thomaswcy @WYC-2020

thomaswcy, 2022-04-03:
I don't have the fix. I also want to fix it, but I don't know how to fix it.

thomaswcy, 2022-04-13:
@WYC-2020

WYC-2020, 2022-04-14:
> I don't have the fix. I also want to fix it, but I don't know how to fix it.

哈哈忘记这事了，等下我看下

WYC-2020, 2022-04-14:
> @WYC-2020

https://github.com/WYC-2020/lede/commit/da695b8e183e2e39ada5853a0266bd97a8458fa9
简单修复了一下，我更加倾向于吧virtio搞成模块的样子，而不是像这样字作为默认的模块，毕竟容量有限

thomaswcy, 2022-04-26:
谢谢（不过我已经改用官方21.02了...）

coolsnowwolf, 2026-09-05:
GitHub Actions 日志来自 2022 年 rockchip 5.15 配置，`kmod-fs-virtiofs` 缺少 virtio 模块依赖；当前目标配置已演进，无法在现行代码复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf