coolsnowwolf/lede #9135: 挺尴尬的，又出现了，newifi-d2使用5.10.108内核时lan口不工作，WiFi正常，wan口正常。按照#8925修复之后出现flow_offload_hw_path错误

uniessey, 2022-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有，但是没起作用；
## 2. 详细叙述

### (1) 具体问题

A：newifi-d2使用5.10.108内核时lan口不工作，WiFi正常，wan口正常。按照#8925修复之后出现flow_offload_hw_path错误。
编译信息附后

### (2) 路由器型号和固件版本

A：newifi-d2使用5.10.108内核

### (3) 详细日志

A：
内核日志：
[    0.000000] Linux version 5.10.108 (uniessey@uniessey-machine) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4369-5a916c954) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Fri Mar 25 15:00:09 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30608 r8192 d22640 u61440
[    0.000000] pcpu-alloc: s30608 r8192 d22640 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00039a10
[    0.000000] Readback ErrCtl register=00039a10
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508392K/524288K available (6705K kernel code, 623K rwdata, 1396K rodata, 1244K init, 235K bss, 15896K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x3d0/0x5e4 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000013] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007950] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.016937] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.083101] pid_max: default: 32768 minimum: 301
[    0.087873] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.095077] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... 550 lines cut ...]
[   80.852084] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x36, hi_lna1 = 0x36
[   80.859210] mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x36, hi_lna1 = 0x36
[   80.866227] original vga value(chain0) = 42
[   80.870412] original vga value(chain1) = 42
[   80.874589] <==== rt28xx_init, Status=0
[   80.878414] get_dev_name_prefix(): dev_idx = 0, dev_name_prefix=wdsi
[   80.886486] get_dev_name_prefix(): dev_idx = 0, dev_name_prefix=wdsi
[   80.894595] get_dev_name_prefix(): dev_idx = 0, dev_name_prefix=wdsi
[   80.902654] get_dev_name_prefix(): dev_idx = 0, dev_name_prefix=wdsi
[   80.910771] get_dev_name_prefix(): dev_idx = 0, dev_name_prefix=apclii
[   80.919078] RTMPDrvOpen(1):Check if PDMA is idle!
[   80.923839] RTMPDrvOpen(2):Check if PDMA is idle!
[   80.928584] WiFi Interface Up Time (rai0): 5.360s
[   80.935292] br-lan: port 2(rai0) entered blocking state
[   80.940577] br-lan: port 2(rai0) entered disabled state
[   80.946336] device rai0 entered promiscuous mode
[   80.951371] br-lan: port 2(rai0) entered blocking state
[   80.956618] br-lan: port 2(rai0) entered forwarding state
[   94.160003] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   94.166935] jffs2_build_filesystem(): unlocking the mtd device... 
[   94.166941] done.
[   94.175075] jffs2_build_filesystem(): erasing all blocks after the end marker... 
[  133.386948] done.
[  133.396549] jffs2: notice: (4359) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[  134.386499] overlayfs: upper fs does not support tmpfile.

uniessey, 2022-03-27:
系统日志：
Fri Mar 25 23:00:35 2022 kern.notice kernel: [    0.000000] Linux version 5.10.108 (uniessey@uniessey-machine) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4369-5a916c954) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Fri Mar 25 15:00:09 2022
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] MIPS: machine is Newifi-D2
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Mar 25 23:00:35 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Zone ranges:
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Early memory node ranges
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30608 r8192 d22640 u61440
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s30608 r8192 d22640 u61440 alloc=15*4096
Fri Mar 25 23:00:35 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
Fri Mar 25 23:00:35 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00039a10
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00039a10
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] Memory: 508392K/524288K available (6705K kernel code, 623K rwdata, 1396K rodata, 1244K init, 235K bss, 15896K reserved, 0K cma-reserved, 262144K highmem)
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Fri Mar 25 23:00:35 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x3d0/0x5e4 with crng_init=0
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.000013] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.007950] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.016937] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.083101] pid_max: default: 32768 minimum: 301
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.087873] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Mar 25 23:00:35 2022 kern.info kernel: [    0.095077] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... cut ...]
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/uniessey/lede/target/linux'
time: target/linux/compile#12.80#7.66#19.43
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/uniessey/lede'
make[1]: *** [target/Makefile:18: /home/uniessey/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/uniessey/lede'
make: *** [/home/uniessey/lede/include/toplevel.mk:230：world] 错误 2

uniessey, 2022-03-27:
编译信息：

ake[5]: Leaving directory '/home/uniessey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.108'
make -C /home/uniessey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.108 KCFLAGS="-fmacro-prefix-map=/home/uniessey/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/uniessey/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Mar 26 01:36:39 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/uniessey/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/uniessey/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.108 CC="mipsel-openwrt-linux-musl-gcc" vmlinux vmlinuz modules
make[5]: Entering directory '/home/uniessey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.108'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/net/ethernet/ralink/mtk_eth_soc.o
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:41:
drivers/net/ethernet/ralink/mtk_eth_soc.h:533:15: warning: 'struct flow_offload_hw_path' declared inside parameter list will not be visible outside of this definition or declaration
        struct flow_offload_hw_path *src,
               ^~~~~~~~~~~~~~~~~~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1519:10: warning: 'struct flow_offload_hw_path' declared inside parameter list will not be visible outside of this definition or declaration
   struct flow_offload_hw_path *src,
          ^~~~~~~~~~~~~~~~~~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c: In function 'fe_flow_offload':
drivers/net/ethernet/ralink/mtk_eth_soc.c:1524:9: error: dereferencing pointer to incomplete type 'struct flow_offload_hw_path'
  if (src->dev != dest->dev)
         ^~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1529:44: error: passing argument 4 of 'mtk_flow_offload' from incompatible pointer type [-Werror=incompatible-pointer-types]
  return mtk_flow_offload(priv, type, flow, src, dest);
                                            ^~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:41:
drivers/net/ethernet/ralink/mtk_eth_soc.h:533:37: note: expected 'struct flow_offload_hw_path *' but argument is of type 'struct flow_offload_hw_path *'
        struct flow_offload_hw_path *src,
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1529:49: error: passing argument 5 of 'mtk_flow_offload' from incompatible pointer type [-Werror=incompatible-pointer-types]
  return mtk_flow_offload(priv, type, flow, src, dest);
                                                 ^~~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:41:
drivers/net/ethernet/ralink/mtk_eth_soc.h:534:37: note: expected 'struct flow_offload_hw_path *' but argument is of type 'struct flow_offload_hw_path *'
        struct flow_offload_hw_path *dest);
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c: At top level:
drivers/net/ethernet/ralink/mtk_eth_soc.c:1551:3: error: 'const struct net_device_ops' has no member named 'ndo_flow_offload'
  .ndo_flow_offload = fe_flow_offload,
   ^~~~~~~~~~~~~~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1551:22: error: initialization of 'int (*)(struct net_device *, int,  u8 *)' {aka 'int (*)(struct net_device *, int,  unsigned char *)'} from incompatible pointer type 'int (*)(enum flow_offload_type,  struct flow_offload *, struct flow_offload_hw_path *, struct flow_offload_hw_path *)' [-Werror=incompatible-pointer-types]
  .ndo_flow_offload = fe_flow_offload,
                      ^~~~~~~~~~~~~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1551:22: note: (near initialization for 'fe_netdev_ops.ndo_set_vf_mac')
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:280: drivers/net/ethernet/ralink/mtk_eth_soc.o] Error 1
make[8]: *** [scripts/Makefile.build:497: drivers/net/ethernet/ralink] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/net/ethernet] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: *** [Makefile:1822: drivers] Error 2
make[5]: Leaving directory '/home/uniessey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.108'
make[4]: *** [Makefile:25: /home/uniessey/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.108/.modules] Error 2
make[4]: Leaving directory '/home/uniessey/lede/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/uniessey/lede/target/linux'
time: target/linux/compile#12.80#7.66#19.43
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/uniessey/lede'
make[1]: *** [target/Makefile:18: /home/uniessey/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/uniessey/lede'
make: *** [/home/uniessey/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-03-27:
什么8925。贴链接，这个玩意5.10是不支持offload的，用5.4，有人说xr那边修复了，我看了，并没有，那只是解决了编译，也是不能支持odfload的，都没有任何地方调用offload相关代码

uniessey, 2022-03-27:
> 什么8925。贴链接，这个玩意5.10是不支持offload的，用5.4，有人说xr那边修复了，我看了，并没有，那只是解决了编译，也是不能支持odfload的，都没有任何地方调用offload相关代码

好的，大佬。就是这个连接中的修复：https://github.com/coolsnowwolf/lede/commit/29d4ee71ed4ce51fceeb0e5b3b85555f7f81ede0

WYC-2020, 2022-03-27:
> > 什么8925。贴链接，这个玩意5.10是不支持offload的，用5.4，有人说xr那边修复了，我看了，并没有，那只是解决了编译，也是不能支持odfload的，都没有任何地方调用offload相关代码
> 
> 好的，大佬。就是这个连接中的修复：[29d4ee7](https://github.com/coolsnowwolf/lede/commit/29d4ee71ed4ce51fceeb0e5b3b85555f7f81ede0)

这个修理我说过，不行，只是当时没人在意而已，你用我仓库的编译 试下可以工作不，然后我在给你说怎么修复

uniessey, 2022-03-28:
> > > 什么8925。贴链接，这个玩意5.10是不支持offload的，用5.4，有人说xr那边修复了，我看了，并没有，那只是解决了编译，也是不能支持odfload的，都没有任何地方调用offload相关代码
> > 
> > 
> > 好的，大佬。就是这个连接中的修复：[29d4ee7](https://github.com/coolsnowwolf/lede/commit/29d4ee71ed4ce51fceeb0e5b3b85555f7f81ede0)
> 
> 这个修理我说过，不行，只是当时没人在意而已，你用我仓库的编译 试下可以工作不，然后我在给你说怎么修复

收到，马上去试试，然后反馈。

uniessey, 2022-03-30:
> 

用您的代码的确可以正常编译，也能很好的刷进去工作，除了总览界面信息不完整之外，其他没发现问题。编译了接近40个小时。

uniessey, 2022-03-30:
> > > 什么8925。贴链接，这个玩意5.10是不支持offload的，用5.4，有人说xr那边修复了，我看了，并没有，那只是解决了编译，也是不能支持odfload的，都没有任何地方调用offload相关代码
> > 
> > 
> > 好的，大佬。就是这个连接中的修复：[29d4ee7](https://github.com/coolsnowwolf/lede/commit/29d4ee71ed4ce51fceeb0e5b3b85555f7f81ede0)
> 
> 这个修理我说过，不行，只是当时没人在意而已，你用我仓库的编译 试下可以工作不，然后我在给你说怎么修复

用您的代码的确可以正常编译，也能很好的刷进去工作，除了总览界面信息不完整之外，其他没发现问题。编译了接近40个小时。

WYC-2020, 2022-03-30:
> 用您的代码的确可以正常编译，也能很好的刷进去工作，除了总览界面信息不完整之外，其他没发现问题。编译了接近40个小时。

什么信息不完整，截图看下

closed 2022-07-25 by aiamadeus