coolsnowwolf/lede #9136: 默认配置选择就选择 Subtarget (AMD Geode based systems)  编译报错

fanhengyuan, 2022-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：两台电脑编译AMD都是出现这个问题。 又虚拟机安装了 ubuntu 20.04 编译还是出现这个问题。

### (2) 路由器型号和固件版本

A：x86 AMD

### (3) 详细日志

A：

```bash
make[3]: *** [modules/video.mk:263: /home/ff/lede/bin/targets/x86/geode/packages/kmod-drm_5.15.31-1_i386_pentium-mmx.ipk] Error 1
make[3]: Leaving directory '/home/ff/lede/package/kernel/linux'
time: package/kernel/linux/compile#10.33#2.09#11.66
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/ff/lede'
make[1]: *** [package/Makefile:110: /home/ff/lede/staging_dir/target-i386_pentium-mmx_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ff/lede'
make: *** [/home/ff/lede/include/toplevel.mk:230：world] 错误 2
```

fanhengyuan, 2022-03-27:
`Package kmod-drm is missing dependencies for the following libraries:`
请问一下这个问题要怎么解决

fanhengyuan, 2022-03-27:
自行修改 `package/kernel/linux/modules/video.mk` 254 行增加
```bash
 242 define KernelPackage/drm
 243   SUBMENU:=$(VIDEO_MENU)
 244   TITLE:=Direct Rendering Manager (DRM) support
 245   HIDDEN:=1
 246   DEPENDS:=+kmod-dma-buf +kmod-i2c-core +kmod-i2c-algo-bit  +PACKAGE_kmod-backlight:kmod-backlight
 247   KCONFIG:=     \
 248         CONFIG_DRM      \
 249         CONFIG_DRM_PANEL_ORIENTATION_QUIRKS=y   \
 250         CONFIG_DRM_FBDEV_EMULATION=y \
 251         CONFIG_DRM_FBDEV_OVERALLOC=100 \
 252         CONFIG_HDMI
 253   FILES:= \
 254         $(LINUX_DIR)/drivers/gpu/drm/drm.ko \
 255         $(LINUX_DIR)/drivers/gpu/drm/drm_panel_orientation_quirks.ko \
 256         $(LINUX_DIR)/drivers/video/backlight/backlight.ko \
 257         $(LINUX_DIR)/drivers/video/fbdev/core/fb.ko \
 258         $(LINUX_DIR)/arch/x86/video/fbdev.ko \
 259         $(LINUX_DIR)/lib/fonts/font.ko
 260   AUTOLOAD:=$(call AutoLoad,05,drm)
 261 endef

```
现在又报错：
```bash
Packaged contents of /home/ff/lede/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/packages/ipkg-i386_pentium-mmx/kmod-drm-ttm into /home/ff/lede/bin/targets/x86/geode/packages/kmod-drm-ttm_5.15.31-1_i386_pentium-mmx.ipk
rstrip.sh: /home/ff/lede/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/packages/ipkg-i386_pentium-mmx/kmod-drm-kms-helper/lib/modules/5.15.31/drm_kms_helper.ko: relocatable
Packaged contents of /home/ff/lede/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/packages/ipkg-i386_pentium-mmx/kmod-drm-kms-helper into /home/ff/lede/bin/targets/x86/geode/packages/kmod-drm-kms-helper_5.15.31-1_i386_pentium-mmx.ipk
Package kmod-drm-i915 is missing dependencies for the following libraries:
intel-gtt.ko
video.ko
make[3]: *** [modules/video.mk:1111: /home/ff/lede/bin/targets/x86/geode/packages/kmod-drm-i915_5.15.31-1_i386_pentium-mmx.ipk] Error 1
make[3]: Leaving directory '/home/ff/lede/package/kernel/linux'
time: package/kernel/linux/compile#11.60#2.57#12.99
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/ff/lede'
make[1]: *** [package/Makefile:110: /home/ff/lede/staging_dir/target-i386_pentium-mmx_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ff/lede'
make: *** [/home/ff/lede/include/toplevel.mk:230：world] 错误 2

```

coolsnowwolf, 2026-09-05:
当前 `kmod-drm` 已显式依赖 `kmod-backlight`，并同步打包所需 fb/backlight 模块；旧 Geode 缺少 `backlight.ko`/`fb.ko` 的依赖错误已修正，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf