coolsnowwolf/lede #9137: 最近固件编译失败

sildem, 2022-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译失败(仅加了open clash和argon theme)

### (2) 路由器型号和固件版本

A：TP-Link Archer C7 v2， Lean大最新版本固件

### (3) 详细日志

A：请参见：

2022-03-27T09:07:41.2407936Z sort: write failed: 'standard output': Broken pipe
2022-03-27T09:07:41.2408494Z sort: write error
2022-03-27T09:07:45.6643320Z Unrecognized dir format '/etc/rc.d/K15
2022-03-27T09:07:45.6644188Z openclash ../init.d/openclash 777 0 0' line 396
2022-03-27T09:07:45.6644899Z Unrecognized dir format '/etc/rc.d/S99
2022-03-27T09:07:45.6645587Z openclash ../init.d/openclash 777 0 0' line 437
2022-03-27T09:07:45.7105029Z make[6]: *** [usr/Makefile:63: usr/initramfs_data.cpio] Error 255
2022-03-27T09:07:45.7105843Z make[5]: *** [Makefile:1734: usr] Error 2
2022-03-27T09:07:45.7109195Z make[4]: *** [Makefile:42: /home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.187/.image] Error 2
2022-03-27T09:07:45.7111749Z make[5]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.187'
2022-03-27T09:07:45.7112229Z make[3]: *** [Makefile:11: install] Error 2
2022-03-27T09:07:45.7114691Z make[4]: Leaving directory '/home/runner/work/lede/lede/target/linux/ath79'
2022-03-27T09:07:45.7115623Z make[3]: Leaving directory '/home/runner/work/lede/lede/target/linux'
2022-03-27T09:07:45.7116408Z time: target/linux/install#13.13#4.21#15.51
2022-03-27T09:07:45.7119309Z     ERROR: target/linux failed to build.
2022-03-27T09:07:45.7121354Z make[2]: *** [target/Makefile:25: target/linux/install] Error 1
2022-03-27T09:07:45.7126895Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2022-03-27T09:07:45.7163140Z make[1]: *** [target/Makefile:19: /home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
2022-03-27T09:07:45.7163859Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2022-03-27T09:07:45.7166776Z make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
2022-03-27T09:07:45.7406788Z ##[error]Process completed with exit code 2.
2022-03-27T09:07:45.8374787Z Post job cleanup.
2022-03-27T09:07:46.0705968Z [command]/usr/bin/git version
2022-03-27T09:07:46.0747599Z git version 2.35.1
2022-03-27T09:07:46.0795459Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2022-03-27T09:07:46.0816106Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2022-03-27T09:07:46.1148731Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2022-03-27T09:07:46.1168188Z http.https://github.com/.extraheader
2022-03-27T09:07:46.1176159Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2022-03-27T09:07:46.1228929Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2022-03-27T09:07:46.1670624Z Cleaning up orphan processes

WYC-2020, 2022-03-27:
单线程

1054009064, 2022-03-27:
> ...
> A：云编译失败(仅加了open clash和argon theme)
>...
> A：TP-Link Archer C7 v2， Lean大最新版本固件
>...

[image]
默认编译能过的

jim-jiang-github, 2022-05-26:
我也遇到了，加入OpenClash就编译不过了。。。
WSL2下，单线程编译。
去掉OpenClash就没有问题

closed 2022-07-25 by aiamadeus