coolsnowwolf/lede #9138: rtl8812au驱动导致编译失败

canghaiwuhen, 2022-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：pull代码后，编译不通过。

### (2) 路由器型号和固件版本

A：x86版本

### (3) 详细日志

A：
make[4]: Entering directory '/Data/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.31'
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_cmd.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_security.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_debug.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_io.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_ioctl_query.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_ioctl_set.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_ieee80211.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_mlme.o
  CC [M]  /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_mlme_ext.o
/Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_mlme_ext.c: In function 'mgt_dispatcher':
/Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_mlme_ext.c:756:5: error: this statement may fall through [-Werror=implicit-fallthrough=]
   if(check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
     ^
/Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_mlme_ext.c:761:2: note: here
  case WIFI_ASSOCREQ:
  ^~~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:277: /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/core/rtw_mlme_ext.o] Error 1
make[4]: *** [Makefile:1868: /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8] Error 2
make[4]: Leaving directory '/Data/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.31'
make[3]: *** [Makefile:57: /Data/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ct-2020-12-07-1e9689c8/.built] Error 2
make[3]: Leaving directory '/Data/lede/package/kernel/rtl8812au-ct'
time: package/kernel/rtl8812au-ct/compile#5.82#0.89#6.68
    ERROR: package/kernel/rtl8812au-ct failed to build.
make[2]: *** [package/Makefile:116: package/kernel/rtl8812au-ct/compile] Error 1
make[2]: Leaving directory '/Data/lede'
make[1]: *** [package/Makefile:110: /Data/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Data/lede'
make: *** [/Data/lede/include/toplevel.mk:230: world] Error 2

Checkra2n, 2022-03-27:
我是80211编译错误
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.o
/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_temp_notif':
/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:207:3: error: implicit declaration of function 'thermal_notify_framework' [-Werror=implicit-function-declaration]
   thermal_notify_framework(tz_dev->tzone,
   ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:277: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.o] Error 1
make[11]: *** [scripts/Makefile.build:540: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm] Error 2
make[10]: *** [scripts/Makefile.build:540: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel] Error 2
make[8]: *** [scripts/Makefile.build:540: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1868: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: *** [Makefile:584: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.85#0.86#9.07
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-03-27:
哈哈

canghaiwuhen, 2022-03-27:
在kernel mode中去掉这个模块编译就可以通过，再加上就又报错

WYC-2020, 2022-03-28:
> 上一次的编译是在2月22日，同样的配置文件正常。 3月27日编译报错，在kernel mode中去掉这个模块编译就可以通过，再加上就又报错
 
没办法内核升级太快，没事别追新
https://github.com/WYC-2020/lede/commit/2442499bc9a5c653e1a346b57b26e4b64f0be5db

canghaiwuhen, 2022-03-30:
今天git pull后编译OK

closed 2022-03-30 by canghaiwuhen

www159, 2022-04-04:
感谢这个发射，解决我的燃煤之急