coolsnowwolf/lede #9139: 网易云插件保存失败

MGBB, 2022-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
网易云解锁插件golang版本切换nodejs版本保存后，再由切换nodejs版本切换 golang版本无法保存，luci界面显示“一个或多个必选项值为空！”
### (1) 具体问题

A：切换多个主题均无法保存配置

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

MGBB, 2022-03-27:
[](url)
[image]

WYC-2020, 2022-03-27:
这玩意以前就修复过了，自己找相似issue

MGBB, 2022-03-27:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ ]  没有类似的issue
> 
> ## 2. 详细叙述
> 网易云解锁插件golang版本切换nodejs版本保存后，再由切换nodejs版本切换 golang版本无法保存，luci界面显示“一个或多个必选项值为空！”
> 
> ### (1) 具体问题
> A：切换多个主题均无法保存配置
> 
> ### (2) 路由器型号和固件版本
> A：X86
> 
> ### (3) 详细日志
> A：

closed 2022-03-27 by MGBB